FormerLurker/Octolapse #305: Octolapse hangs on "Taking Snapshot"

lawdog1425, 2019-01-19:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version:1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Installed Octolapse and configured for Ender 3 Pro with BLTouch
2. Sliced object with Cura
3. Printed object 

#### What should have happened?
Octolapse should have created timelapse of print and object should have printed

#### What happened instead?
Auto bed leveling completed.  Hot end traveled to front left to print a line along the left side of the print bed to clear extruder.  Hot end gets to rearmost part of line and Octolapse says Taking Snapshot and printer stops completely

#### Operating System running OctoPrint and Octolapse

OS Name: Windows
Os Version: 10 Home

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: Vanilla Marlin 1.1.9 "bugfix"

#### Browser and version of browser, operating system running browser

Browser: Google
Browser OS: Chrome

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: multiple Gcode files

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

lawdog1425, 2019-01-19:
I forgot to mention I am using a PiCam V2 for my camera

FormerLurker, 2019-01-21:
Can you download your settings.json file (octolapse settings within the OctoPrint settings screen, there is a download button there) and your gcode file?  I just helped someone get an ender 3 (not pro) working, so perhaps it is settings and not firmware related.  Speaking of which, do you have stock firmware?

Also, in the meanwhile you might want to disable all other plugins and reboot.  Then try another print and see if you still have an issue.  I've seen a few issues involving conflicts with other plugins, especially those that capture and process gcode.

FormerLurker, 2019-01-21:
Great!  If you are willing, it would be great if you could turn the plugins back on one at a time and see which one is causing conflicts.  I could then try to figure out what is breaking and how to fix it.

lawdog1425, 2019-01-21:
Ok I found it.  It is the Grbl support 1.0.2 plugin.  I was trying to use octoprint on a laser engraver.  It never worked but I never removed the plugin and when I re-enabled my plugins after making octolapse work I didn’t re-enable it because I don’t use it.   So when I re-enabled it bingo problem reappeared.

FormerLurker, 2019-01-21:
I figured out the issue.  M400 and M114 are being replaced by the plugin, which will kill Octolapse.  I might be able to work around the M400 issue, but M114 is critical.  Also, it looks like those replacements are pretty important for the Grbl plugin, so there may be no solution :(  I may install it and see if I can find a workaround.

Anyway, thanks for hunting that down for me!

lawdog1425, 2019-01-21:
Great, glad you figured it out.  Like I said I don’t use it so don’t try to fix it on my account.