FormerLurker/Octolapse #306: [Request] After finishing "Animated - Printing" run, have print head move to a final location for Post-Roll

cmeister2, 2019-01-20:
#### If this is a feature request describe it here

For the "Animated - Printing" stabilization, this oscillates around the X axis with a fixed Y axis. This is great and gives a cool effect, but causes the print head to show up in the final post-roll, which can ruin shots with shadow, e.g. https://www.youtube.com/watch?v=rX_bg4b8J5I

It'd be nice if there was an option to, once the print was complete, move the print head to a final location for post-roll. Then I could set this to "Center-Right" or whatever worked for my system, and the final shot would be lit correctly.

Alternatively I'm open to whatever solutions are available!

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-01-21:
I've considered this before, and believe this can be done.  The only way I can think to make it work reliably involves analyzing the entire gcode file ahead of time.  Fortunately I'm working on pre-processing gcode files right now for other reasons (settings extraction).  I'll keep this in mind and see if I can come up with some reasonable way to implement at least something similar.

Thanks for your submission!

cmeister2, 2019-01-22:
In case anyone else has this problem I worked around it with the following solution:

1. In Slic3r, I added custom G-code to the "After layer change" section of my printer - the M240 code is for taking snapshots.

```
;AFTER_LAYER_CHANGE
;LAYER: [layer_num], Z HEIGHT: [layer_z]

M240 ; Take snapshot
```

2. I added the M240 to the end G-code:
```
G4 ; wait
M221 S100
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head up
M240 ; Take final snapshot   <<< new code
G1 X0 Y200; home X axis
M84 ; disable motors
```

3. I configured Octolapse to detect "M240" as the snapshot code, and then selected "Gcode" as the snapshot profile.

This gives you a jump between the penultimate and last frames, but the print head is well out of the way.

Demo: https://www.youtube.com/watch?v=nb0TfTM1Vlo

Looking at it again, I could possibly have a section of Gcode which is literally 30 of the if-statements with M240 commands included as well, increasing layer_z by 1 each time. This would give the smooth transition up to the top.

filipgoc, 2019-03-28:
That's pretty awesome @cmeister2. Good thinking.

seltix5, 2019-07-29:
Hello,
I use cura and I was locking for a solution for this problem too. The animated printing mode is great and it does not really increase printing time or oozing, the only downside really is the post-roll with the printer head... One vanilla solution for this wold be great =)

FormerLurker, 2019-10-12:
I know this is an old issue, but I finally have an update!  I've added the preprocessing that is necessary to make this feature request work, but have not enhanced the stabilizations themselves to support this as of yet.  I'm planning to solve this by adding some snapshot gcode parameters that allow you to specify coordinates within the snapshot command itself.  Also, I still need to allow a mix of snapshot gcodes AND the normal layer triggers to the new version.  It might be a while before all this is finished (probably in V0.5 or 1.0 depending on which comes first).

I'm going to leave this open and will update it at some point when I've made more progress.

nledenyi, 2020-02-14:
I just ran into this will looking for this exact solution. Looking forward to this being implement and many thanks for coming back to old issues for keeping them updated. 
I'm wondering if maybe addig this: "Also, I still need to allow a mix of snapshot gcodes AND the normal layer triggers to the new version" could be a short term workaround (by using end g codes) and maybe a low hanging fruit?