FormerLurker/Octolapse #307: Monoprice Mini Delta Profile [Request]

gotTech18, 2019-01-21:
#### If this is a feature request describe it here

I would like a profile for the Monoprice Mini Delta because I have seen the cool timelapses made with OctoLapse and I would like to make some cool timelapses also. I know this is possible because this Reddit user got it to work, and I don't know how.

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

gotTech18, 2019-01-21:
Here is the [here is a link](https://www.reddit.com/r/mpminidelta/comments/abbvee/i_got_octolapse_sort_of_working/) to the Reddit post.

gotTech18, 2019-01-28:
Anyone there?

gotTech18, 2019-01-29:
> Hey, sorry for the delay. Been a bit underwater with these issues.
>
> Check out this video: https://www.youtube.com/watch?v=YsxjQhWKz5w
>
> It should pretty much explain how to setup a new printer profile, except
> the part about slicer settings is incorrect since those settings have been
> updated since v0.3.4.
>
> Use the print volume described in the reddit post, and set XYZ and E axis
> modes to 'default to absolute' That should be about it!
>
> Let me know if you need any additional help. It's possible that your
> printer is unique somewhat, and I can help if things go wrong.

So, I did what you said and I started a print. I got this (look at
attachment) on Octoprint and it crashed the Web UI, so I had to reload.

FormerLurker, 2019-01-29:
Hmm, that error explosion isn't supposed to happen in 0.3.4.  I thought I rate limited those messages to 1 per second, which give the previous error time to close.  I guess that's not the case.

Anyway, it looks like your print volume is off.  Can you export and post your settings.json file please?  You can find it within OctoPrint Settings (wrench/spanner icon)->Octolapse.  There is a button on every tab to download the file.

Also, if you could slice a very small print for me and post the gcode I would be able to run some tests within the debugger.  That might not be necessary, but sometimes that makes problems obvious.

gotTech18, 2019-01-29:
[Here is the link to the .json file (I had to do it through Google Drive)](https://drive.google.com/file/d/1iWfVDqzSamMpsIs6PmIhXSP_e4C6R74H/view?usp=sharing)

[Here is a link to a sliced file. (Also in Google Drive)](https://drive.google.com/file/d/1H5R8VlTMnJGjgKWcHy0OKlUd0skYSs_f/view?usp=sharing)

FormerLurker, 2019-01-29:
Your Y - Depth is off in your printer volume in the Octolapse printer profile.  Your Y axis is set from -55.0 to -55.0, so it has no depth at all.  I thought I had a validation in the UI for this!  I'll take a look.  Anyway, it is easy to fix.

Change your Y Depth in your Octolapse printer profile to the following:
Y Depth Min = -60
Y Depth Max = 60

That means your Y axis is 120mm, starting at y=-60 all the way to y=60

gotTech18, 2019-01-29:
Oh! How did I overlooked that? I will give that a try and let you know from
there.

gotTech18, 2019-01-29:
Yes! It's working! Thanks!

FormerLurker, 2019-10-12:
Hey @gotTech18, I should have closed this issue out a long time ago, so I'm going to do this now.  FYI, the next version of Octolapse has full support for delta printers and circular beds.  I don't think there is a profile for yours in the default settings (now configurable via a simple make/model selection), so if you want to try it out and work on a profile I'd be happy to include it in the new profile repository!  Let me know.

For now I'm going to close this issue out.  Thank you so much for your participation!!!

closed 2019-10-12 by FormerLurker