FormerLurker/Octolapse #308: exposure auto priority

foxabilo, 2019-01-22:
In answer to your request for further information regarding this feature of some webcams.

According to the UVC spec, 

"The Auto-Exposure Priority Control is used to specify constraints on the Exposure Time Control when the Auto-Exposure Mode Control is set to Auto Mode or Shutter Priority Mode. A value of zero indicates that the frame rate must remain constant. A value of 1 indicates that the frame rate may be dynamically varied by the device."

So the auto-exposure priority control is only useful when the auto-exposure mode is set to auto or shutter priority. Some webcams manufacturer decided to ignore the control if the auto-exposure mode is not one of those, others decided to return an error, and some others probably decided to crash the firmware

FormerLurker, 2019-07-14:
I was reviewing issues and noticed I never responded to this.  I was sure I did.

Thank you for answering my question!  I'm actually redoing the camera settings pages and will be updating my description for that item (apparently also called 'Exposure, Dynamic Framerate').  I'm very glad to have something other than 'I have no idea what this does!"

Again, thank you so much.  I'm going to close this issue since I've taken care of the description.  Post back any time!

FormerLurker, 2019-07-14:
Description updated, closed!

closed 2019-07-14 by FormerLurker