FormerLurker/Octolapse #309: [Request] Using GoPro for timelapse

gabimal, 2019-01-24:
I was wondering if it is possible to use a GoPro for timelapse. It would be interesting to control the GoPro via WiFi, some discussions here:  https://www.raspberrypi.org/forums/viewtopic.php?t=83530 
Thank you!

yet-another-average-joe, 2019-02-05:
Yes you can !

I'm just experimenting with. At the moment I take snapshots on the GoPro SD card. It works over WiFi. The GoPro applications make video conversion, color corrections, cropping, ProTune corrections, and so on really easy, so asking the author for the post is pointless (IMHO...)

I'm using GoPro-Py-Pi library. There are functions to dowload stills to the RasPi, but is it usefull ? I don't think so.

Just install Python 3.6 (or 3.7), create a Python script using GoPro-Py-Pi, and a bash script calling this Python script. Call this bash in "before snapshot script". You can also set all parameters with another script : photo mode, resolution, ProTune, etc.

Worls a treat !

Note that I am an absolute beginner with Python and Bash (but not in codeing). Took me 24H to get it together. If I can, everybody can.

If y'oure interrested, let me know : a few lines to type.

FormerLurker, 2019-02-05:
That's great!  Most of the solutions I have seen involve hacking the camera and voiding the warranty.  I didn't really want to go this route if possible.

Yes I am interested in getting some more info!  Can you post your python, .sh scripts, and Octolapse camera profile settings?  I can probably find the library myself :)  If I get it working I'll post a tutorial.

Then all I need is a gopro.  Fortunately I have a little Patreon cash available (thanks guys!)  Which cameras have wifi capability?  Do some models work/others not with the library?  If you know these answers that would save me some time.

FormerLurker, 2019-02-05:
Well, it seems to me the best way to go about this would be to write a plugin that transforms the GoPro into a streaming server/still snapshot provider.  That way the extra library can be used elsewhere and won't interfere with Octolapse installation if something goes wrong with the package.  It would be trivial to add a hook to allow other plugins to communicate with  it, and to add an api call to fetch images via http get.  It's an added bonus that this library was written in Python, making it really simple to get working with OctoPrint.

Now I just need to get one cheap :)  Thanks for all of the info.  I'll leave this open and will update when I've made some progress.  FYI, I'm finishing up a MAJOR overhaul to the program, so it will be several weeks at least.  Performance massively improved, setup simplified, quality enhanced, and new features added.  Lots to test and debug, but a new camera preferences/control page is next on my ToDo list, so I will definitely get to it.

In the meanwhile I'll try to think of a way for you to make your script work within the current framework and have decent performance.  Currently I can't think of anything, but that doesn't mean I won't!  Feel free to post any other ideas/comments/information in this thread, especially if you have done experiments and have results.

Thanks so much for your time!

yet-another-average-joe, 2019-02-05:
I did some googling about persistance and serialization, and found shelves and pickles. But it is beyond my understanding (I'm familiar with this only under Windows with MFC, and I don't see how it could avoid initiating a communication everytime a program is called and therefore the object recreated !). I will ask the GoPro-py-api author.

Or some synchronisation : knowing the delay between the starting point of the script and the snapshot, the "Before Snapshot Script" could be launched in advance. Complicated : displacements speeds and accelerations would have to be considered. Not a great idea...

I don't know anything about Linux. But is it possible for a Python script to communicate with another one using messages, and exchange data ? "Before Print Start Script" could instanciate the GoPro object, and then receive SHOOT messages from "Before Snapshot Script", and shoot stills. Then, "Before Render Script" could send a STOP message, and terminate "Before Print Start Script".

I just did something funny. In order to simulate a message loop, I defined a pre-print script with an infinite loop. It crashed OctoPrint service, and can't be restarted, even from a console. Had to sudo reboot. The scripts and OctoPrint run in the same process, I presume.

FormerLurker, 2019-02-06:
My instinct is to look into communicating with the process via stdin (just like you would in windows).  I'll look into this while waiting for my tests to complete next time I run one.  I don't think it will be too difficult, but I could be wrong, lol!

FormerLurker, 2019-02-06:
[This looks interesting](http://www.roman10.net/2011/04/21/named-pipe-in-linux-with-a-python-example/)

yet-another-average-joe, 2019-02-06:
Oh yes ! 
I'm an absolute beginner. In order to test comms, 2 consoles might be opened, with a script in each one ?

FormerLurker, 2019-02-06:
I believe that is the idea.  Or you could launch the script via ```sh script_name.sh`` where that script executes the python.  That should launch the script in the background I believe, then you should be able to communicate in the same console.  At least I think, lol!

yet-another-average-joe, 2019-02-06:
I'll try to learn about this tomorrow. Time to sleep. Less than 5 hours left... Good night !

yet-another-average-joe, 2019-02-07:
Hello FormerLurker,

I experimented with named pipes as suggested. And succeded... but only in command line mode. The scripts work this way.

1st Python script instanciates a GoPro object, a FIFO file (on disk, I will see later for a memfile...). It also runs a message loop. It processes 2 messages : "take photo' and "terminate". It starts with the creation of the FIFO file using mkfifo(), then opens the file in 'r' mode, then it loops looking for messages. On the "Terminate" msg, it deletes the file and exit().

2nd Python script is the snnapshot one. It sends "Shoot" messages to the 1st one after opning the FIFO file in 'w' mode.

3rd Python script send the "Terminate" msg to the 1st one. It's supposed to be calles befoire post processing.

I also created 3 bash scripts that call the matching Python scripts. They are invoqued in the Custom Camera Scripts section.

Everything runs perfectly in terminals. 

BUT : when run from within OctoLapse, the 1st one fails, with this error : "File "/home/pi/gopro-py-api/zGoPro.py", line 20, in os.mkfifo(MsgQueue_path) PermissionError: [Errno 13] Permission denied
The 1st one being the heart of this, I can't test the others from OctoLapse.

The error :

Cannot create the FIFO, and I suspect some owner rights I don't understand.

I also failed with the suggested ```sh script_name.sh``, because it is blocking. I Googled about this, read about an ampersand, but whatever I do, it is blocking. I can't create a background process.

Please, be indulgent, I'm a 3 days old Python (and Linux) coder ! I will soon post on Youtube a short video showing a GoPro doing timelapses, without any hack, driven by OctoLapse, so you can see the potential. GoPro's also allow streaming over WiFi (low res), and could do everything (webcam and hi reolution timelapses). My (very) old Hero3+ Black can shoot at 12MP ! Without Bacpac and battery (USB feed), it's very lign weighhed it can be attached to the bed of a cartesian. This being said, the first method I used works great. There's only a performace issue, the object instanciation being very time consuming.

(sorry for my bad english)

FormerLurker, 2019-02-07:
Your English is great :)  

Post your scripts and I'll play with them.  I found someone willing to let me borrow their Hero 3+ black so we'll be on the same page! Once I have acquired it I should be in business for testing.

If I am able to create a plugin for this that would eliminate the instantiation time, but I'd like to get your scripts version done as a proof of concept before too much effort goes into developing a plugin.  That will be a neat project!

Also, your on-the-fly learning is impressive!  When I started Octolapse I had no experience in Python or Knockout, and had very little experience with Linux.  We all have to start somewhere, and diving right in is usually the best way!  Besides, what you are trying to do is quite advanced, so bravo!

FormerLurker, 2019-02-07:
By the way, I'm curious what your native language is if you don't mind telling.

yet-another-average-joe, 2019-02-07:
This very naive code ! There's no error processing !

All files are 'z' prefixed, doing so, they are easy to find (I do all testing with VNC and Pixel)

zGoPro.py :

```
#!/usr/local/bin/Python3.6

import pickle, os, sys
from goprocam import GoProCamera
from pathlib import Path

MsgQueue_path = "/home/pi/gopro-py-api/zMsgQueue"

# let's create the GoPro object
gopro = GoProCamera.GoPro()
print("GoPro object created")

# FAIL FAIL FAIL FAIL FAIL
# get rid of a pre-existing message queue
#MsgQueue = Path(MsgQueue_path)
#if MsgQueue.is_file():
#  print("Previous MsgQueue exists : deleting")
#  os.remove(MsgQueue_path)

# let's create the message queue
os.mkfifo(MsgQueue_path)
MsgQueue = open(MsgQueue_path, 'r')

# message loop
while True:
  msg = MsgQueue.read()
  if msg=="S": # "S" = Shoot
      print("zGoPro received ",  msg)
      print("take photo")
      gopro.take_photo();
  elif msg=="T": # "T" = Terminate
      print("zGoPro received ",  msg)
      print("terminating message pump")
      break;

MsgQueue.close()
print("message queue closed")
os.remove(MsgQueue_path)
print("message queue removed from disk")
```

zShoot.py

...
#!/usr/local/bin/Python3.6

import pickle
import os

# send shoot messages to the gopro process

#wfPath = "./Shoot_2_GoPro"
MsgQueue_path = "/home/pi/gopro-py-api/zMsgQueue"

MsgQueue = open(MsgQueue_path, 'w')
MsgQueue.write("S")		
MsgQueue.close()
...

zTerminate.py

...
#!/usr/local/bin/Python3.6

import pickle
import os

# send shoot messages to the gopro process

#wfPath = "./Shoot_2_GoPro"
MsgQueue_path = "/home/pi/gopro-py-api/zMsgQueue"

MsgQueue = open(MsgQueue_path, 'w')
MsgQueue.write("T")		
MsgQueue.close()
...

sStart.sh :

...
#! /bin/bash

#sh /home/pi/gopro-py-api/zGoPro.sh . &
python3.6 /home/pi/gopro-py-api/zGoPro.py
...

zShoot.sh

...
! /bin/bash

python3.6 /home/pi/gopro-py-api/zShoot.py
...

zTerminaye.sh

...
#! /bin/bash

python3.6 /home/pi/gopro-py-api/zTerminate.py
...

Not sure about the shebangs ;)
I'm french

yet-another-average-joe, 2019-02-08:
Some success...
I get shoots from a script fired by OctoLapse.
Octolapse can delete the GoPro object living in another process.
But I can't figure how to tell OctoLapse to instanciate  GoPro object. But everything's fine if I do it from a console. The 1st problem was solved adding sudo and piping the passwd (safety nazis will have an heart attack, I don't care...)

The bash scripts : (I changed some file names, in order to get things clearer)

1_BeforePrintStart.sh is called by OctoLapse, calling 1_InitGoPro.sh that calls in turn 1_InitGoPro.py
In a console, no problems. In Octolapse, it does not not work.

1_BeforePrintStart.sh :

```
! /bin/bash

echo "raspberry" | sudo -S /home/pi/gopro-py-api/1_InitGoPro.sh

```

1_InitGoPro :

```
#! /bin/bash

echo "raspberry" | nohup sudo -S python3.6 /home/pi/gopro-py-api/1_InitGoPro.py &
 
```

2_BeforeSnapshot.sh : (works in console and in Octolapse)

```
#! /bin/bash

echo "raspberry" | nohup sudo -S python3.6 /home/pi/gopro-py-api/2_Shoot.py

```

5_AfterRender.sh : (works in console, but NOT from OctoLapse)

```
! /bin/bash

echo "raspberry" | nohup sudo -S python3.6 /home/pi/gopro-py-api/5_DeleteGoPro.py

```

I can get GoPro timelapses from Octolapse, but the GoPro MUST be instanciated and deleted from a console... It's extremely frustrating, because it's probably just my lack of knowledge about bash scripts.

FormerLurker, 2019-02-08:
I bet it's something subtle.  Learning a new language or using new tools almost always causes me to lose some hair :)  Does your init script exit when you run it, or does it just run forever?  It's possible that Octolapse kills all of the subprocesses (it does some cleanup to make sure other long running scripts terminate if they time out) even if the script finishes (though I don't think it should in this case).  What if you added the script to your Pi's boot script and just have it always be in the background?

I still haven't received the GoPro I'm going to be borrowing, but I got confirmation today that it's being prepared, just need to figure out how to meet and pick it up.

FormerLurker, 2019-02-08:
Oh, and I'll take a look at your scripts tomorrow to see if there's something I can fix even without the hardware.

yet-another-average-joe, 2019-02-08:
No, I can't get the script to return to cmd line in console, and the problem is there. OctoLapse is stuck until I kill the GoPro objet from a console (it then returns OK, test validated or something like that). I tried combinations of nohup and &, and whatever I do the result is always the same : Octolapse is stuck by an infinite loop from the "GoPro ctor script". It's not a problem with OctoLapse. It's a problem with the script.. Oh, and it it looks like I solved the problem with "5_AfterRender.sh" : I deleted the command "nohup", and OctoLapse can now destroy the GoPro object, the FIFO file, and exit the loop. But still have to launch the 1st script in a console in order to instanciate GoPro and do (virual) timelapses. I feel like it's just a matter of 3 or 4 keystrokes in teh scripts !

I'd like you to have a really working proof of concept before you get your GoPro ;)

FormerLurker, 2019-02-08:
It is much appreciated!  It sounds like you are making progress.  In my experience it always takes a little longer than you expect to make things like this work, especially when you think you're very close ;)

yet-another-average-joe, 2019-02-09:
It seems I forgot to click "Comment" yesterday !

I found the problem for the 1st script not returning to command line. See here https://stackoverflow.com/questions/28281467/running-bash-script-does-not-return-to-terminal-when-using-ampersand-to-run

redirecting to null does nothing. I removed the print calls in my 1st python script : nothing. I even replaced all the print calls in gopro-pi api with no-op and exit commands : nothing ; but I could have done it wrong.

It seems that gopro-py-api refuses to cooperate with background processes ! I will post an issue on Git's author.

yet-another-average-joe, 2019-02-11:
Here's 2 scripts and a few commands that will make Python 3.6 DL/compilation and gopro-py-api installation fast and easy, 
This as been tested about 10 times and should not cause problems.

Video : https://www.youtube.com/watch?v=TML8L3RmN60&feature=youtu.be
This 1st video is only about installation, had no time to shoot timelapses. Coming soon.

1 - Python 3.6 download/compilation/installation - takes 15mn

command lines :

```
sudo -s
nano install-python3.6.sh (creates script file, then just copy/paste text)
chmod +x  install-python3.6.sh
./install-python3.6.sh
```

script (text to be copied/pasted into nano) :

```
echo
echo
echo "Python 3.6 install script for Octopi/Octoprint"
echo
echo

##########################################
# asksure() function found on :
# https://stackoverflow.com/questions/25809999/press-y-to-continue-any-other-ke$
asksure() {
echo -n "This will install Python 3.6. Do you want to proceed (Y/N)? "
while read -r -n 1 -s answer; do
  if [[ $answer = [YyNn] ]]; then
    [[ $answer = [Yy] ]] && retval=0
    [[ $answer = [Nn] ]] && retval=1
    break
  fi
done

echo # just a final linefeed, optics...

return $retval
}

### using it
echo
echo
echo

if asksure; then
 echo "Installing Python 3.6"
else
 echo "Aborting installation..."
 exit
fi

echo "Installing Python 3.6..."
echo
echo

##########################################
# commands found on :
# https://liftcodeplay.com/2017/06/30/how-to-install-python-3-6-on-raspbian-linux-for-raspberry-pi/

apt-get update
# check build tools are installed
apt-get install build-essential checkinstall
#install libraries
apt-get install libreadline-gplv2-dev libncursesw5-dev libssl-dev libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev
# change dirctory to /usr/src
cd /usr/src
# donwnload Python 3.6 sources archive
wget wget https://www.python.org/ftp/python/3.6.0/Python-3.6.0.tgz
# unzip sources
tar xzf Python-3.6.0.tgz
# change directory to sources
cd Python-3.6.0
# configure
bash configure
# compile and link ; -j3 makes compilation 2x faster
# (7:30 vs 15:30 on a RaspBerry Pi 3 B+
make -j3 altinstall
# return to root directory
cd /
echo
echo
echo
echo
echo
echo "test launch Python 3.6 : python3.6 command"
echo "type exit() to quit Python3.6"
echo
echo
python3.6
```

2 - gopro-py-api installation

command lines :

```
sudo nano install-gopro-py-api.sh
sudo chmod +x  install-gopro-py-api.sh
./install-gopro-py-api.sh
```

script "install-gopro-py-api.sh" (text to be copied/pasted into nano) :

```
echo
echo
echo "Python install script for Octopi/Octoprint"
echo
echo

##########################################
# asksure() function found on :
# https://stackoverflow.com/questions/25809999/press-y-to-continue-any-other-ke$
asksure() {
while read -r -n 1 -s answer; do
  if [[ $answer = [YyNn] ]]; then
    [[ $answer = [Yy] ]] && retval=0
    [[ $answer = [Nn] ]] && retval=1
    break
  fi
done

echo # just a final linefeed, optics...

return $retval
}

### using it
echo
echo
echo
echo -n "This will install gopro-py-api. Do you want to proceed (Y/N)? "

if asksure; then
 echo "Installing gopro-py-api"
else
 echo "Aborting installation..."
 exit
fi

echo "Installing gopro-py-api..."
echo
echo

git clone http://github.com/konradit/gopro-py-api
cd gopro-py-api
sudo python3.6 setup.py install
```

yet-another-average-joe, 2019-02-14:
The second video, with OctoLapse configuration and the 2 tiny scripts.

https://youtu.be/GLyWBZpqbUg  (the result is at the very end)

1 - python script

command lines :

```
cd /home/pi/gopro-py-api
nano gopro_take_photo.py
```

script gopro_take_photo.py :

```
#!/usr/bin/ python3.6

from goprocam import GoProCamera
goproCamera = GoProCamera.GoPro()
goproCamera.take_photo()
```

test command :

`python3.6 gopro_take_photo.py
`

2 - bash script :

commands :

```
nano gopro_take_photo.sh
chmod +x gopro_take_photo.sh

```

script :

```
#!/bin/bash

python3.6 /home/pi/gopro-py-api/gopro_take_photo.py
```

test command :

`./gopro_take_photo.sh`

3 - command to be written in "Before Snapshot :

`/home/pi/gopro-py-api/gopro_take_photo.sh`

yet-another-average-joe, 2019-02-23:
Some progress...

I experimented with these 2 gopro-py-api functions :

take_photo()
downloadLastMedia()

take_photo() takes a photo, and returns the file name on the GoPro, this way :

`http://10.5.5.9:8080/videos/DCIM/108GOPRO/GOPR0136.JPG`

downloadLastMedia(filename) takes this file name as argument and downloads the file to the current directory. The file is named like so : 

`108GOPRO-GOPR0136.JPG`

It now *100% SURE* that OctoLapse can handle GoPros without any change in its code. It's just a matter of bash and Python scripts.

1- the http file name must be parsed to get the disk file name that gopro-py-api will set
2- the "take_snapshot.sh" script must use $4, $5 and $6 to rename and move the GoPro file to OctoLapse working directory 

Moving, renaming is not a problem. It must be the same operations that with MS-DOS.
But parsing is another story : I will have to learn string and parsing functions, hoping that there are some, working like C standard string library.

Or it could be done by giving the parameters to the Python script, and then using Python string functions.

FormerLurker, 2019-02-24:
PROGRESS!!!  I still haven't gotten the gopro.  The person who will be lending it to me decided to take it on a ski trip, so it will be after they return.  Not sure exactly when that is unfortunately.

I'm looking forward to seeing the final script!  If you need any help parsing let me know.  I am currently working on a C based gcode parser/position tracker.

yet-another-average-joe, 2019-02-24:
Done !

I have something that could work... but gives timeouts with timeout set to 25000ms !!!! Th GoPro is set to 5Mpx images (2560x1920, 2.5 MB)

2 scripts. I first declared an external camera : 
Snapshot Acquire Script = /home/pi/gopro-py-api/gopro-take-snapshot.sh

```
#!/bin/sh
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# IMPORTANT - You must add gphoto2 to your /etc/sudoers file in order to execute gphoto2 without sudo
# otherwise the following line will fail.

#sudo gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"

#############################################################
#############################################################

python3.6 /home/pi/gopro-py-api/gopro_take_photo.py $6

#############################################################
#############################################################

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_PATH}'." >&2 
  exit 1
fi

```

Now, the Python script called by gopro-take-snapshot.sh : 

```
#!/usr/bin/ python3.6

from goprocam import GoProCamera
import sys
import shutil
import os

# argv[1] : SNAPSHOT_FULL_PATH
#print(sys.argv[1])

#lives in /home/pi/gopro-py-api
os.chdir("/home/pi/gopro-py-api")

#print("init GoPro")
goproCamera = GoProCamera.GoPro()
#print("take photo")
gopro_file_name=goproCamera.take_photo()
#print("start download")

# TAKES LIKE 30sec OR SO !!!!!
# about 1sec with GoPro App on phone !!!
# is this library broken ???????
goproCamera.downloadLastMedia(gopro_file_name)
#print("download : done")

#print(gopro_file_name)

# http://10.5.5.9:8080/videos/DCIM/108GOPRO/GOPR0136.JPG
# 108GOPRO-GOPR0136.JPG
# search for DCIM
# left trim

file_name_pos = gopro_file_name.find("DCIM")

#print(file_name_pos)

gopro_file_name = gopro_file_name[file_name_pos+5 : len(gopro_file_name)]

#print(gopro_file_name)

# replace '/' with '-'
# -> filename

slash = gopro_file_name.find("/")

print(slash)

# can't do gopro_file_name[slash]='-' !!!!!!!!!!!!!!
# "Python strings are immuable" => ??????????????	

guns=gopro_file_name[0 : slash]
roses=gopro_file_name[slash+1 : len(gopro_file_name)]

#print(guns)
#print(roses)

axel_rose=guns+'-'+roses

#print(axel_rose)

# argv[1] : SNAPSHOT_FULL_PATH

# src dst
shutil.move(axel_rose, sys.argv[1])
#print("terminé")
```

But there is a huuuuuuuuuuuge problem : downloading the snapshot takes 30+ seconds. Using GoPro App on the smartphone, it takes only one second...

FormerLurker, 2019-02-24:
Hmm.. sounds like the download will need to happen in the background, then the render script will have to wait for all downloads to finish before starting.  Octolapse does this already internally, but not for scripts :(

Can't wait to finish up what I'm working on now and to get a gopro to play with :)  I'll dig into your script more tomorrow, have to leave here soon.

yet-another-average-joe, 2019-02-24:
I think there's a problem. 22 sec for medium resolution. On my phone, the GoPro app needs 1-2 sec. On the author's demo video, with a higher res, it takes 11 sec from a Hero 3.

Is "$6" correct for passing this parameter to the Python script ?
How could I get a longer max timeout (I mean wher could I hack your sources ?)

FormerLurker, 2019-02-24:
The time out is in camera settings.  It should obey that.

yet-another-average-joe, 2019-02-24:
I'd like to set it to 35 or 40 sec in order to really test ; there's a limit at 25000ms . At the moment I can't get a timelapse, it timeouts with 25000.

yet-another-average-joe, 2019-02-24:
Yesssss !!!!!!!!!!!!!!!!!!!!!!!!!!!!! I found a solution to the problem !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Got it !!!!!!!!!!!!!!!!!!

The bottleneck was the integrated WiFi controller or it's configuration.

I suspected a WLAN performance issue or whatever : I deactivated it, and found a way to get a USB dongle to work (remember, I'm a RasPi/Linux noob). Link below :

And now it creates the GoPro object and donwloads the images at light speed (and probably even a little bit faster !!!). At first I believed that the script exited, but no, it finished the work before I realized it was just working as expected, maybe even faster than the Android GoPro App ! It seems to be faster to DL 8Mpx to the Pi than it is with 5 Mpx to the phone.

I will post on gopro-py-api, maybe it's a known issue.

No need for background processes, it works out of the box with 2 scripts.

So, at the moment, the workaround is to use an external dongle. This is what I wanted to do at first, (in order to keep a WLAN) ; but I gave up, as I did'nt find a way to get 2 WiFi networks to work together. I still have to find how to set the onboard controller with another SSID. I don't know, but there's certainly a solution. Until, now, I failed...

The scripts work OK. The timelapse is created by OctoLapse. Done on a Virtual Printer, the GoPro shooting a watch, but really can't be shown (too high a FOV, to small the watch : on a 30mm watch at 300mm, mode wide, 7Mpx, 3000x2250)

It's 3:45 AM here, I will redo it tomorrow, and put a short unreferenced video on Youtube (15-20 sec).

This request will be completed and GoPro fully supported when I find the solution about the networks. The GoPro part is done (but there is no errors/exceptions handling). I imagine  that Octolapse redirects stderr to the red popups ? I will give it a try tomorrow.

FormerLurker, 2019-02-24:
Oh the sweet taste of success.  Relish these moments, they don't come often :) looking forward to your next update!

yet-another-average-joe, 2019-02-24:
Next update could be powering on/off the camera . But it depends on the model, and there are 2 power on functions. The sensor and/or the chipset tend to get hot. Adds about 5 sec. Doing this, a snapshot would take less than 10 sec : power on, shoot, download, power off. Mode can also easily be set, but depends on the model and the user's preferences. But something's wrong at least with Hero 3 ; see here : https://github.com/KonradIT/gopro-py-api/issues/83 

I opened an issue about file transfers : https://github.com/KonradIT/gopro-py-api/issues/88

yet-another-average-joe, 2019-03-04:
Today, I stress tested Octolapse with a triple timelapse while using TouchUI on a 7" HDMI LCD touchscreen + VNC client duplicating it + OctoPrint in a browser.

- GoPro 3 (WiFi, with a USB dongle and onboard deactivated)
- D5100 (USB)
- 1920p webcam (USB)
- virtual printer

Worked like a charm ! Just a little bit long to take it all togeter at render time :)))
The PSU made a huge difference : a 5 amps brick I had for months and was too lazy to use (a clean and insulated connector to be made for the GPIO, and some plastic protectors for safety)

From time to time, a photo seamed to timeout, DSLR or GoPro, but not sure it really did.

As I also tested some details in the script, I had aborted and failed timelapses ; I found that OctoLapse doesn't always free space in data folder. Found a bunch of folders and files in /home/pi/.octoprint/data/octolapse/snapshots (total 1GB after a few hours of testing). This could be a problem after a few weeks ! OctoLapse should do some cleaning, for example when it starts. But some "autoexec.bat" could do it (don't know much about Linux...) 

I tested a script with GoPro power on then off after taking a picture, but there's a problem : the function to be called for power on depends on the model.

I also attempted to save the GoPro parameters before timelapse, setup the camera for 3D printing, and then restoring previous parameters. Unfortunately, some funtions are buggy (for example syncTime() )
and many others depend on the model (pre or post Hero 3). I gave up !

Tomorrow, I will post the slighly modified scripts. and a short video (1-2mn) with the setup and the results, shooting a clock while printing with the virtual printer. It will be time to do real printer (there is still the exposure and lighting problem to solve)

LMITAB, 2019-03-11:
I would like to try this one... i have a Gopro 6 and running Octolapse on Prusa MK3S

yet-another-average-joe, 2019-03-11:
Ooops, I forgot to upload...

There's no big difference. It's just more readable, with instructions in the comments.

It should work with the H6. You will have to install Python 3.6+ and gopro-py-api

The 2 scripts are in the ZIP :

[gopro-take-snapshot.zip](https://github.com/FormerLurker/Octolapse/files/2954573/gopro-take-snapshot.zip)

yet-another-average-joe, 2019-03-12:
Octopi comes with 2.7 and 3.5
This will install 3.6 without overwriting 2.7
3.x and 2.7 are différent.

LMITAB, 2019-03-12:
should these be edit or ???

SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

yet-another-average-joe, 2019-03-13:
NO !!!
These are the parameters sent by OctoLapse to the scripts.

1 - install Python 3.6 (script is above)
2 - install gopro-py-api (script is above)
3 - create or copy the scripts from the archive to /home/pi/scripts (instructions are in the scripts headers)
4 - call gopro-take-snapshot.sh from OctoLapse field "External Camera Setup, Snapshot Acquire Script : see screen copy

[image]

and that's it !

I made a video about the Python and gopro-py-api (link is above)
But do NOT follow the instructions of the second video, as it was experimental, and it is obsolete (I have to make another video for the last 2 scripts. But I'm currently busy on another project...

Maybe it would be better you wait a few days, so I could find time to upload some footage. I thought that nobody was interested in Octolapse+GoPro timelapses. As a result, I delayed the updated tutorial video...

LMITAB, 2019-03-13:
ok, how should the gopro be connected.. thru wifi or

yet-another-average-joe, 2019-03-13:
WiFi ! Do NOT use the onboard adapter. Deactivate it. It sucks. Use a WiFi dongle, or you will get many timeout errors. 

Use a USB power supply. With the H3, if the USB is connected to a computer, it turns into mass storage mode. So, use a mains adapter or a power bank, or chances are you will not be able to take photos. I don't know the H6 (nor 5 and 4), but with the H3 it is better not to use a case, to turn the LEDs and LCD display OFF, and also the beep, or it will drive you crazy !

LMITAB, 2019-03-13:
ok, right now my gorpo just keep on saying wifi not ready ....

LMITAB, 2019-03-13:
got something runnig :)

/home/pi/scripts/gopro-take-snapshot.sh
Creating directory:
mkdir: cannot create directory ‘’: No such file or directory
/home/pi/scripts/gopro-take-snapshot.sh: 34: /home/pi/scripts/gopro-take-snapshot.sh:
        We just replace the call to gphoto2 with a call
        to a Python script that will shoot photos with the GoPro.

        It takes one argument : ▒6, the full path name to the
        temporary file that OctoLapse will use to build timelapse

: not found
Traceback (most recent call last):
  File "/home/pi/scripts/gopro-take-snapshot.py", line 34, in <module>
    os.chdir("/home/pi/gopro-py-api")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/gopro-py-api'
The snapshot was not found in the expected directory: ''.

LMITAB, 2019-03-13:
pi@octopi:~ $ /home/pi/scripts/gopro-take-snapshot.sh
Creating directory:
mkdir: cannot create directory ‘’: No such file or directory
/home/pi/scripts/gopro-take-snapshot.sh: 34: /home/pi/scripts/gopro-take-snapshot.sh:
        We just replace the call to gphoto2 with a call
        to a Python script that will shoot photos with the GoPro.

        It takes one argument : ▒6, the full path name to the
        temporary file that OctoLapse will use to build timelapse

: not found
Waking up...
Waking up...
Waking up...
Error, camera not defined.
Traceback (most recent call last):
  File "/home/pi/scripts/gopro-take-snapshot.py", line 57, in <module>
    urlFileName=cam.take_photo() # default : delay = 1sec
  File "/usr/local/lib/python3.6/site-packages/goprocam-3.0.3-py3.6.egg/goprocam/GoProCamera.py", line 453, in take_photo
    if "HERO5 Black" in self.infoCamera(constants.Camera.Name) or "HERO6" in self.infoCamera(constants.Camera.Name):
TypeError: argument of type 'NoneType' is not iterable
The snapshot was not found in the expected directory: ''.

yet-another-average-joe, 2019-03-13:
Waking up means the GoPro was not ON, or the WiFi OFF or communication error.
WiFi must be configured in /boot/octopi-wpa-​supplicant.txt. if you use a WiFi dongle, the configuration is more complicated.

At least, you have the libraries and the scripts installed !

LMITAB, 2019-03-13:
when i execute the script now... it seams to lock the octoprint for a while (not sure how long).. i get no errors...but octopi seams to lock it self and " waiting" for the script too come back...

after a few minutes it seam to timeout and octoprint is back operating

LMITAB, 2019-03-13:
There were 1 errors while applying custom camera settings/scripts:
camera_initialization_error: The custom camera initialization script failed with the following error message: mkdir: cannot create directory ‘’: No such file or directory
Traceback (most recent call last):
File "/home/pi/scripts/gopro-py-api/gopro-take-snapshot.py", line 32, in 
urlFileName=cam.take_photo() # default : delay = 1sec
File "/home/pi/scripts/gopro-py-api/goprocam/GoProCamera.py", line 453, in take_photo
if "HERO5 Black" in self.infoCamera(constants.Camera.Name) or "HERO6" in self.infoCamera(constants.Camera.Name):
TypeError: argument of type 'NoneType' is not iterable
The snapshot was not found in the expected directory: ''.

LMITAB, 2019-03-13:
/home/pi/scripts/gopro-take-snapshot.sh
Creating directory:
mkdir: cannot create directory ‘’: No such file or directory
Waking up...
Waking up...
Waking up...
Error, camera not defined.
Traceback (most recent call last):
  File "/home/pi/scripts/gopro-py-api/gopro-take-snapshot.py", line 32, in <module>
    urlFileName=cam.take_photo() # default : delay = 1sec
  File "/home/pi/scripts/gopro-py-api/goprocam/GoProCamera.py", line 453, in take_photo
    if "HERO5 Black" in self.infoCamera(constants.Camera.Name) or "HERO6" in self.infoCamera(constants.Camera.Name):
TypeError: argument of type 'NoneType' is not iterable
The snapshot was not found in the expected directory: ''.

FormerLurker, 2019-03-13:
FYI, I just got a hero 3 to play with thanks to a generous friend.  I will hopefully have time to poke around in the next few weeks.  Currently I'm finishing up some stuff for the next release of Octolapse, which is exciting.

yet-another-average-joe, 2019-03-13:
Once again, this means that the GoPro and gopro-py-api do not communicate. These messages are issued by gopro-py-api. You MUST solve this problem. You have to test according to the tutorial  video by the author of gopro-py-api
Camera ON
WiFi ON
Raspi WiFi set to GoPro access point
GoPro set to GoPro App mode

Can you communicate using a smartphone with GoPro App ?

If the GoPro is set to be used with the remote, it will NOT work as acces point. See GoPro WiFi configuration.

LMITAB, 2019-03-13:
Sweden...

yet-another-average-joe, 2019-03-13:
@FormerLurker : fine ! About Python : I read that 2.7 is end of life, and it seams to me that Foosel explained in a recent video that she was working on a new OctoPrint rewritten to work with v 3.x (to be verified... She speaks too fast for me !). This could make gopro-py-api installation as easy as gphoto2 if OctoPi and OcloPrint come with 3.6+ in a future release. At this time it is 3.5, and the OctoPi developper wanted to stick with 3.5, because it comes with Jessie (I asked, it was his answer).

@LMITAB : I'm from France (currently GMT+1, we will change within 2 weeks for GMT+2). Are we on the same Time Zone ? I think so. I'm generally coding, CADing, routing PCBs or editing videos between 11:00PM and 3:00AM (working days) and from 2:00PM and 3-4AM (sunday and monday). I start tonight a new video about OctoLapse and GoPros, with the last scripts, and connectivity tests. It will be easier for you (and others).

I realize that KonradIT made no videos about H6 !!! But you could give it a try with other models tutorials.

There's a potential problem : in some functions the MAC ID must be given as a parameter. For example the init() function.https://github.com/KonradIT/gopro-py-api/blob/master/docs/docs.md
It's not easy to me because, as I already said, I'm a Python beginner (a few weeks). And to be honest, I hate this language (C++ lover).

I just found this (the exact same problem as yours, but no really usefull answer):

https://github.com/KonradIT/gopro-py-api/issues/32
https://github.com/KonradIT/gopro-py-api/issues/36

KonradIT did NOT test this one. But it is in the compatibility list. I know no H6 owner, except you...

LMITAB, 2019-03-13:
Ii found the error... i must have ths Gopro in 2.4 Ghz not 5 Ghz

pi@octopi:~ $ /home/pi/scripts/gopro-take-snapshot.sh
Creating directory:
mkdir: cannot create directory ‘’: No such file or directory
HERO6 Black
HD6.01.02.01.00
Camera successfully connected!
{}

{}

Its stays like the above and ? waits maybe for next call (i dont know)

LMITAB, 2019-03-13:
if i try execute the script once more in octoprint it wont respond...and gives me error.

But if i just waits for the error and then excute the script once more it works again...????

There were 1 errors while applying custom camera settings/scripts:
camera_initialization_error: The custom camera initialization script failed with the following error message: mkdir: cannot create directory ‘’: No such file or directory
Traceback (most recent call last):
File "/home/pi/scripts/gopro-py-api/gopro-take-snapshot.py", line 32, in 
urlFileName=cam.take_photo() # default : delay = 1sec
File "/home/pi/scripts/gopro-py-api/goprocam/GoProCamera.py", line 453, in take_photo
if "HERO5 Black" in self.infoCamera(constants.Camera.Name) or "HERO6" in self.infoCamera(constants.Camera.Name):
TypeError: argument of type 'NoneType' is not iterable
The snapshot was not found in the expected directory: ''.

[image]

LMITAB, 2019-03-13:
i got some more progress...
I started a print with octolapse in test mode and the gorpro take a snap then take 1 more snap right after the first one...then this error comes up...
and when the layer is ready it takes "2" more snaps... and keep on doing this

I found out that i had the script call "/home/pi/scripts/gopro-take-snapshot.sh" in 2 places...in octolapse gui

Octolapse - Camera Error
snapshot_script_error: The snapshot script failed with the following error message: - The snapshot script timed out in 5.0 seconds.

LMITAB, 2019-03-13:
i also tested in real mode and i get the timeouts all the time... and also this error ...

Octolapse Rendering Failed
insufficient-images: No snapshots were found for the 'GoPro' camera profile.

The script does fire the gopro and i takes snapshot thats in the camera...

yet-another-average-joe, 2019-03-14:
All the errors you report are neither OctoLapse related, neither script related. They are gopro-py-api + GoPro related.

Once again you must first setup and test gopro communications from a Python console, as explained by KonradIT in his videos. The same way FormerLurker explains a DSLR has to be tested when using gphoto2. 

Do this https://vimeo.com/209181246 ; it is for H3, but it shouyld work. If it does not work, you will have to try with the other tutorials or ask on gopro-py-api git page. 

If you are using the onboard WiFi adapter (and I feel like you do), you will NOT get any results with OctoLapse. It does NOT work. Even with timout cranked up to 10s. , You MUST use a WiFi dongle AND deactivate onboard WiFi adapter, or it will timout every time. This is a known issue, confirmed by the author of gopro-py-api : https://github.com/KonradIT/gopro-py-api/issues/88 . It is hardware related., and it can be seen in the KonradIT videos (pics take a while to download : 11s for a photo !). 

Use a WiFi dongle with a chipset from this list :

https://elinux.org/RPi_USB_Wi-Fi_Adapters

Deactivate onboard WiFi adapter :

in /boot/config.txt, add this :

dtoverlay=pi3-disable-wifi

if you don't use Bluetooth, add also this, it does not hurt :

dtoverlay=pi3-disable-bt

(and keep an Ethernet connection !)

Then, follow these steps in order to activate the WiFi dongle :

https://www.electronicshub.org/setup-wifi-raspberry-pi-2-using-usb-dongle/

And you will get a good WiFi connection with the GoPro. Without a WiFi dongle, it will fail 100%. It will takes 10+ seconds to take and download a snapshot.

You must also have a good PSU. I had many problems until I used a 5 amps 5V one, connected directly on the GPIO pins. I even slightly overvolted at 5.2V instead of 5.0V, in order to overcome the voltage drop in the wires. All 3 amps mains wall adapters I tried where just crap ! But your PSU seems to be OK, as there is no alarm icon in your screenshot. Are you using the last version 1.3.10 ?

A torture test with 3 simultaneous timelapses : DSLR + GoPro + webcam : https://www.youtube.com/watch?v=yA1-0jj4F8A

FormerLurker, 2019-03-14:
@yet-another-average-joe, I'm aware of the python 3 compatibility in an upcoming release of OctoPrint, and even understand how to implement it.  I'm not worried about this.  I'm going to be making Octolapse compatible with both 2.7 and 3+ soon.  It is mostly already, but there are a few pieces that aren't, or that will operate differently in 3.

When I start on the GoPro plugin I will make it 3.6+ only, for obvious reasons.  It won't work until the next version of OctoPrint is released, but I doubt I'll beat Gina :)  She's on it!

FormerLurker, 2019-03-14:
@LMITAB, try increasing your camera timeout to from 5 seconds to 30 seconds for testing purposes.  it seems like there is a lot of variability in how long it takes to get an image from the gopro, and having a short timeout is not good for debugging.

LMITAB, 2019-03-14:
ok, will try it... im all over trying to get a usb wifi to work ... the one i had (old one) doesnt seam to work with rpi3 anymore...

Bus 001 Device 005: ID 07b8:8188 AboCom Systems Inc AboCom Systems Inc [WN2001 Prolink Wireless-N Nano Adapter]

Going to find a new one tomorrow...hopefully they have one thats comp with the rpi

yet-another-average-joe, 2019-03-14:
It should work. I just Googled the WN2001, it is a Realtek RTL8192U chip, and it is what I am currently using (noname dongle with this chip on my side).

LMITAB, 2019-03-14:
[image]

[89 more replies]

closed 2019-10-12 by FormerLurker