FormerLurker/Octolapse #310: Octoprint printer profile override not working

pandel, 2019-01-25:
Hi!

I am trying to use the current Octolapse version with my delta printer, but am having issues with printbed recognition.

Everytime I start a print job, Octolapse is telling me errors about not having found the printbed origin, blabla... AND those messages appear for every move command (it would be nice, if one could disable those nasty popups coming up every half of a second)!

My printbed is roughly 240mm in diameter with (0,0) in the middle. Even though my Octoprint printer definition contains the same values (in my case "round", 240mm, and a custom bounding box setup), I set up an override in Octolapse, just to be sure: -110 to 110 for X and Y, and 0 to 200 for Z (smaller sizes, to be on the safe side). But that doesn't interest Octolapse at all.

What am I doing wrong here?

The printer sets a G28 before everything starts and so Octolapse should be able to take a note of the printers home position and start from there. In combination with the values I dialed in manually, everything should be fine.

Regards,
Holger

FormerLurker, 2019-01-25:
What does the error message say exactly?  Pause the print to prevent new errors from popping up while you are copying the message.

pandel, 2019-01-25:
I will find out, perhaps this evening....

FormerLurker, 2019-01-25:
Thanks!  Im suspecting a conflict between octoprint and octolapse here.  I haven't finished support for circular beds yet, so a minor hack may be necessary until full support is added, hopefully to the next release.

pandel, 2019-01-28:
I am very sorry! As always, when you're trying get things done, things get messed up... I am trying to post an example message as soon as possible!

pandel, 2019-02-01:
Finally from the log file:

`2019-1-31 22:10:59,104 Position - Coordinates (X:-101.36300,Y:-23.27200,Z:0.30000,E:4.50500) are out of printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.`

These are my current profile settings (G90 and M82 are definitely the first two lines of GCODE in the file I get from Simplify3D):
_{
      "small_perimeter_speed": 2925.0, 
      "bridging_speed_multiplier": 110.0, 
      "perimeter_speed": 2925.0, 
      "outline_speed_multiplier": 50.0, 
      "first_layer_travel_speed": 4800.0, 
      "top_solid_infill_speed": 3120.0, 
      "top_solid_infill_speed_text": null, 
      "movement_speed": 4800.0, 
      "solid_infill_speed": 3120.0, 
      "ooze_shield_speed": 3822.0, 
      "guid": "ac021985-44a7-4e0b-8a1a-4345b40b28de", 
      "priming_height": 0.3, 
      "small_perimeter_speed_multiplier": null, 
      "has_been_saved_by_user": true, 
      "origin_z": null, 
      "abort_out_of_bounds": true, 
      "min_y": -116.0, 
      "min_z": 0.0, 
      "detract_speed": 3000.0, 
      "suppress_snapshot_command_always": true, 
      "default_firmware_retractions": false, 
      "speed_tolerance": 1.0, 
      "z_hop": 0.0, 
      "maximum_z_speed": null, 
      "infill_speed": 3900.0, 
      "small_perimeter_speed_text": null, 
      "max_z": 290.0, 
      "e_axis_default_mode": "require-explicit", 
      "external_perimeter_speed_multiplier": null, 
      "max_y": 116.0, 
      "gap_fill_speed": 3900.0, 
      "ooze_shield_speed_multiplier": 98.0, 
      "description": "", 
      "support_structure_speed_multiplier": 81.0, 
      "bridge_speed": 4290.0, 
      "first_layer_speed_multiplier": 55.0, 
      "solid_infill_speed_text": null, 
      "slicer_type": "simplify-3d", 
      "top_solid_infill_speed_multiplier": null, 
      "g90_influences_extruder": "false", 
      "external_perimeter_speed_text": null, 
      "solid_infill_speed_multiplier": 80.0, 
      "origin_y": null, 
      "origin_x": null, 
      "above_raft_speed_multiplier": 30.0, 
      "min_x": -116.0, 
      "print_speed": 3900.0, 
      "first_layer_speed": 2145.0, 
      "retract_speed": 3000.0, 
      "default_firmware_retractions_zhop": false, 
      "axis_speed_display_units": "mm-min", 
      "auto_position_detection_commands": "", 
      "prime_pillar_speed_multiplier": 99.0, 
      "name": "Anycubic Kossel LP - PLA", 
      "retract_length": 5.1, 
      "snapshot_command": "snap", 
      "override_octoprint_print_volume": true, 
      "support_speed": 3159.0, 
      "external_perimeter_speed": 1950.0, 
      "skirt_brim_speed": 3900.0, 
      "first_layer_speed_text": null, 
      "z_hop_speed": 1002.0, 
      "num_slow_layers": 1, 
      "max_x": 116.0, 
      "prime_pillar_speed": 3861.0, 
      "printer_position_confirmation_tolerance": 0.001, 
      "units_default": "millimeters", 
      "xyz_axes_default_mode": "require-explicit", 
      "auto_detect_position": true, 
      "above_raft_speed": 1170.0
    },_

FormerLurker, 2019-02-04:
@pandel, I actually have the settings extraction direct from gcode for Simplify, Slic3rPE and Cura working already :)  It will be a bit before it's ready because there were some really extensive modifications.  If you feel like beta testing at some point let me know and I'll send you a link once I've got all of the kinks worked out.

I'm not sure what is going on here, I'm sorry to say.  Can you post the entire settings.json file for me?  This is likely some unknown bug that is related to either other settings, your printer responses to M114 commands, or something I haven't even thought of before.   90% of the time these errors are due to incorrect printer volume, but your printer profile settings look correct to me.

pandel, 2019-02-04:
Hi!

No problem ;-) As I am still learning to use my delta, I will have many more chances to try to get a good timelapse video. In the meantime I'll be printing the boring parts ;-) Here is my settings file:

[settings.json.txt](https://github.com/FormerLurker/Octolapse/files/2828032/settings.json.txt)

FormerLurker, 2019-10-12:
OK, so it's been way too long since I got back to you.  Sorry about that.  The issue list had a huge backlog, and I'm trying to remedy that now.

You'll be pleased to know that I've added full support for delta printers to V0.4 alpha of Octolapse.  Let me know if you'd like to try it out.  In the meanwhile I'm going to close this since I think the root problem has been resolved (lack of good support for circular beds).  If you have any other questions or comments I can easily reopen this issue, or feel free to post a new one!

Again, thanks for your post!!

closed 2019-10-12 by FormerLurker