FormerLurker/Octolapse #311: [Request]

notoriousbad, 2019-01-25:
Language/Translation issue.

Under "Extruder Retraction Settings", the wording is probably wrong: "Detraction settings". It should be Deretraction settings". "Detraction" has a whole different meaning with even negative connotation.

Screenshot OctoLapse printer settings:
[image]

Screenshot Slic3r PE extruder settings: 
[image]

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

FormerLurker, 2019-01-25:
Lol, I cant claim any language barrier issue here.  I actually noticed this while working on some enhancements (settings extraction from gcode).  That mistake was copy/pasted everywhere, so ill make sure it is fixed for the next release.  Thanks for pointing it out so clearly!

FormerLurker, 2019-10-12:
Wow, was this difficult to solve.  I decided not only to change this in the verbiage you showed, but also within the code itself.  It has actually been fixed for a while, but I've been way behind on issue maintenance.  I'm going to go ahead and mark this closed.  Thanks for the report!

closed 2019-10-12 by FormerLurker