FormerLurker/Octolapse #312: invert axis

alopicaso, 2019-01-27:
#### Version of Octolapse

Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: Version 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Change printer min volume
2. use g90/91 and m82/83
3. look into firmware and find out that my printer xyz is invert, and it only works with that setting. 

#### What should have happened?
print normally and take snapshot

#### What happened instead?
1.Print on edge of the bed, hitting and smashing
2. If I don't change the min volume I would get Position - Coordinates (X:110.31200,Y:-84.26700,Z:-114.27000,E:25.03600) are out of the printer area! 

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi 
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: FlashForge Creator Pro 2016
Printer Firmware Version: sailfish 7.7

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/alopicaso/35635244f4d89f1db472e101e56b0f69

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/alopicaso/09d5d77a6e73b88bf115ebaf123b5a98#file-gistfile1-txt

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Is there anyway to disable the position checking function? I really love this plugin, and I am desperately looking for a solution for almost 1 week before I would bother your precious time, but I found no luck. Thank you for helping me.

alopicaso, 2019-01-27:
This is my gcode
https://gist.github.com/alopicaso/fa4ec80a3d15faea3350a600674f8d18
Im using sailfish firmware 7.7 and my setting is something like this https://www.photobox.co.uk/my/photo/full?photo_id=501617197021
When I use m114 in octoprint to detect the current position, it shows xyz in negative value like x-100,y-100,z-145
I am not really sure if the invert axis setting causing this negative value or not, but that the only thing I found so far. Thanks

FormerLurker, 2019-01-28:
That photobox link didn't work.  You can just download your settings from OctoPrint Settings->Octolapse.  There is a download button in there.  That will help!

Also, I looked at your gcode, and it doesn't seem unusual at all.  It must just be the position reporting!  I wonder why that is the case.  I may send you a list of gcode commands to run for testing after I get settings.json depending on what I find.

alopicaso, 2019-01-29:
Sorry this is the picture I was referring to, hope it helps. I will upload the Octolapse setting as well when I get home tonight. Thanks
[image]

FormerLurker, 2019-01-29:
I see.  Thanks!  However, what I really need is your printer profile settings from Octolapse.  Look in OctoPrint settings (wrench/spanner icon)->Octolapse.  There is a download button in there for settings.json.  If you can post that for me that would be great!

I'm looking to see if I can get a virtual printer to simulate the behavior of your printer.  If so I should be able to get it working pretty fast unless there is some unforeseen issue!

alopicaso, 2019-01-29:
My setting.json Thanks! 
https://gist.github.com/alopicaso/2a02a9f70dde7e6cc41ef3e95f31bbea

FormerLurker, 2019-01-29:
It looks like your Min and Max values for your print volume are the same, which means your bed has no volume at all as far as Octolapse is Concerned.  For Example if MinX=-100 and MaxX = 100, your bed is 200mm wide, and '''G1 X=-100.0''' would bring you to one extent and '''G1 X=100`` would bring it to the opposite extent.

What you need to find is the minimum and maximum X, Y, and Z coordinates that your printer can handle without hitting an endstop.  Just remember that MaxX - MinX = printer width.  The same is true for Y and Z.

Octolapse needs to understand the coordinate system for 1 main reason:  It needs to know where to move to stabilize.  If you tell octolapse to stabilize at the back left, it will go to position X=MinX and  y=MaxY.  The front right would by X=MaxX and Y=MinY

If octolapse detects movement outside of this area, it decides it can't stabilize because the printer volume/coordinates must be incorrect.  If Octolapse attempts to stabilize when the volume/corrdinates are incorrect it might hit an endstop or something worse.  In this case it decides to tell you want coordinates were out of bounds to make it easier to know what values to adjust.

Clear as mud?

alopicaso, 2019-01-30:
Yes, it makes perfect sense. I will try to mess with the min and max value when I get home. I will let you know. Thanks

alopicaso, 2019-01-30:
This is my **Home** position using jog mode to find it Home: X:0.04 Y:0.03 Z:-150.03 E:0.00
This is my **Max** position using jog mode to find it Recv: ok X:-230.00 Y:-150.00 Z:28.32 E:0.00
I tried to change the print volume in octolapse and it still give me position error because before the print start X and Y axis tried to move out of the print area and smashed the printer.
I think I have to match my slicer print volume profile with octolapse? but I can't enter negative value in my slicer.

[image]

FormerLurker, 2019-06-24:
Hey, I'm sorry I haven't gotten back to this issue in so long, but I had a lot going on, and lots of code needed to be added to support inverted axes and circular beds.  However, I've finally gotten around to implementing inverted axis support.  Unfortunately I have no good way to test it!  There's not much info out there on what inverting an axis actually does.  Some firmware can invert and make this invisible to the user, and some other folks are running with stepper motor polarity reversed without firmware modifications, so I'm pretty confused.

I've created a post on the [octoprint forum](https://community.octoprint.org/t/inverted-axis-questions/10339), and am hoping to have a few questions answered before asking you to test the new version.  I hope to run some tests on the virtual printer first.

Ok, and now for the 'whoops' news.  After reviewing your original issue, I see that Octolapse is reporting that the out of bounds Y position is -160.85, so maybe changing Y-Depth-Min to -161.0 will solve your problem without any of this inversion stuff?

Lastly, if you have time at some point can run the following commands (carefully since I'm not sure how your printer will respond) to see if your extruder moves left or right?  I've annotated each gcode, and they should be executed one at a time.  If anything odd happens at any step, stop the scripts and discontinue the test (safety first!):

```
G28; Home XYZ
G90; Absolute XYZ Mode
G0 Z-145 ; Lift about 5mm
G0 X-100 y-100 ; move to -100,-100
G91; Set XYZ to relative Mode
G0 X20 ; Not sure what direction this will move
G0 X-20 ; Not sure what direction this will move, but it should return to -100, -100
```
Anyway, stay tuned, and sorry for the long long delay!

FormerLurker, 2019-10-12:
So, circular bed support has been added to Octolapse V0.4 (the alpha)!  Let me know if you'd like to try it out.  I don't believe there is a default profile for your printer at the moment (I added new make/model based printer configuration), but we could remedy that.  Anyway, I'm going to mark this issue as waiting for response and will give you some time to respond.  Thanks for positing your issue, just sorry it took so long to resolve :)

FormerLurker, 2019-11-09:
@alopicaso, since this issue has been resolved by adding circular bed support, I'm going to go ahead and close it down.  Thanks for posting!

closed 2019-11-09 by FormerLurker