FormerLurker/Octolapse #313: Octolapse causing prints to pause mid print.

benomoreno3, 2019-01-31:
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/2815642/plugin_octolapse.log)
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/2815643/octoprint.log)

#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Downloaded octolapse plugin and inserted all settings
2. started new print with octolapse enabled
3. print stopped without me doing anything

#### What should have happened?
print should have continued and finished normally with the timelapse done

#### What happened instead?
print paused a few layers in without stopping the heating

#### Operating System running OctoPrint and Octolapse

OS Name: MacOS
Os Version: 10.14.2

#### Printer model & used firmware incl. version

Printer Model: Prusa Mk3
Printer Firmware Version: latest

#### Browser and version of browser, operating system running browser

Browser: safari
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: different gcode files tried, all worked with octolapse disabled

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-10-12:
I've been going through old tickets today and noticed I never responded to this!!!!  I'm SO sorry.  I did just take a look at your log, and am having a hard time figuring out where the problem is occurring.

There have been LOTS of changes to Octolapse over the last 9 months, so if you are still having this issue, it might be time to try the alpha version of Octolapse V0.4.  I'll mark this issue 'Waiting for Response' in the meanwhile.  Please let me know how you'd like to proceed.

Again. SO SORRY I didn't respond, it must have fallen through the cracks somehow, but I'd like to remedy that if possible.

FormerLurker, 2019-11-09:
@benomoreno3, it's been several weeks since this thread was active, so I'm going to go ahead and close it down.  Thank you for posting!  Again, I'm SO sorry I missed your original issue somehow.  Feel free to post again any time.  I'm getting better at managing tickets (hopefully) :)

closed 2019-11-09 by FormerLurker