FormerLurker/Octolapse #315: DSLR Image Capture With No Download problem

kyoungin90, 2019-02-04:
Version of Octolapse
0.3.4rc1.dev3

Version of OctoPrint
1.3.10

First, Thank you for your excellent program, octolapse.

I used Octolapse with DSLR very well.

Now, I'm trying to use 'DSLR Image Capture With No Download' function.

Therefore, I followed guide exactly same way. 

When I finished my work, my camera took a picture. 
However, the picture was not saved on my camera SDcard and pi.

When I did step3 - test camera script, I couldn't see " TODO: FILL THIS IN" and "Todo:  Fill this out" message.

How can fix this problem??

thank you for reading.

FormerLurker, 2019-02-04:
Hey, sorry that guide isn't 100% finished yet.  That 'TODO' was for me :)

Can you post your output from trying to run this in the command prompt:

```
/home/pi/scripts/initialize-camera-save-to-sd.sh
```

and also:

```
/home/pi/scripts/trigger-snapshot.sh
```

If I have some time today I'll update that guide to reflect actual output from those commands.

kyoungin90, 2019-02-04:
Wow! what was changed in your script?!
It was working very well :)

My camera was Canon 100D.
Thank you for your effort!

FormerLurker, 2019-02-04:
You said that it 'WAS' working very well.  Do you mean that images are now being saved to your camera?  I'm hoping your problem is solved :)

kyoungin90, 2019-02-04:
> You said that it 'WAS' working very well. Do you mean that images are now being saved to your camera? I'm hoping your problem is solved :)

I changed now was to is :) 
Thank you!

kazzthehobbyist, 2019-04-28:
Hallo.
I am testing this option with my Canon Eos 1200D. But It will not safe the pictures on my SD card. The camera snaps every time it needs to, but the picture is simply not there. I have tried to add the sleep command and set it to 10, just as a test, but still not there.
This is also the case then taking a test photo in Putty.

The camera is hooked up to a Rasberry pi 3b+ via usb.

The camera safes pictures i take manually

FormerLurker, 2019-04-28:
@kazzthehobbyist, you probably should't be adding any sleep commands, since it will just slow down the process.

What version of gphoto2 are you using?  [The documentation](http://gphoto.org/doc/remote/) specifies libgphoto2 2.5.16 or newer for your camera.

Did you run this script (mentioned in the guide, sorry I have to check):  /home/pi/scripts/initialize-camera-save-to-sd.sh

Since this is most likely related to gphoto2 and your particular camera, I recommend testing all commands in putty, running gphoto2 manually in a terminal (putty), and testing until you can get an image to save to your SD card.  Based on this you will probably need to make a few changes to the script I mentioned above.

FYI, by default Gphoto2 will try to save photos within RAM (not SD) for speed, which means no image on the SD.  I suspect that something is wrong with the capture target setting (see initialize-camera-save-to-sd.sh for this command), which should be telling your camera to save to SD.  GPhoto2 has some options to print out the current capture target, and to list the available options.  [See the command line documentation here](http://gphoto.org/doc/manual/ref-gphoto2-cli.html).  I tried to test some of these commands for you, but my camera PSU is missing!  I'm charging some batteries and will tag this on in a few hours.

kazzthehobbyist, 2019-04-28:
@FormerLurker My version of gphoto is 2.5.11-1 And I should have followed the guide down to the letter, but I check again and give you a heads up

kazzthehobbyist, 2019-04-28:
@FormerLurker I have checked the guide and followed every step there is and included the script ans you asked. Its the same result.
So that means that I need to look into the script?

FormerLurker, 2019-04-28:
@kazzthehobbyist, following the guide to the letter may have been the problem, lol!  It's beta and it seems the instructions need to vary per camera.

Try running this command:
```
gphoto2 --get-config capturetarget
```
I've updated [the guide](https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download#note-about-the-capturetarget-setting), so you might want to check that out.

FormerLurker, 2019-04-28:
@kazzthehobbyist, FYI, I also changed '''capturetarget=2''' to ```capturetarget=1```, though the former option seemed to work on many cameras.  Not sure why I used 2 in the first place honestly.  It was probably very late.

kazzthehobbyist, 2019-04-28:
@FormerLurker Thanks a lot, I will look into it tommorrow morning. I will give you a heads up as soon as I have tried it.

kazzthehobbyist, 2019-04-29:
@FormerLurker I have tried the new guide and the pictures know is stored on my SD card, but the TODO commands dont show up in the consol then I follow the guide. And the pictures is not downloaded for the render.

But hey thats 1 problem fixed and a bonus one that needs to be looked at. I count that as a suscess xD

EJtechandDiy, 2019-10-05:
> `capturetarget=2` to `capturetarget=1`

On the guide you still have this on 2 and not on 1, after not getting any pictures on my sd card I changed this to one and the now the pictures are been save on the sd card.

Im working with a nikon D5300

FormerLurker, 2019-10-05:
@EJtechandDiy, I corrected this on the deferred image download tutorial but not the no download tutorial.  I have corrected this.  Please take a look at the updated instructions and let me know if it looks good now.  I added some information regarding how to determine the appropriate capture target as well.

EJtechandDiy, 2019-10-05:
> @EJtechandDiy, I corrected this on the deferred image download tutorial but not the no download tutorial. I have corrected this. Please take a look at the updated instructions and let me know if it [...]

You are a legend, thank you so much for all of this, it looks really good now.

I really don't like petreon but I think I gonna have to make one to support people like you.

FormerLurker, 2019-10-05:
@EJtechandDyi, you are too kind!  I appreciate any and all support, and wish you the best of luck with all your endeavors!  Thanks for letting me know about the problems in the wiki, as this improves things for everyone.

FormerLurker, 2019-10-12:
OK, it looks to me like this issue is resolved.  If anyone would like me to reopen it, please let me know!  Thank you to everyone who contributed!

closed 2019-10-12 by FormerLurker