FormerLurker/Octolapse #316: [Request] octolapse camera moving function?!

kyoungin90, 2019-02-04:
Hello!

I really pleasantly use octolapse these days.

And I want some other function when I am making timelapse with DSLR.

My suggestion is that if I added one stepper and made DSLR roller which can make DSLR move horizontally little by little.
Then I control a stepper with octolapse, I will make more cool timelapse.

[image]

FormerLurker, 2019-02-04:
Are you planning to control this via gcode, or some other way?  If you are planning to use gcode, check out the camera profile settings and look at the camera type 'Gcode Camera'.  With this you can send gcode before any snapshots are taken.

If you plan to use a script to control this, it will depend on how exactly you have your hardware configured.  If you let me know a bit more about your setup, I can probably help.  Fyi, some other folks have gotten this to work already.  I think three is even a project in github for this.  Let me know if you need some links and I'll google them up for you.

FormerLurker, 2019-10-12:
Hi again @kyoungin90!  This issue is a bit old, so I'm going to go ahead and close it.  Let me know if you'd like it reopened for any reason, and thank you for posting!

closed 2019-10-12 by FormerLurker