FormerLurker/Octolapse #317: button to take picture

michaelmart77, 2019-02-05:
I'm new to octoprint. I found this plugin nice. I'm asking for a feature. I'm trying to have a button to take a picture (execute shell script) not only while printing but also anytime I want? Is it hard?

michaelmart77, 2019-02-05:
Thank you. not webcam. it's a usb cam. it captures a picture by executing a python or shell script. I just want to let it take a picture (execute the script) when clicking the button (probably in octolapse). The purpose is that I want to test the pictures manually before starting. The picture is saved somewhere.

FormerLurker, 2019-02-05:
Hmm.. I have a workaround for you for the time being:  Create a new camera profile, mark it as 'disabled' so it won't interfere with your timelapses.  Then set the 'Camera Initialization Script' path to your sh script for grabbing your pictures.  There is a 'test' button next to that script for various reasons (it was the easiest one to do) which simply executes the script it's pointed at.  Be sure to give the script execute permissions, and if you use any commands that require sudo there is more work to be done (either embedding the password in the file or modifying sudoers so that the commands can run without a password).

It might not be particularly convenient to open the camera settings to nab a picture, but it will work.  There are some other Octoprint plugins that can execute scripts (gcode system commands comes to mind), but I'm not sure if any allow you to attach a script to a button yet.  Octolapse might actually be the only one, but if so It's only a matter of time :)

FormerLurker, 2019-10-12:
@michaelmart77, I've been thinking about this feature request, and I don't think it is a good fit with Octolapse ATM, especially with the direction the new version is taking.  However, there ARE some plugins out there that can do what you are wanting by adding buttons to Octoprint that can run shell scripts.  Please let me know if you would like a link to one of these.

I'm going to go ahead and close this issue, but please let me know if you'd like it reopend for further consideration.  Also, thank you so much for posting and suggesting improvements!  I'm always very glad to hear your ideas!

closed 2019-10-12 by FormerLurker