FormerLurker/Octolapse #318: [REQUEST]

yet-another-average-joe, 2019-02-05:
Hello,
and thanks for OctoLapse.

I've been experimeting with OctoLapse. DSLR and GoPro (using GoPro-Py-Pi with a WiFi connected Hero3+BE).

I take stills with the GoPro through 2 scripts : a bash calling a Python script. Works fine.

But timelapses and therefore scripts are deactivated if there is no declared camera. I see at least 2 reasons to allow timalapses to be allowed with no camera, but launching the 5 scripts :

- a DSLR shooting raw pics (real men shoot raws ;) )
- a GoPro (but GoPro-Py-Pi has functions to download pics over WiFi)

FormerLurker, 2019-02-05:
Hmm, if you're using Octolapse to acquire the image via a script, it should work.  You need a camera profile in order to do this, right?  I've got examples of scripts that allow you to shoot with a DSLR and not download any images to the pi, or to wait to download until all of the images are acquired before rendering, or you can disable rendering if you don't want a timelapse to be produced automatically (you might want to render it manually?)

Am I missing something?

yet-another-average-joe, 2019-02-05:
I just set the default webcam as camera profile, and use a script to shoot with the GoPro. Then, the rendered timelapse is not used. Until now, it's been with a virtual printer and a RasPi alone. But I tested with a Nikon D5100 on a real printer. Everything seems to run smoothly. The only issue is the GoPro Camera object : it has to be created for each and evry pic, it takes time and I had to set timeout at 7000ms. Don't know if it can be kept persistant from one call to another (I really doubt !)

The bash script called from OctoLapse :

```
#!/bin/bash
python3.6 /home/pi/gopro-py-api/test.py
```

The Python script :

```
#!/usr/bin/ python3.6
from goprocam import GoProCamera
goproCamera = GoProCamera.GoPro()
goproCamera.take_photo()
```

"goproCamera = GoProCamera.GoPro()" takes a while...

Remember : this lines of code are my very first ones (with bash and Python)

yet-another-average-joe, 2019-02-05:
Bad click, I closed !

FormerLurker, 2019-10-12:
Been doing some issue maintenance (as you probably have noticed), and I believe we've taken care of this, no?  Going to close this issue, but will reopen if you'd like me to.  THANK YOU for posting!

closed 2019-10-12 by FormerLurker