FormerLurker/Octolapse #319: [Request] Snapshots without movement

loudsound-org, 2019-02-05:
#### If this is a feature request describe it here

As far as I can tell, there's no option to do this.  I like the Animated timelapse, but with an unrestricted X, so the printhead randomly moves around, but the Y axis stays stabilized.  However, this causes the printhead to have to move for the snapshots, both slowing down the print, and creating some artifacts.  Is it possible to opportunistically take snapshops at any point where the Y axis happens to be at 50% (or some user configured value)?  That way the bed appears stable in the timelapse, but the printhead is never moved just to take a snapshot.  Obviously some layers may never get a snapshop if the printhead never passes through that Y value (although it would be even cooler if there were a way to recognize this, and then you could have portions of the timelapse where the Y axis HAS moved, for that portion of the print).  

Another option might be some sort of smoothed stabilization, so you can have Y axis changes, but it only takes them when it's changed in a small amount and in a certain direction, so you don't have lots of jumping back and forth.  Not easy to implement I'm guessing, but would look nice.

Bottom line, I love the idea of octolapse, but I don't like having my print slowed down, and even more, I don't like some of the artifacts I'm getting when using it.  I'm trying some different options but not sure how much they'll help.  And I figured out why my prints were jumping really far...I have high quality feature detection on, but there was no infill or inner walls in large sections of the print, so it would just not take a snapshot for hundreds of layers!

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-02-05:
So you'll be pleased to know that you can actually already turn off stabilization for either X or Y!  Just edit your stabilization profile and set one axis to 'disabled' like so:
[image]

That should solve that issue.  Regarding the time added to prints, it can be very small (less than 1 second per layer depending, but can be several seconds in a worst case).  There are lots of things you can do to reduce the time, and I'm working on even more as we speak.  You might be interested to know that I've reduced the time required to take a snapshot with no stabilizations enabled (no x/y axis movements at all) to something that's hardly noticeable.  For now I recommend a 'center' stabilization.  If you REALLY want a small amount of time look into 'Position Restrictions' in the snapshot settings and keep the snapshots close to the restriction.  There is a link to a help page in the plugin next to the setting in case you want to learn more.  Also, try to minimize your camera delay.  250MS is the default, but I've been able to get my camera to work at between 75MS and 100MS at 15FPS depending on the amount of light.  You may have to increase your frame rate and your light.  The formula for the optimal delay (if all conditions are perfect) is 1/(frames per second)*1000.  Also, you might want to tick on the 'Snapshot Time Calculation' in the main settings if you want to know how much time is being added by Octolapse.  
[image]

This is disabled by default because some printers don't like all of the extra M400 commands this requires (My Prusa for one when linear advance is enabled).  If you encounter any lockups when using that setting, disable it and let me know what printer you have so I can add it to the list :)  FYI, sending an M114 command via the terminal can sometimes fix the issue.  Otherwise there is a 10 minute timeout :(

Regarding artifacts, this can happen.  Some printers/materials are worse than others.  My Prusa does really well with PLA and ABS, not so well with PETG (though it strings a lot anyway).  I've been able to get prints with virtually no artifacts whatsoever, though some limitations apply (missing frames in some cases).  Look into the 'high quality mode' profile.  You can prevent snapshots over things like bridges and exterior perimeters.  Currently the weak point with this method is the 'minimum layer time' slicer setting, which reduces all speeds by some percentage and throws off my calculations, leading to missed layers.  I'm working on this issue and may have a solution for most cases.  Time will tell.  Other than that, printing outside perimeters last may give you better results.

Also, regarding quality, there's a known issue for some slicers when using absolute extrusion and another for printers where G92 command affects the extruder state.  These bugs are fixed I believe, but the release isn't ready.  If you get noticeable pimples, but have your retraction/deretraction speed and length entered properly, either one of those bugs could be the cause.  There is also a (somewhat outdated) troubleshooting guide in the wiki that goes deep into quality.  Take a look if you haven't seen it yet.  At some point I'll do a video on the subject.

You might be interested to know there are some BIG changes coming to Octolapse.  I have been working almost exclusively on reducing artifacts, increasing performance (decreasing CPU and IO/reducing lines of gcode sent), and simplifying setup.  I'm adding some new stabilizations that pre-process the gcode, which makes it possible to search for the (hopefully) best places to take a snapshot for maximum quality and/or to minimize time.  I've already got one working that would probably meet your needs where you tell Octolapse to lock on to the Min or Max X and/or Y position of your print and take a snapshot there with practically no noticeable pause (depending on your camera delay and capture speed, of course).  I call this a quasi-stabilization since it's not perfectly still the whole time.  However, I'm super excited about this feature because it could be turned on all of the time with (hopefully) no effect on quality and virtually no time cost (in the seconds, not minutes).

At some point here I'll start releasing development builds for testing.  Check back for news.  Let me know if you have any other specific questions regarding quality or print time.  As you can tell I love going into detail and I just got off of work :)

FormerLurker, 2019-10-12:
So, the next version of octolapse has some greatly enhanced features for preventing travel during snapshots.  The delay has been minimized as much as possible by placing post processing on a background thread as well as a few other tricks.  I also noticed that increasing the camera's framerate also decreases the delay, since mjpgstreamer will wait for the next frame to become available before delivering a snapshot.  If you want to test the new version out, let me know and I will deliver instructions.

Because of the above, I'm considering this thread resolved and will close it now.  Please let me know if you'd like this to be reopend.  Thanks!

closed 2019-10-12 by FormerLurker