FormerLurker/Octolapse #320: [REQUEST] Test Script button for all 5 scripts

yet-another-average-joe, 2019-02-07:
Hello,

a "Test Script" button for each and every camera script line would be great ! There's only one, for Before Print Start Script. Therefore, testing scripts from within OctoLapse need lots of copy/paste and right clicks !

FormerLurker, 2019-02-07:
Yes, I want to get those features implemented for sure!  It's just an effort thing at this point.  I just need to figure out the best way to indicate success and to show the results (especially in the case of the snapshot acquire script) without further complicating the scripts.  I have some ideas and will mark categorize this request.

yet-another-average-joe, 2019-02-07:
Until now, the button is great : it returns the error of the invoqued script (at least my errors ;) !

FormerLurker, 2020-02-26:
This feature has been implemented and is available since v0.4.0rc1.dev3 (currently dev4, soon to be dev5).  You can install it from the plugin manager with the following url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

If you have any issues, please report them [here.](https://github.com/FormerLurker/Octolapse/issues/455)

FYI, the tests aren't bulletproof, since I have no way to know what the goal of each script is, but I tried to simulate a real world test as much as possible.  FYI, there are 7 test script buttons, since I've added a few more scripts :)

Closed at long last!

closed 2020-02-26 by FormerLurker