FormerLurker/Octolapse #321: can't install octolapse

DrMarkyMarc, 2019-02-08:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 1.3.9+

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Try to install octolapse
2. via octopi
3. via octoprint
4. updated and upgraded everything I could 
5. googling and installing all the pillow library recomanded to make this work (no succes)

#### What should have happened?
install octolapse. 

#### What happened instead?
Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip...
/home/pi/oprint/bin/python2 -m pip install https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip --no-cache-dir
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
Downloading https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
Collecting pillow (from Octolapse==0.3.4)
Downloading https://files.pythonhosted.org/packages/3c/7e/443be24431324bd34d22dd9d11cc845d995bcd3b500676bcf23142756975/Pillow-5.4.1.tar.gz (16.0MB)
Requirement already satisfied: sarge in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4) (0.1.5.post0)
Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4) (1.11.0)
Requirement already satisfied: OctoPrint>1.3.8 in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4) (1.3.10)
Requirement already satisfied: psutil in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4) (5.4.8)
Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.1)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.8.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2018.2.21)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.8.3)
Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (4.5.3)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.10)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.2.11)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.3.5)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.9)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10)
Requirement already satisfied: markdown<2.7,>=2.6.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.6.11)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.4)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.7.19)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.8.3)
Requirement already satisfied: netifaces<0.11,>=0.10.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.7)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.9)
Requirement already satisfied: rsa<3.3,>=3.2.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.2.3)
Requirement already satisfied: pkginfo<1.5,>=1.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.4.2)
Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.18.4)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.4.2)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (6.2)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.6.5)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (5.2.1)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.2)
Requirement already satisfied: future<0.16,>=0.15.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.15.2)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.3)
Requirement already satisfied: websocket-client<0.41,>=0.40.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.40.0)
Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.6.1)
Requirement already satisfied: wrapt<1.11,>=1.10.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.10.11)
Requirement already satisfied: futures<3.2,>=3.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.1.1)
[... 299 lines cut ...]

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-02-08:
It took me like 30 sec to scroll through that to reply, lol!  Maybe post a link next time instead?

So when you said you tried installing pillow, what happened?  Did you install it into the octoprint virtual environment?  If so, what errors did you receive?

So ive seen issues with pillow before, and the solution is usually either upgrading pip like this:

source home/pi/oprint/bin/activate

Then

pip install --upgrade pip

Then reinstalling.

The other issue was really difficult to figure out.  Pillow apparently fails if your psu is not powerful enough.  I had a hard time believing that at first, but ive seen it several times now.  Check out [this](https://github.com/FormerLurker/Octolapse/issues/234) closed issue.

DrMarkyMarc, 2019-02-12:
**well i got this from what you ask me to do**
Requirement already up-to-date: pip in /home/pi/.local/lib/python2.7/site-packag                         es (19.0.2)

**if i try to install it via console**  ended up with this
`Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?`
`Command python setup.py egg_info failed with error code 255 in /tmp/pip-64i4ibgi-build`

**So i figure I mith not have the good python so I did this**
/home/pi/oprint/bin/python -m pip install "https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip"

**and it get stuck till I Ctrl+c** 
It just freeze withing 2 second of running that command when it get to this line (the thing that usualy turn even stop turning)

building wheel for pillow setup.py 

Iv waited for 30 min nothing happend Ctrl+c work

FormerLurker, 2019-02-12:
Can you post your console output where you try to install Pillow via the command line, including the lines where you activate the virtual environment?

However, the error you are seeing trying to install pillow is probably related to your power supply.  Double (triple) check that your PSU is powerful enough to run your pi.  This causes all kinds of insane issues.  Octoprint has a feature to detect undervoltage if you're running the latest version.  Not sure if you read the previous thread on this issue, but [here is is again](https://github.com/FormerLurker/Octolapse/issues/234) in case you haven't.

DrMarkyMarc, 2019-02-21:
i'm powering my rpi via my 3d printer board via gpio pin I will look how much power it give but I do not see a not enouff power logo in octoprint

FormerLurker, 2019-02-21:
Can you post all console output for when you are trying to install pillow via the command line please, include everything after you sign in.  Thanks!

wtfdic, 2019-03-09:
Having a very similar issue. Went through everything in the other thread about PSU and got pretty far now I am facing this error when installing Octolapse:

Command "/home/pi/oprint/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-ETznR_/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-Vv0saC/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/pillow" failed with error code 1 in /tmp/pip-install-ETznR_/pillow/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

FormerLurker, 2019-03-09:
@wtfdic, have you updated pip or tried installing pillow manually?  There are instructions above.  Let me know if you need help.

wtfdic, 2019-03-10:
@FormerLurker got it working here are the steps that worked for me:

source oprint/bin/activate
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo apt-get install python-dev
pip install --upgrade pip
sudo apt-get install libjpeg62
sudo apt-get install libjpeg62-dev
sudo apt-get install zlib1g-dev
sudo apt-get install libfreetype6-dev
sudo apt-get install liblcms1-dev
sudo apt-get install libopenjp2-7
sudo apt-get install libtiff5
sudo apt-get install ffmpeg
sudo apt-get install libjpeg8-dev
pip install pillow

then install octolapse from plugin manager

FormerLurker, 2019-03-10:
@wtfdic, are you running the latest octopi image?  Glad you got it working!

wtfdic, 2019-03-10:
@FormerLurker yes running 0.16.0 on a Raspberry Pi 3 A+

FormerLurker, 2019-03-14:
I'm really confused as to why Pillow seems to install fine for some people on the newest OctoPi, but not for others.  If it's not PSU related, perhaps some other version of libjpeg62-dev was installed previously before your PSU issues were sorted out?  In any case, since we probably can't replicate the issue anymore (it works for me with a fresh install), I'm going to close this issue.  I really wish Pillow would give me fewer install problems, but it is free and extremely useful, so I won't complain too much :)

Let me know if you think I should reopen this issue for any reason.

closed 2019-03-14 by FormerLurker