FormerLurker/Octolapse #323: Strange Printer Behavior When Starting Print With Octolapse On

MetalTurtle18, 2019-02-19:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse_0.3.4

#### Version of OctoPrint

OctoPrint Version: Octoprint_1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Started Print With Octolapse Enabled

#### What should have happened?
Printer leveled, primed nozzle, started printing. After first layer, octolapse starts to take pictures.

#### What happened instead?
Printer leveled, started priming nozzle, moved to center of build plate, moved back to finish priming, moved to center for picture, started printing, jumped right back to center, then begin printing normally.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.0PE

#### Printer model & used firmware incl. version

Printer Model: Original Prusa I3 MK3
Printer Firmware Version: 3.3.1-845

#### Browser and version of browser, operating system running browser

Browser: Google Chrome 72.0.3626.109
Browser OS: Running on Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/MetalTurtle18/18ee21bd5afb3351a7eb429abec7b447

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/MetalTurtle18/9f7ad6b77264daf5bb49f72192895901

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/MetalTurtle18/a8dd2c3aa5f8d6c136b8f556432b1143

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://drive.google.com/open?id=1FM_vyOOs7iprDhPXOnTBk0VEbOT706W6

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-02-19:
That behavior is expected in the current version of Octolapse for various reasons.  It's possible to prevent in the current version (albeit a bit tricky) by priming with a slightly thicker line than you are printing at and setting the 'priming height' to that value.  I think the MK3 primes at 0.25mm by default?  

However, the next version has some new features to eliminate such behavior, including snapshot boundaries to exclude the priming area, better layer detection so that the first snapshot would be taken BEFORE that prime started, and the ability to prevent snapshots before the first layer (lots of people want to have the first snapshot be of an empty bed, so there's some debate about when to take the 1st pic).

Let me know if I've answered your question, and thanks for the video!  That made things more clear.

MetalTurtle18, 2019-02-19:
That answered my question. Thank you. Do you have any estimate of when it could be fixed?

FormerLurker, 2019-02-19:
It will me a month or two.  This next release includes some major rewrites and new features, so it will take a while to get it all tested and ready for release.  I'm adding gcode preprocessing, automatic slicer settings extraction from gcode files for Slic3r PE, Simplify 3d and Cura, new stabilizations and more.  Stay tuned :)

MetalTurtle18, 2019-02-19:
Ok. That helped me out. Thank you.

closed 2019-02-19 by MetalTurtle18

2probro, 2023-02-16:
> 

Is it already implemented? I can't find the setting to enable this. I don't want my printer to start taking pictures while priming my nozzle