FormerLurker/Octolapse #324: Octolapse stops/pauses print after some layers

AnduriI, 2019-02-19:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
I tried to print different models with Octolapse enabled. It works in the way that the nozzle goes to the correct position on each layer (at back center) and returns to continue. Some layers in the model the nozzle will not return anymore and stay until stepper motor timeout (where I aborted the print). This happens reproducable at the same layer for the same gcode, but on different layers for different models.

#### What should have happened?
It should have continued to print.

#### What happened instead?
It stayed at back center.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version
Printer Model: modified CR10-mini
Printer Firmware Version: Marlin 1.1.9

#### Browser and version of browser, operating system running browser
Browser: Firefox 65.0.1
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: [gcode_failing.zip](https://github.com/FormerLurker/Octolapse/files/2881578/gcode_failing.zip)

#### Link to settings.json
Link to settings.json with all passwords removed: [Settings.zip](https://github.com/FormerLurker/Octolapse/files/2881614/Settings.zip)

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log: [plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/2881618/plugin_octolapse.log)

#### Link to octoprint.log
Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser
Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-02-19:
I'll see if I can reproduce the issue, but first, can I ask if you have the 'Display real snapshot time' option enabled in the Main Settings (OctoPrint Settings [wrench/spanner icon]->Octolapse->Main Tab in Octolapse Settings->Edit Main Settings->Display real snapshot time)?  If so, try turning that off and see if the freezes go away.

I've learned that some printers don't like to have lots of M400s thrown at them, and some seem to lock the serial interface if more than one arrives before moves have completed.  I believe I've worked around this by adding some additional checks to prevent a second M400 from being sent if no movement occurs since the last one, and will need a tester once this code is ready.

Without the 'Display Real Snapshot Time' only 1 M400 should be sent when taking a snapshot, which most printers seem to be OK with.

Also, if you encounter another freeze, try sending 'M114' via the Octoprint terminal.  That may work depending on what the problem is.

Thanks for your post!

FormerLurker, 2019-02-19:
One more thing, can you disable the 'Detail Progress Plugin' also when you run your next test?  It's probably not causing any issues, but it is sending gcodes while Octolapse is taking snapshots, so better safe than sorry.  Also, if by chance that plugin is involved, I'd like to take a look at the code and see if I can discover/work around the issue.

Thanks again!

AnduriI, 2019-02-19:
well I did a little test print just before your second answer arrived, here the results:
I turned off the snapshot time and issued a print (20 layers, 6min). The nozzle was parked without return on layer 8, just as before. Sending a M114 made it continue for same layers, I had to issue this 3 or 4 times on this print.
Now I will try with Detail Progress disabled.

As a side note: I also see the progress bar in the left column going crazy, showing values of 80% already in layer 3 and going way beyond 100% (but don't know if that's relevant or purely cosmetic)

FormerLurker, 2019-02-19:
Hmm.  If it fails again, can you disable all other plugins and try again?  Octolapse shouldn't affect the progress bar at all.  Sounds like a conflict that i need to know about

AnduriI, 2019-02-19:
completed print with detailed progress disabled... that's a solution for me. In case you need to test future hotfixes just drop a message, I would be glad to help

FormerLurker, 2019-02-19:
Great, I appreciate that!  I'll take a look at that plugin and see what's going on.  Honestly it's probably not an 'issue' with either plugin, just a conflict since both read and modify gcode.  There is probably a workaround though, and I'll find it if possible.

I might close this issue and add a new one called something like 'incompatible with detailed progress plugin' if I discover a workaround on my end.

Again, thanks for the report!

FormerLurker, 2019-02-19:
FYI, I believe I've found the issue.  Not sure why the detailed progress plugin is involved, but it probably has something to do with multithreading in OctoPrint.  Octolapse looks for M114 responses, which usually look something like this:
X:149.90 Y:109.90 Z:2.00 E:0.00 Count X:11992 Y:8792 Z:741

However, when your print stopped I see this in the log:
line:X:149.84 Y:109.89 Z:2.00: 0.00 Count X:11987 Y:8791 Z:741

Notice that the E parameter is missing.  In fact, I even had trouble copying and pasting that line from my editor, so I suspect some hidden ascii characters are in there too.

I'll do a bit more digging and see if I can find out more, but I'm in the middle of a massive refactor, so I best get back to that before I forget what i'm doing, lol!

FormerLurker, 2020-02-26:
@AnduriI, I made some changes to the way M114 is parsed, so it's possible your issue has been solved.  Sorry for the huge delay in getting back with you, but I've had quite a backlog to deal with and not enough freetime to work on it all.

You can install from the plugin manager with this url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

A TON has changed, so be sure to read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) all the way through before installing, paying close attention to the *automatic slicer settings* configuration, and the sub-sections on each slicer type.

If you have any problems, please report them in [this thread](https://github.com/FormerLurker/Octolapse/issues/455).

I'm going to go ahead and close this issue, and will wait for your responds on the v0.4.0rc1.dev4 thread.  THANKS FOR POSTING!

closed 2020-02-26 by FormerLurker