FormerLurker/Octolapse #325: [REQUEST] Keep images on the DSLR

yet-another-average-joe, 2019-02-22:
Hi !

I'm currently playing with Octolapse and gphoto2. At first, something is very disturbing : pics are not written (or deleted) to (from) the SD card (maybe the pics are jut taken from the buffer and not even written).

So, I played with gphoto2 CLI options.

I see at least one easy enhacement for OctoLapse : "keep" JPG on the SD card. Just a check box somewhere in the options.

In take-snapshot.sh, I replaced :

`# sudo gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"`

with 

`# sudo gphoto2 --keep --set-config capturetarget=1 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"`

It could be even better with :

`# sudo gphoto2 --keep --keep-raw --set-config capturetarget=1 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"`

as it prevents gphoto2 from downloading raw pictures if they are enabled on the DSLR. Really counter intuitibve, but it seems to work this way.

I tested with take-snapshot.sh ; at this time I did'nt test on a real life print, as my printer's OctoPrint seems to be completely messed up, and I gave up for now (too much prototypes to be printed) So I'm playing with another Pi.

I could'nt find anything about DSLR configuration in gphoto2-Mosaic-compatible-website (I can even hear a V22 modem connecting to Compuserve every time I open the site) ...

It would be cool to have the options about picture resolutions, RAW, keep on SD card or not, directly in OctoLapse. At least "keep on SD card". Just a little checkbox, and a $7 parameter in take_snapshot.sh : SNAPSHOT_KEEP.

FormerLurker, 2019-02-22:
I'll consider this.  I'm also considering just integrating gphoto2 itself.  Once the current round of enhancement to Octolapse is complete I plan on overhauling the camera setup, so that might be a good time to look into one or the other.  Here are some things that are planned:

1.  Create a much enhanced webcam settings page that lets you preview live stream and adjust settings on the fly.
2.  Crate a plugin and hook for GoPro cameras that have wifi.
3.  Create some standard camera scripts and/or integrate gphoto2 with Octolapse.  One or the other of those may come later.  It's MUCH easier to just add parameters as you suggest, but I'd like to be able to do it in a customizable way, perhaps with some pre-configured options like the one you mentioned.  Perhaps a 'Custom Arguments' section?
4.  Integrate GPIO triggers for things like flash, external rig control, etc.

That's pretty much in order of priority, though I've not thought everything through 100% yet.

FormerLurker, 2019-02-22:
Oh, btw, I'm assuming you've seen [this wiki page](https://github.com/FormerLurker/Octolapse/wiki/DSLR-Image-Capture-With-No-Download), but since you were disturbed to learn that images weren't on the SD perhaps not?  [This guide for deferred download](https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download) might also interest you.  You'll need to play with the CLI a bit to determine what works for your camera, however.  I've fielded a lot of issues where specific models don't work like mine.

Note that it takes some time to run -set-config capturetarget, so that's best to perform those operations in the camera initialization script.  Also, if you only want images on the SD card you want to use --trigger-capture, which (on some cameras at least) will return after the image is captured to internal ram but before it is saved to SD.  This REALLY cuts down on the snapshot time a lot.  I mean a whole lot.  In my case it went from around 5 seconds to a under 2 seconds.

yet-another-average-joe, 2019-02-22:
Oh thanks ! I was sure I read something about this, but I thought it was some trick you gave as an answer to a feature request. Spent half an hour searching. Now the link is on the desktop ! These pages are not accessible from Configuring an External Camera (DSLR), or I missed something...

About set-config capturetarget, yes : I found on a forum that it is persistent from a snapshot to another.

FormerLurker, 2019-02-23:
I've been helping several people with DSLRs over the last couple days, and the issues are starting to bleed together, lol!  That script is still beta (you'll notice that there are unfinished sections).  If you have any suggestions or comments for any of those (or the download script), please let me know!

yet-another-average-joe, 2019-02-23:
There's something I don't understand in "DSLR Image Capture With No Download". I see a potential issue with a script.

First of all, I had a hard time before I undertood how to get the parameters for gphoto2 commands ! As soon as I understood, I did : 

`gphoto2 --get-config capturetarget`

and got only 2 options : 0 : internal and 1 memory card

In initialize-camera-save-to-sd.sh, you wrote :

```
...
gphoto2 --set-config capturetarget=2
...
```

What is "capturetarget=2" ? (it does not exist on my Nikon D5100) Maybe a 2nd SD card ? According to NikonHacker, D5100 and D7000 share the same Fuji chipset (AFAIR), but the 5100 has no second slot. capturetarget=2 does work, with no such a target ! Should'nt work. The firmware could redirect to the 1st and only slot ?

Could capturetarget=2 be a problem with other camera brands or chipsets or firmwares ?

FormerLurker, 2019-02-23:
The problem is the correct option depends on your model.  I have a D5200, and that option works for me.  This is the main reason these guides are still beta, and you're not the only person to have issues with the script relating to the capturetarget parameter.  When I have some time I'll add the steps I used to determine which target to use.  There are ways of querying the options available for (some?) cameras in gphoto2.

yet-another-average-joe, 2019-02-23:
I had no issue at all ! It works fine. I think it is because the hardware is made for 2 SD cards. I'm curoius : what does "gphoto2 --get-config capturetarget" return on your D5200 ? (if you have time)

FormerLurker, 2019-02-23:
Hmm.. I must have worked on this late at night because this is the output:
```
Label: Capture Target
Readonly: 0
Type: RADIO
Current: Memory card
Choice: 0 Internal RAM
Choice: 1 Memory card
END
```
I would have sworn it wasn't working unless I set it to 2, lol!  Anyway, I stumbled on this variant to set the capture target.

'''gphoto2 --set-config-value /main/settings/capturetarget="Memory card"'''
and
'''gphoto2 --set-config-value /main/settings/capturetarget="Internal RAM"'''

Does this work for you?  That seems more intuitive.

Also, I'm not sure why I thought you were having problems with capture target, you were pretty clear.  I've been working on a C python extension, and have been working on an error that's driving me nuts.  I think it's slowly destroying my brain.

yet-another-average-joe, 2019-02-23:
Don't know where you live, but where I live, you answered at 4H AM. On my side, slept 4 hours, and back ! Could'nt sleep until I understand how to use set config value Today, I will probably  put my hands on a D7000 (2SD cards) and see what the value 2 does, and what get config value returns.

Thre's something that makes the code you just wrote above not 100% usable : it depends on localization.  

This is what I get (RasPi french, Nikon english) : 
gphoto2 --get-config capturetarget

```
Label: Cible d'acquisition
Type: RADIO
Current: Carte mémoire
Choice: 0 Mémoire interne
Choice: 1 Carte mémoire
```
other --get-config xxxxxx are also localized

get/set-config-value is exciting because it could allow OctoLapse to save and restore DSLR settings : mode, ISO, speed, aperture, etc. (if lightings remain the same, it makes sense)

I imagine a page with with all values returned by --list-config, with 100's combo boxes filled with the answers for each and evry settings. (just kidding, it's irrealistic and unusable)

Some data are usefuil, and should exist on most DSLRs : datetime, flash, batterylevel, iso, and some others.

An idea : the user could set his DSLR as he wants, then push a button in some dialog box in order to save his settings. For further timelapses, he could restore these values.

FormerLurker, 2019-02-23:
Localizations strike again, of course!  I wonder if the camera is localizing the return values or gphoto2.  If it's gphoto2, I bet there's a way to set the localization too.  Either that or I could localize my guide.  Do you know any good translators :)  I really need to add localization to Ocotlapse as well, but it's a daunting task.  Eventually I'll get around to it, hopefully before the 1.0 release.

Regarding your sleep, you sound like me!  I got up way early because I was dreaming about a solution to my c++->python problem.  Will see in an hour or two if the fix I literally dreamed up works, lol!

I like your idea regarding extracting and saving settings, but there sure are a lot of them!  This would be easier with a direct integration with gphoto2, but there are distribution issues I need to figure out, like how to supply the correct library for libgphoto2, and what to do about windows/macos (or rather what not to do...). 

I'm in Chicago, BTW.

yet-another-average-joe, 2019-02-23:
It's easy to see. ASAP I will flash a non localized version of OctoPi, and I'll keep the DSLR on english settings (it was sold on french market, so who knows...). I bet it's localized in gphoto, as the developper  lives in Montreal, Quebec, Canada... Finally, I will not be able to test with a D7000 (dual SD) until a couple of weeks.  :(

I just went back to GoPro scripts, and I will be attempting to feed OctoLapse with GoPro captures within minutes. Should'nt be too difficult (I hope so...)

yet-another-average-joe, 2019-02-25:
I tested on a fresh OctoPi install, with no localization (so it should be en-US as there are en-GB and fr translation files in po subdirectory) : gphoto2 speaks american english.

I had a look to main.c, there is these lines (2192) : 

```
/* for translation */
setlocale (LC_ALL, "");
        bindtextdomain (PACKAGE, LOCALEDIR);
        textdomain (PACKAGE);
```

But I don't know what it does. I imagine that ignoring this should let the program locale alone...

Adding an option "--nolocale" to gphoto2 could be the idea. I can't help : Microsoft environments are not supported, so I can't mod and compile to see ; Visual Studio user ! I don't know anything about Linux systems, and VS does not cross compile for Linux (iOS and Android only)

Maybe a feature request on gphoto2 Git ?

You wrote : 

>  Do you know any good translators :) I really need to add localization to Ocotlapse as well, but it's a daunting task.

Do you mean creating localization files, like the *.po I found in gphoto2 sources ? Do some environments exist to do this ? Because translations are often difficult to do : too short or too long strings could kill the controls placements, etc. I've seen too much softwares complely broken in french, with unreadable BUTTONS, STATICS, LISTBOX, etc. In Windows development, resource editor makes things not too difficult. I had a look to jinja : I understand nothing ! Is there a WYSIWIG IDE ?

FormerLurker, 2019-02-28:
@yet-another-average-joe, hey sorry about the delay.  I got (and still have) a bad case of the flu, so I've got a huge backlog both at work and in my git issues.  I'll start trying to clear them out ASAP after I start feeling better.

yet-another-average-joe, 2019-02-28:
Take care of you ! Got this once, and only once : I remeber how bad I feeled. Timelapses can wait. On my side, I just came back to gopro api : I was finalizing a quick change hotend system.

yet-another-average-joe, 2019-03-15:
I had an idea... A workaround until you release a next version.

In the Python script, photos MUST be downloaded, or OctoLapse will complain.
The idea : using PIL (Pillow ?), a fake image can be created and given to eat to OctoLapse, after the pic is taken but not downloaded (modded gphoto cmd line). Let's say a dummy 1px JPG image. 3 lines in a Python script.
Every time a photo is taken, this dummy image is copied to the name given by OctoLapse to the script. Therefore, everybody will be happy : the user who does want to create timelapses by hand, and OctoLapse who will have something to work with, without throwing errors. And processing 1px images will be very fast so the printer will be available again immediatly.

It is also very interesting for GoPros, as there is an excellent (Cineform based, but abandonned) Win and Mac app for timelapses, and GoPros seem to easily timout. GoPros H4+ can take snapshots in Protune mode (raw). This app automatically apply filters on raw images when it comes to create timelapses. This would solve 2 problems : timeouts, and Protune.

I will works on these tiny lines of bash and Python scripts tonight. It should work.

yet-another-average-joe, 2019-03-16:
I got this to work : snapshots, no download, keep images on SD card. Needs a python script and a modded take-snapshot.sh

As explained in the previous post, it takes a snapshot (DSLR), does not download. Then it calls a python script that create a dummy JPG (1x1 pixel, red), and rename it according to what is expected by OctoLapse.

```
keep-snapshot-no-download.sh :

#!/bin/sh
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# IMPORTANT - You must add gphoto2 to your /etc/sudoers file in order to execute gphoto2 without sudo
# otherwise the following line will fail.
#sudo gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"

sudo gphoto2 --capture-image

python /home/pi/scripts/keep-snapshot-no-download.py $6

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_PATH}'." >&2 
  exit 1
fi

```

keep-snapshot-no-donwload.py :

```

import sys
import shutil
import os

sys.path.append("/home/pi/oprint/lib/python2.7/site-packages")

from PIL import Image

img = Image.new('RGB', (1,1), (255, 0, 0))
img.save("/home/pi/scripts/dummy.jpg", "JPEG")
shutil.move("/home/pi/scripts/dummy.jpg", sys.argv[1])

```

The scripts are in /home/pi/scripts

yet-another-average-joe, 2019-03-17:
I followed your suggestions about keeping images on SD card. And it successed.

```
#!/bin/sh
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# IMPORTANT - You must add gphoto2 to your /etc/sudoers file in order to execute gphoto2 without sudo
# otherwise the following line will fail.

# /home/pi/.gphoto/config change contents : everybody

#sudo gphoto2 --set-config capturetarget=0
sudo gphoto2 --set-config capturetarget=1
sudo gphoto2 --capture-image-and-download --keep --filename "${SNAPSHOT_FULL_PATH}"
#sudo gphoto2 --capture-image

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_PATH}'." >&2 
  exit 1
fi

```

At first, it didn't work : "camera error" for each and every photo. Searching the web, I found a thread where gphoto2's author explained that the file /home/pi/gphoto/settings had to be writable by the programm calling the functions. This flag being set locally, and not on the camera. I changed its rights for everybody (it was for owner only).

Then it worked, but not immediately. But I'm not sure. Because I edit the bash script while a timelapse is running, between 2 shots. (quick and dirty tests, so I can see the error messages in "real time" !). Will have to re-flash the backup OctoPi image and see again. (but in command line, I had no problems to shoot downoad and keep). 

I think that the script approach should be much less work for you. It could just be the rights on "/home/pi/gphoto/settings" to be modified in the gphoto2 install script (?)

There's probably a performance downside. Downloading pics from the SD card rather than from the buffer is probably slower.

I also tested double shots : one with target = 0 (for dowload/delete), the 2nd with target = 1 (to be kept on the SD card). It is slow ! 100% manual mode, no time consuming autofocus.

It would be interesting to see if the mirror can stay upside between photos, and the effect on performance. But I did'nt see such a command in gphoto, and when the DSLR is connected with USB, Liveview is deactivated. At least on Nikon D5100.

Some thoughts about defered downloads...

It is complicated to setup for the user. Then it takes time to download. And then it takes time to make the video. Meanwhile, the printer is theorically unusable.

A dedicated Pi could be great. Communication via Ethernet or via some serial port (GPIO). A minimalist photo/video cluster, with the cheap Pi Zero, could do it.

yet-another-average-joe, 2019-03-26:
about "--set-config capturetarget=xx"

I tested with a D7000 (2 SD slots). No difference if xx is 1 or whatever. This parameter does not save the image to a specific SD card. Somewhere in gphoto2 or firmware, there should be "if( param != 0 )" and that's it.

"--get-config capturetarget" returns 0 and 1, as with the D5100 (1 slot)

Dragon2781, 2019-07-31:
Hi yet-another-average-joe,
I'm a total noob with scripting and Raspberry Pi.
Is this working now? I also want to make my Timelapses by hand.
Can you please give me a step-by-step description what to change?
What commands are needed to add/change this in Putty?

FormerLurker, 2019-07-31:
@Dragon2781, have you looked at [this guide?](https://github.com/FormerLurker/Octolapse/wiki/DSLR-Image-Capture-With-No-Download)  Depending on your camera, this could work.  However, not all cameras function the same, and there are some known problem cameras (newer sony mirrorless cameras, for example).

Dragon2781, 2019-08-04:
Thank you, this is exactly what I need! Great job as always @FormerLurker

Dragon2781, 2019-08-04:
@FormerLurker 
Sadly it is not working with my Canon 7D. Even with 10sec sleep the picture is not on the SD. :(

FormerLurker, 2019-08-04:
I recommend experimenting with gphoto2 on the command line to see how your camera responds.  If it works on the command line, it will most likely work within a script.  You can query your camera's capabilities via gphoto2 also.  Check out this site for details:  http://gphoto.org/doc/manual/ref-gphoto2-cli.html

Check out the --list-all-config and --list-config parameters.  I think they should help you figure out what capturetargets your camera supports.  There are lots of methods for capturing images too that you could try.  It might be that none of them work for your particular camera, but I think it's worth a shot.

Dragon2781, 2019-08-04:
@FormerLurker I have really no idea how to do that scripting stuff... How can I get to this command line?

FormerLurker, 2019-08-04:
The terminal screen you used in the guide.  If you are using windows, maybe you have used putty before?

Dragon2781, 2019-08-04:
I use putty like in your description but this is all I did with it.

FormerLurker, 2019-08-04:
So connect to your pi and paste in this command and tell me what you see:

 gphoto2 --get-config capturetarget

Dragon2781, 2019-08-04:
pi@octopi:~ $ gphoto2 --get-config capturetarget
Label: Capture Target
Type: RADIO
Current: Internal RAM
Choice: 0 Internal RAM
Choice: 1 Memory card

FormerLurker, 2019-08-04:
Ok, so try this next:
```
gphoto2 --set-config capturetarget=1

gphoto2 --capture-image
```

Then see if the image is on your SD card.  If it is there, try this command next:
```
gphoto2 --trigger-capture
```

See if that image is on your SD now.  Let me know what happens.

Dragon2781, 2019-08-04:
Both images are on the camera, yes

FormerLurker, 2019-08-04:
Well, it should be working then :)  Do you have the initialization script location (from the guide, basically the following line:gphoto2 --set-config capturetarget=1)?

If you screencap all of your script settings in your camera profile and paste it in here I can look at it.  Also, if you want to paste in the actual scripts I can look at those too (there should be 2, the initialization and capture script).

Also be sure to disable rendering (the enabled check box in the rendering profile), else you'll get an error after your print ends.  However, the error won't hurt anything.

Dragon2781, 2019-08-04:
I just redid all your description and it seems to work. Let me do a Testprint...

Dragon2781, 2019-08-04:
It is working!!! You are the BEST!
One thing: Is it possible to take a picture after the last layer? It seems like the last picture is after the second-last layer.

FormerLurker, 2019-08-04:
Yes, you can use the gcode trigger and add the snapshot command pseudo gcode command to your layer change script and at the very first line of your end gcode script.

FormerLurker, 2019-08-04:
Fyi, the default command is snap, and can be changed in your printer profile.

Dragon2781, 2019-08-04:
No idea hahahah! Nevermind, I'm ok with it how it is now. Thank you again!

FormerLurker, 2019-10-12:
OK, so I've gotten several reports now that this appears to be working for many cameras, and have updated the guides with instructions on how to find the correct capturetarget.  I'm going to go ahead and close this issue, but if anyone wants it reopened for some reason, let me know!

closed 2019-10-12 by FormerLurker