FormerLurker/Octolapse #326: [REQUEST] gphoto2 built into octolapse with multi-camera support.

MineThingIssues, 2019-02-24:
#### If this is a feature request describe it here
To save people the hassle of trying to install gphoto2 with the possibility of making a mistake and having to either start again or send an issue here where other people have to deal with it (Which could obviously get annoying at some point), having gphoto2 built right into octolapse could be incredibly handy. Having the latest version already installed (or maybe something that you click a button to install) would result in fewer bugs and could support more cameras (Such as the Canon EOS 1300D and the 1400D (Both are very popular at the moment as it is "budget-friendly") which had major bug fixes in 2.5.18 which is more recent than the version installed via "sudo apt-get install gphoto2")

Although it may be easy to install yourself, some people (like me) who are not very technically advanced may find the task of SSH-ing into a Pi and copying and pasting complex code that they don't understand a very daunting task (Although I have learnt a lot from the experience).

Having multi DSLR support is possible with gphoto2 by specifying the port for each camera (use gphoto2 --auto-detect to get the info before each octolapse and parse out the port and name into variables for easy use, then specify the port and/or camera when calling a gphoto2 command). It may take a bit of setting up but I think that for some people It could be really useful.

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-02-28:
@MineThingIssues, sorry about the delay, been (and still am) really sick with the flu.  I think I can respond to this though. 

The problem is GPhoto2 isn't something that I can add to the Octolapse installer (at least I don't know how).  I would LOVE for that to happen automatically.  I think our best bet would be to convince the good folks who maintain OctoPi to include it in their image (along with pillow).  I've looked into including libgphoto2, and will continue to investigate that.  What do you think?

I apologize in advance for being late to respond.  Need sleep :)

MineThingIssues, 2019-03-05:
Hope you get better soon! Sorry for the late reply, I have been pretty busy recently and I forgot to check for a reply. I am not a developer and have very limited knowledge of python or any other programming language so forgive me if I say something that doesn't make sense or is outright stupid:

My idea is to have some sort of button that the user can press that runs a script that runs these commands in the terminal (or whatever it is called. The thing that you have immediate access to when you SSH into the Pi) (Source: https://hyfrmn.wordpress.com/2015/02/03/install-libgphoto2-and-gphoto2-from-source-on-raspberry-pi/): 
Install dependencies:

1. `sudo apt-get install libltdl-dev libusb-dev libusb-1.0 libexif-dev libpopt-dev`

Download latest stable versions:

2. `wget http://downloads.sourceforge.net/project/gphoto/libgphoto/INSERT.LATEST.VERSION/libgphoto2-INSERT.LATEST.VERSION.tar.gz`
3. `wget http://downloads.sourceforge.net/project/gphoto/gphoto/INSERT.LATEST.VERSION/gphoto2-INSERT.LATEST.VERSION.tar.gz`

Install libgphoto2:

4. `tar -xvzf libgphoto2-INSERT.LATEST.VERSION.tar.gz`
5. `cd libgphoto2-INSERT.LATEST.VERSION`
6. `./configure`
7. `make`  (Takes a long time especially on an older raspberry pi)
8. `sudo make install`

Then gphoto2:

9. `tar -xvzf gphoto2-INSERT.LATEST.VERSION.tar.gz`
10. `cd gphoto2-INSERT.LATEST.VERSION`
11. `./configure`
12. `make` (Much faster then libgphoto2)
13. `sudo make install`

This way, It will not take up the users time or storage if they don't necessarily want to use a DSLR as they have the option to install it. Again, I do not know what the capabilities of plugins are and I completely understand if this is not possible or practical.

The scripts that you modify and create (in the guide you created for using a DSLR with octolapse) could then be either downloaded or modified (but backup first in the case of a malfunction or problem) (I do not know the capabilities of python read/write/append, especially in 2.7, and have no knowledge of shell scripts).

Then it should do some checks to make sure that everything is running correctly. If it is not, then it should restore the backup and say that it failed.

Again, I do not know enough about python or programing in general to see if this is possible, but I did write a very basic script in python 3 that writes a file, reads a file and appends a new line of text to an already existing file just for my own satisfaction. (This is what lead me to this probably impossible idea in the first place).

Edit: what you are about to read probably either sounds like I am repeating myself or just making things harder, so sorry for the confusion and you may want to just ignore this next paragraph:

Just brain storming now, but would it be possible to just download a shell script that has all the commands already layed out and ready to run (sorry if this makes no sense)? That way, the button could just run that shell script to install everything. 

Also, if you would like, I could spend a bit of time (in my free time) learning how to write shell scripts and then I could try and get one working for you so that you can work on other more important things? I would really like to do this for my own satisfaction and to help others anyway, so I might just get started now and see how far I get 😜 

Thank you for everything you are doing to help the community and I hope you get better soon

MineThingIssues, 2019-03-06:
ok, I have written a very basic script that may or may not work. I have not tested as of yet, but if it does work, It would be cool to have it linked in one of the guides to help people who are afraid of forgetting a step or messing something up. WARNING: Needs a restart after completion for the sudoers file to be updated or something like that. I did not add that to the script because someone may not want it to restart or they may panic if they see it restarting when they weren't expecting it too.

2.5.22 is the current latest libgphoto2 version and 2.5.20 is the latest gphoto2 version. There is a way to get the version number for the latest version automatically and use variables to set the version numbers in the script to that version, but I do not know how to filter the results I get when doing the command and save it to a variable correctly.

`#!/bin/bash
#script for installing gphoto2 and libgphoto2 and also setting up a DSLR with OctoLapse
#this script comes with no warranty and I am not held responsible for any damage done
#This script is still in BETA and has not been full tested (Or in fact tested at all)

#Go main directory
cd

#Install dependencies
sudo apt-get install libltdl-dev libusb-dev libusb-1.0 libexif-dev libpopt-dev

#Download latest stable versions
wget http://downloads.sourceforge.net/project/gphoto/libgphoto/2.5.22/libgphoto2-2.5.22.tar.gz
wget http://downloads.sourceforge.net/project/gphoto/gphoto/2.5.20/gphoto2-2.5.20.tar.gz

#Install libgphoto2
tar -xvzf libgphoto2-2.5.22.tar.gz
cd libgphoto2-2.5.22
./configure
make #Takes a long time especially on an older raspberry pi
sudo make install

#back to main directory
cd

#Install gphoto2
tar -xvzf gphoto2-2.5.20.tar.gz
cd gphoto2-2.5.20
./configure
make #Much faster than libgphoto2
sudo make install

#make it work
ldconfig

#----------Setting it up with OctoLapse----------#

#Setting up sudoers
echo 'pi ALL = (root) NOPASSWD: /usr/bin/gphoto2' >> /etc/sudoers
echo 'pi ALL = (root) NOPASSWD: /usr/local/bin/gphoto2' >> /etc/sudoers

#move to scripts directory
cd /home/pi/scripts

#download script
wget https://gist.githubusercontent.com/MineThingIssues/6f29fbdd623e77d09afe37b1eaef4982/raw/1e95240ec2981022ef72526f88a2508adba96d6c/take-snapshot.sh
chmod +x take-snapshot.sh

#now add "/home/pi/scripts/take-snapshot.sh" to the snapshot aquire script box in the settings`

FormerLurker, 2019-03-06:
I haven't had a chance to review the quite large volume of scripts you've sent (impressive, btw), but I wanted to show you [this script](https://github.com/gonzalo/gphoto2-updater) if you haven't seen it.  It only downloads and installs (or uninstalls if necessary) gphoto2, but it's quite through.  Git has been scripted in OctoPi, so I did have to make a minor change due to this in order to get that script running.  See [this](https://github.com/gonzalo/gphoto2-updater/issues/64) issue for details about the changes.

MineThingIssues, 2019-03-06:
Wow, that looks complex, yet it looks like it does a pretty similar technique for installing gphoto2, but I guess it has a lot of checks to make sure that it all works perfectly and replaces the older version of gphoto2 correctly (something I couldn't work out how to do manually at all and had to do a fresh install of OctoPi originally). My script is designed to run on a pi that does not have gphoto2 or libgphoto2 installed already (although may still work anyway) and has absolutely no checks or anything (probably going to cause problems but it is the best I can do at my skill level).  My script also uses a different source (sourceforge) to download gphoto2 and libgphoto2 as that is what worked for me and I don't have time to test anything else yet. The issue you faced will most likely not happen with my script.

after looking through that script, I have found that you can install and compile gphoto2 a lot faster by running multiple jobs in parallel. I have edited my script to do the same thing (but slightly different because I don't use any variables lol)