FormerLurker/Octolapse #327: Camera tested OK but no streams?

dannieboiz, 2019-03-03:
I'm connecting Octolapse to an IP cam stream. In the settings, I'm able to test with no error. This same stream works in the built in timelapse but for Octoprint, I'm unable to get any image up. 

I've tested both streams from blue iris as well as the direct stream from the camera but no go.

FormerLurker, 2019-03-03:
Octolapse doesn't show the webcam stream in its tab.  Instead, it shows the most recent snapshot, which will appear after you start printing.  Let me know if that helps!

FormerLurker, 2019-03-14:
@dannieboiz, I'm going to close this issue for now, since I don't think there is any error here.  I am planning to make it more clear what that image preview is for, and might even add live stream or snapshot capabilities in an upcoming release.

Let me know if you want me to reopen!

closed 2019-03-14 by FormerLurker