FormerLurker/Octolapse #328: How can I force rendering?

ulitiy, 2019-03-04:
Version: 0.3.4
OctoPrint Version: 1.3.10
Diagnostic Logging was Enabled: NO

1. I printed
2. **Rpi had some issue after the print so I couldn't access it until hardware reset**
3. Now **I don't have a video result but I have the snapshots in the folder**

**How can I force rendering?**

OS Name: Octopi
Os Version: 0.15.1

Printer Model: Prusa i3 mk3
Printer Firmware Version: 3.5.1-1778

Logs: https://gist.github.com/ulitiy/361c3b28f6407ffa5c47a2ee07decbc0

FormerLurker, 2019-03-05:
Yeah, I probably need to add some routine to render failed timelapses.  As of now you'll need to download the images and manually render them (a pain in the butt, I know).  The good news is that should be properly named to use either ffmpeg or avconv to generate your timelapse.

I'll think about some kind of routine to search for and render timelapse images that are found on reboot to fix this in the future.

FormerLurker, 2019-03-14:
FYI, the images should be available in /home/pi/.octoprint/data/octolapse/snapshots/{some crazy guid}/{another crazy guid}/

I'm going to mark this closed.  Let me know if you want me to reopen, or if you need help getting your images.  Look for the option to render failed timelapse images in a future version of Octolapse.  Probably not the next version, but definitely the one after that.

closed 2019-03-14 by FormerLurker

D3473R, 2019-03-31:
Today my print too came back with only snapshots and no video, maybe the automatic shutdown plugin turned octoprint off before the rendering was done :/
Anyway, maybe this ffmpeg command can help someone in the same situation 
`ffmpeg -framerate 30 -i image%03d.jpg video.mp4`
The `%03d` stands for three digit number (i had 800 snaphots), if you have between 10 and 99 snapshots use `%02d` and so on.
With the framerate option you can specify the target framerate for the rendered video.