FormerLurker/Octolapse #329: Communication error

pingouin84, 2019-03-07:
#### If this is a feature request describe it here

Communication error

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Go print
2. position snapshot
3. snapshot
4. communication error

#### What should have happened?
the impression should have been

#### What happened instead?
nothing, the imprinate does not move anymore

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.13.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 mk2
Printer Firmware Version: 3.1.0

#### Browser and version of browser, operating system running browser

Browser: chrome
Browser OS: 72.0.3626.119

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/pingouin84/45350aada29879c9ae312ba5f1928a0d

#### Link to settings.json

Link to settings.json https://gist.github.com/pingouin84/c712241aa08c067f74b5127b0b3002af

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/pingouin84/ebbc5e6ee1bfe176d3f82ffd24837b03

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/pingouin84/409be2850d5b5bc6a30afa80aa8c2d3e

FormerLurker, 2019-03-07:
I see that you are running multiple cameras.  It looks like your DSLR is timing out.  Can you try running without the DSLR to see if you get the same error?  If not, turn off the webcam and try it again with ONLY the dslr and let me know what happens.  Also, if you post your snapshot script for the DSLR that may help a bit too.

Also, have you ever gotten the DSLR to work, or is this your first time?

pingouin84, 2019-03-07:
web cam OK
DSLR (nikon coolpix S3600) KO

`#!/bin/sh
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# IMPORTANT - You must add gphoto2 to your /etc/sudoers file in order to execute gphoto2 without sudo
# otherwise the following line will fail.
sudo gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_PATH}'." >&2 
  exit 1
fi`

pingouin84, 2019-03-07:
test /home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg" OK

FormerLurker, 2019-03-07:
Please surround your code with triple `s like so:
```
#!/bin/sh
```

Due to the formatting I can't be sure, but if you have an apostrophe in front of this command like so:
```
`#!/bin/sh
```
the scrip will not work.

When you run that test script does a jpg get downloaded?  Finally, I assume you edited your sudoers file and rebooted like the guide says, right?

pingouin84, 2019-03-07:
' error post comment

pingouin84, 2019-03-07:
the script function very well, tested with /home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"

FormerLurker, 2019-03-07:
@pingouin84, don't be fooled into thinking that just because the test above succeeded that there can be no problems executing this command within OctoPrint.  Here are the log entries where the snapshot script is being executed:

```
2019-03-07 17:29:29,694 Running the following snapshot script command with a timeout of 20.0: /home/pi/scripts/take-snapshot.sh 0 0.0 /home/pi/.octoprint/data/octolapse /home/pi/.octoprint/data/octolapse/snapshots/d5785913-687d-40e8-8fb2-c698d6248d4b/9779a539-ec45-4817-847c-156f36d7d8ae heart_box_3_base_0.2mm_PLA_MK2S000000.jpg /home/pi/.octoprint/data/octolapse/snapshots/d5785913-687d-40e8-8fb2-c698d6248d4b/9779a539-ec45-4817-847c-156f36d7d8ae/heart_box_3_base_0.2mm_PLA_MK2S000000.jpg
2019-03-07 17:29:32,685 Received from printer: line:
2019-03-07 17:29:32,695 Queuing Command: Command Type:temperature, gcode:M105, cmd: M105, tags: set(['trigger:comm.handle_timeout'])
2019-03-07 17:29:33,354 Queuing Command: Command Type:temperature_poll, gcode:M105, cmd: M105, tags: set(['trigger:comm.poll_temperature'])
2019-03-07 17:29:35,707 Received from printer: line:
```
This wait loop continues until right around here:
```
2019-03-07 17:29:44,772 Queuing Command: Command Type:temperature, gcode:M105, cmd: M105, tags: set(['trigger:comm.handle_timeout'])
2019-03-07 17:29:47,776 Received from printer: line:
2019-03-07 17:29:47,800 Printer event received:Error.
2019-03-07 17:29:47,804 Printer event received:UpdatedFiles.
2019-03-07 17:29:47,863 Printer event received:Disconnected.
2019-03-07 17:29:47,866 Printer disconnected.
```

Since handle timeout gets called, it looks like your default communication timeout is being hit before your snapshot is complete.  Check out OctoPrint settings (wrench/spanner icon)->Serial Connection->Intervals & Timeouts->Communication Timeout.

However, your best bet is probably to reduce the amount of time it's taking to download your image.  Make sure your camera is set to only create a JPG/JPEG and NOT Jpeg + raw!  Second, try turning down your image resolution as a test.  Let me know if that helps.  If so, there are ways to greatly reduce the image acquisition time.

pingouin84, 2019-03-07:
the script my 18 sec to execute.

I double the time temperature interval polling and autoport, it works now

FormerLurker, 2019-03-07:
That's a really long time to acquire an image, and I'm sure your print quality will suffer.  Take a look at [this guide](https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download) to defer download until AFTER your print is complete.  You won't get any snapshot previews, but your capture time will be GREATLY reduced.

pingouin84, 2019-03-07:
what are the solutions to reduce the time of the script?

FormerLurker, 2019-03-07:
Did you notice that you embedded a link to a guide in my last post?  With it you can delay download of all images until after the print is s complete.  The scripts are still beta, so you may have to experiment a bit.

pingouin84, 2019-03-08:
I'm going to watch because the --trigger-capture function is not available on my device

closed 2019-03-09 by pingouin84

pingouin84, 2019-03-13:
little HS, I can not trigger the capture on M117 Snap

FormerLurker, 2019-03-14:
@pingouin84, I think in version 0.3.4 that the snapshot command (printer profile) is case sensitive.  Make it all caps in both places for now like so:  M117 SNAP.  The development version has this fixed, but isn't ready for public consumption unfortunately.