FormerLurker/Octolapse #330: 404 error when trying to access control.htm

LithophaneMaker, 2019-03-25:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE = 0.3.4

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE = 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO = No

#### What were you doing when the problem occurred

1. I changed the octopi.txt file from:
#camera_http_webroot="./www-octopi"
#camera_http_options="-n"
to
camera_http_webroot="./www"
camera_http_options=""
2. I rebooted several times.
3. Tested camera and it worked.
4. Went to OctoPi URL:8080 and it says "404 Error. File not found"
5. I  commented out those variables and rebooted.
6. Went to OctoPi URL:8080 and I see snapshots from my camera (Logitech C920).
7. I retried this several times and scoured the internet.
8. I have made several timelapses so I believe mjpg_streamer is functional.

#### What should have happened?
I should have seen mjpg_streamer settings

#### What happened instead?
I got a simple text webpage that said 404 error. File not found.

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE = OctoPi
Os Version: OS_VERSION_GOES_HERE = 0.15.1

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE = CR10
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-03-25:
Try using OctoPi URL/Webcam/control.htm and see if that helps.

Also, I think this issue is probably more of an OctoPi thing and less of an OctoLapse thing.  The control.html page was deactivated by default in the OctoPi a few versions ago (after I implemented the webcam controls) to prevent people from being able to change the camera settings if your instance was exposed on the internet.

FormerLurker, 2019-10-12:
I'm doing some issue maintenance and I see this is pretty old.  I'm going to close this thread for now, but if you want me to reopen it just let me know.  

Thanks for posting!

closed 2019-10-12 by FormerLurker