FormerLurker/Octolapse #331: Keep Pictures On DSLR With no "capturetarget" config

xxsteven69xx, 2019-03-28:
I wasn't sure about where to put this. Sorry if this is the wrong place. I have a Sony A7riii and waiting for the 42 megapixel jpgs coming over usb2 was causing longer than the 25 second max wait time about 1 out of every 200 pics or so. Plus I render myself after anyways so I found your guide on not downloading the pictures. Unfortunately, sony's don't have the config capturetarget and require the picture to be cleared from ram before allowing a new snapshot. I amended the script from the first guide of how to take pictures with a DSLR to make this work.

#!/bin/sh
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# IMPORTANT - You must add gphoto2 to your /etc/sudoers file in order to execute gphoto2 without sudo
# otherwise the following line will fail.

#Capture image but not download
#NOTE: --trigger-capture will not work for this
gphoto2 --capture--image

#Delete the image waiting to be transfered in ram
gphoto2 -D

#Capture a small (in my case 180Kb) jpg to feed back into Octolapse
gphoto2 --capture-preview --filename "${SNAPSHOT_FULL_PATH}"

#Couldn't figure out how to get --capture-preview not to save as "thumb_somefile" so this is to rename it
(cd $4 && mv thumb_$5 $5)

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_PATH}'." >&2 
  exit 1
fi

FormerLurker, 2019-03-28:
This is the right place, and thanks for the submission!  Once I have a chance to review your and test your code, I'll update the guide.  It seems like I need some camera specific guides too, since as you discovered there's a lot of variability out there.

xxsteven69xx, 2019-03-28:
I'm also looking forward to the windows guide. What was the name of the tool to pull the pictures from camera you use in the script? I want to see if i can tackle it. I have a lattepanda that would be great for this but I have to keep windows on it for other reasons.

xxsteven69xx, 2019-03-28:
I also realize you probably don't have every model of camera out there and if there is any testing or logs I can provide for the A7riii or A77mk2, I'm happy to help.

FormerLurker, 2019-04-01:
@xxsteven69xx , I have used [digicamcontrol](http://digicamcontrol.com/) for windows, but have never gotten it to work 100%.  I get random timeouts which don't prevent a timelapse, but some frames are missing.  When I tried it more recently I couldn't get digicamcontrol to open.  I'm sure it's something simple since I upgraded my hardware since the last time I tried, but haven't had the time to fix.  I have no idea what's causing the timeouts.

Here is a script I apparently used in the past for 'capture and download' with digicamcontrol:

```
set SNAPSHOT_NUMBER=%1
set DELAY_SECONDS=%2
set DATA_DIRECTORY=%3
set SNAPSHOT_DIRECTORY=%4
set SNAPSHOT_FILENAME=%5
set SNAPSHOT_FULL_PATH=%6
if not exist %SNAPSHOT_DIRECTORY% mkdir %SNAPSHOT_DIRECTORY%

"C:\Program Files (x86)\digiCamControl\CameraControlCmd.exe" /filename %SNAPSHOT_FULL_PATH% /capture
rem see if the file exists, if not throw an error.
if not exist %SNAPSHOT_FULL_PATH% (
	echo The file was not created! 1>&2
	EXIT /B 1
)
EXIT /B 0

```

yet-another-average-joe, 2019-04-05:
Tested good with the Nikon D5100.

xxsteven69xx, 2019-04-09:
@FormerLurker my A7riii is not supported on digiCam :( I was also looking to see if the sony remote app or capture one could be triggered through command line and I don't see a way.

yet-another-average-joe, 2019-04-19:
@FormerLurker : I wrote "tested good with Nikon D5100". It seems I was wrong, and called another script... There's something weird. Could you test with you D5200 ?

1st problem on my 2 systems : gphoto2 calls in @xxsteven69xx script have to be prefixed with sudo. This is the reason why it was impossible I correctly tested the script : it could not work as is on my systems.

2nd problem on my systems : I am now unable to shoot more than 1 or 2 photos. Then the the USB port remains busy. I don't know why ! Probably DSLR related...

3rd problem with DSLRs like the Nikon D : @xxsteven69xx script shoots 2 times. This is fine with the Sony, as it is mirrorless. But it is unfortunately barely usable with classic DSLRs (too slow)

@xxsteven69xx : could you explain "gphoto2 -D" between cpature_image and capture_preview ? I tested with and without, and didn't see the difference. This function is not really documented. What does it delete ? file on Pi, buffer or SD card ?

> 
> -D, --delete-all-files
> 
>     Delete all files in folder (defaults to --no-recurse).

xxsteven69xx, 2019-04-27:
> @FormerLurker : I wrote "tested good with Nikon D5100". It seems I was wrong, and I called another script... There's something weird. Could you test with you D5200 ?
> 
> Problem with DSLRs like the Nikon D : @xxsteven69xx script shoots 2 times. This is fine with the Sony, as it is mirrorless. But it is unfortunately barely usable with classic DSLRs (too slow)
> 
> @xxsteven69xx : could you explain "gphoto2 -D" between cpature_image and capture_preview ? I tested with and without, and didn't see the difference. This function is not really documented. What does it delete ? file on Pi, buffer or SD card ?
> 
> > -D, --delete-all-files
> > ```
> > Delete all files in folder (defaults to --no-recurse).
> > ```
> 
> I get this error when using the no capturetarget script :
> 
> [image]
> 
> [EDIT] I found why it fails. From command line, "--capture-preview" works one time, and "-D" or not, a 2nd call never returns. (have to ctrl-C). This is disapointing, as this script is brilliant, but it seems it is not usable with the Nikon D5100. After some googling, it seems it's a common issue.

On the sony, the captured image will stay in the cameras buffer in ram untill gphoto deletes it. I assume the "normal capture" would download the image then delete. I'm not downloading it so I have to manually delete the file. If your camera can do video, live view on the lcd, or silent... something where the mirror stays locked up, I would imagine that capture preview would work much better for you.

yet-another-average-joe, 2019-04-28:
Thanks for the answer.

Unfortunately, on the Nikon D5100, LiveView is disabled as soon as the USB connection is detected, even if LiveView is set before pluging the USB connector. In quiet mode the mirror does not stay locked up. It goes up when pushing the shutter button, and down when releasing it. It's not a real silent mode, it's a fake silent mode. And there's no real video mode (the DSLR must be in LiveView mode, and a red button be pushed). The mirror can be locked up for sensor cleaning, but it can't shoot pictures in this mode. It's an old and low end DSLR !

But there's a workaround : the D5100 (and probably many other DSLRs) must be set to RAW + JPG, and --keep-raw used in the script ; therefore, capturetarget=1 is not needed, (and doing so I'm not off-topic !). It will keep NEF (raw) pictures on the DSLR. I still have to see if setting the Nikon for low quality JPG speeds up the transfers, playing with timeout value. It's some kind of preview, if the files are really downloaded at low resolution (not sure...). This could mimic the Sony behaviour.

FormerLurker, 2019-04-28:
@xxsteven69xx, for some reason digicam control seems to be broken on my PC.  An uninstall+reinstall didn't help, so I'm currently stuck.  I've been getting lots of DSLR support requests recently and am trying to come up with some general solutions, like in app testing, which should help debugging problems like this.  I'll try to reply on all of the DSLR support threads if I come up with anything.

xxsteven69xx, 2019-04-29:
@FormerLurker Cool I've lost hope for windows and have converted an old 2011 laptop to ubuntu for octoprint ( I don't already own a raspberry pi) and have been ok with this solution. I would like to use my lattepanda mounted to the frame ( this would give me back half of my desk space footprint) but for now it has to keep windows 10 :(  I've been kicking around the idea of just wiring up one of my unused pins on the minirambo to a remote trigger and using 
M42 P30 S255 ;  turn on pin 30 on ramps
G4 P500 ; wait .5 secs
M42 P30 S0 ; turn off pin 30
G4 P5000 ; wait 5secs for camera capture

The only thing is moving the print head correctly. Would octolapse be able to execute gcode (specificaly M42 commands)

FormerLurker, 2019-04-29:
@xxsteven69xx,  you can send gcodes via camera profile settings, but they get sent AFTER the scripts are executed at the moment (I plan on adding before/after snapshot gcode soon though...)  Currently you will need to use scripts to trigger your GPIO pins.  I've been working with another user [in this issue](https://github.com/FormerLurker/Octolapse/issues/334), but haven't been able to figure out the problem (it seems to work on my printer).  Maybe if you jump in on that thread we can make some progress.

yet-another-average-joe, 2019-05-01:
Hi

Got your idea to work with my D5100, but with a slightly different aproach (otherwise the USB connection is stuck after 1 image). I had to use gphoto2 in shell mode, and there is no need for "capturetarget". But it needs a command list file...

The script :

```
#!/bin/sh

# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# IMPORTANT - You must add gphoto2 to your /etc/sudoers file in order to execute gphoto2 without sudo
# otherwise the following line will fail.

#sudo gphoto2 --capture-image-and-download --keep-raw --filename "${SNAPSHOT_FULL_PATH}"

#gphoto2 --capture-image && gphoto2 -D && gphoto2 --capture-preview --filename "${SNAPSHOT_FULL_PATH}" && (cd $4 && mv thumb_$5 $5)

# the only way I can get previews from my Nikon D5100 !
# gphoto2 in shell mode with a command file, cmd.txt ; must be in scripts folder
# QUICK AND DIRTY !
# capture_preview.jpg should be deleted if already exists !

cd /home/pi/scripts
sudo gphoto2 --shell < cmd.txt
mv capture_preview.jpg $6

# save snapshot to SD card, download preview
#sudo gphoto2 --capture-image && sudo gphoto2 -D && sudo gphoto2 --capture-preview --filename "${SNAPSHOT_FULL_PATH}"

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_PATH}'." >&2 
  exit 1
fi
```

The command file (cmd.txt). It has to be created in the scripts folder :

```
set-config /main/actions/viewfinder=1
capture-image
capture-preview
q
```

@xxsteven69xx : it does exactly what does your script. No capturetarget trick, no image download but the preview, and images stay on the SD card for further processing. Could you give it a try and let me know if it works for you ? (pretty sure it will). Please let me know if the command "set-config /main/actions/viewfinder=1" makes your camera hangry, as it probably does not exist on a mirrorless camera !

FormerLurker, 2019-10-07:
Hey, I have finally gotten things to work again with digicamcontrol for windows!  I know this is an old thread, and that most of these problems have been since dealt with, but I wanted to mention this.  I'm going to go ahead and close this thread, but if anyone wants it reopened, or would like me to add a new script example to the wiki, please let me know!  Thanks again for posting everyone!

closed 2019-10-07 by FormerLurker

epvasquez, 2021-02-05:
Hi, I´ve got a Notebook with an old version of win 10, digicam software for my Canon EOS Rebel t100 DSLR. (digicam software working), Octoprint working, But I can´t been able to conect my DSLR cam to Octoprint...  can anyone help me with is matter?

I've seen much info for rasperry pi and gphoto2 , but  my notebook has windows, not linux..

FormerLurker, 2021-02-05:
You should be able to get digicam control working. It has a snapshot url you can use in Octolapse.  However, this issue is closed, and unrelated to your post.