FormerLurker/Octolapse #332: Octolapse reports that rendering has completed, but the timelapse is not created

Jyckle, 2019-04-01:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Ran a test print with debugging enabled (after failing with several normal prints)
2. Received a notification that the timelapse had rendered and finished syncing
3. Attempted to open the timelapse
4. Discovered that the timelapse was an empty text file despite having the extension .mp4
5. Repeated the process with several different settings and got the same results
6. Looked at the logs to attempt to find the problem
7. Found the command to generate the timelapse
8. SSH'ed into my pi and ran the command manually
9. Took the file (which was not previously present) and moved it to the octopi timelapse directory
10. Downloaded and viewed the file successfully
11. Looked at settings but have found no way to fix the issue as octolapse doesn't appear to know that there is an issue. 

#### What should have happened?
Timelapse should have been generated successfully

#### What happened instead?
Empty text file was created instead

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.15.0

#### Printer model & used firmware incl. version

Printer Model: Prusa Mk3
Printer Firmware Version: 3_6_0

#### Browser and version of browser, operating system running browser

Browser: Chromium/Chrome
Browser OS: Ubuntu/Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: Tried several different files

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/t2adg4fW

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/Jyckle/d0635c9877d7fc72342f89181aa15055

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/Jyckle/6a884f9c2a16192880ed1a7f6c2e76ec

#### Link to contents of Javascript console in the browser

Link to javascript console output: Don't think this is needed

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

Wasn't sure where to put other comments, but just wanted to say thanks for this plugin! I'm hoping one day when my life gets a little less crazy I can contribute to this project, because it's something I use all the time! And when this works it's amazing. My issue is just with the generation of the timelapse, and I've tried over and over again to get it working so I was hoping someone might have an idea what's going on. Thanks again for the awesome work on this plugin and hopefully someday I'll get to help contribute to it!

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-04-01:
Well this is something new!  You don't happen to have the MP4 that is actually a text file still do you?  Octolapse reported no errors because avconv didn't report any errors unfortunately.

FormerLurker, 2019-04-01:
Clicked the wrong button, whoops.

Jyckle, 2019-04-01:
My thoughts exactly. Not sure how to address an error when there's no reporting on the error! But yes I do still have a couple of the mp4 files. They show up in the web interface as 0.0B also. Unfortunately neither github nor dropbox will let me upload the files, so here's a screenshot of them in my file system. 
[image]

Jyckle, 2019-04-01:
One other note, they are all zero lines/zero characters which makes me think that when octolapse attempted to copy the file from the octolapse directory to octoprint's timelapse directory, it created a new empty file because the file didn't exist in the octolapse directory.

FormerLurker, 2019-04-01:
@Jyckle, thanks for the extra info, and for your compliments earlier :)  I'll do a bit of digging and see if I can come up with a theory as to why it's not working and eventually a solution.  If a code change is needed I may ask that you install a development version for testing, but hopefully it won't come to that!

Jyckle, 2019-04-01:
Awesome, I really appreciate it @FormerLurker! And the compliments are definitely well-deserved with all the work you put into this! And sounds great! If you do need me to do any testing I'm more than happy to help with that as well :)

Jyckle, 2019-04-04:
After a whole bunch more messing around, and completely reinstalling octopi/the plugins I use, I figured out the issue! It actually had nothing to do with octolapse itself, rather with the combination of that and the polar cloud plugin I was using. The polar cloud plugin requires GStreamer, which apparently messes with the way octolapse works. I had octolapse working, installed GStreamer and it broke, and then uninstalled GStreamer and it works perfectly again! Silly that it works that way, but figured it out nonetheless! Thank you for the help regardless! I'll mark this as closed!

closed 2019-04-04 by Jyckle

FormerLurker, 2019-04-04:
@Jyckle, that is very strange!  Thanks for your debugging efforts.  If possible I'll dig into this and try to figure out why avconv and gstreamer aren't playing well together.