FormerLurker/Octolapse #333: Controling camera settings

jonas740, 2019-04-07:
I have searched and searched to find answers to my question.

And that is how to get control over my Camera Module inside Octolapse.

I found that i should change 

#camera_http_webroot="./www-octopi"

#camera_http_options="-n"

to this:

camera_http_webroot="./www"

camera_http_options=""

This is inside "boot/octopi.txt". 

But the problem is that i do not have an Octopi version of octoprint. I have it installed on Raspbian Stretch with desktop as a venv. According to this "https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337"

How do i find thoose settings with this install ?

FormerLurker, 2019-04-11:
Hey there @jonas740!  I believe we've been introduced :)  Sorry about being late to respond, but in my mind I equated the reply in the octoprint discourse page with this issue.  I think this is the better place to deal with it.

As you may have read, I'm finally getting a raspicam to figure this out.  My suspicion is that you have done everything correctly, but that mjpegstreamer does not support control on the raspicam in the same way as it does with regular USB webcams.  We will know for sure in a couple days.  If it's not compatible I'll look into some other ways of accomplishing the same thing.

You mentioned in the discourse pages that Octolapse allowed you to enable the custom image preference page, which is probably because all it does it check to make sure you are streaming from mjpegstreamer.  Apparently that is not enough since perhaps not all modules support control.htm.  This is the first time I've heard of this, so thanks for bringing it up.

FormerLurker, 2019-10-07:
@jonas740, You'll be happy to know that the raspberry pi camera V2 is now fully supported by the next version of Octolapse (though there is a bit of setup).  Let me know if you want to test the alpha version.  I'm going to mark this issue as 'waiting for response'.

Thanks for posting here!

qwertpi, 2019-10-07:
> @jonas740, You'll be happy to know that the raspberry pi camera V2 is now fully supported by the next version of Octolapse (though there is a bit of setup). Let me know if you want to test the alpha version. I'm going to mark this issue as 'waiting for response'.
> 
> Thanks for posting here!

I'd be interested in this alpha version if it isn't the one you've already sent me

FormerLurker, 2019-10-07:
@qwertpi, I believe I have sent it to you, but just in case you can install it via this link:

https://github.com/FormerLurker/Octolapse/archive/devel.zip

I'm pretty sure there are changes/bug fixes in there since the last time.  [Here are instructions](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) to get started with the new version, and [here is setup info for the raspi cam](https://github.com/FormerLurker/Octolapse/wiki/Configuring-a-Raspberry-Pi-Camera).  Let me know how it goes!

qwertpi, 2019-10-07:
> @qwertpi, I believe I have sent it to you, but just in case you can install it via this link:
>
> https://github.com/FormerLurker/Octolapse/archive/devel.zip
>
> I'm pretty sure there are changes/bug [...]

Thanks, when I get a block of free time I'm gonna systematically work through the possible causes of my camera problems and then I'll take a look at this.

FormerLurker, 2019-11-09:
@jonas740  and @qwertpi, It's been over 30 days since this thread has been active, so I'm going to go ahead and close it.  If you want me to reopen it, just let me know, or you can just open a new ticket.  Thanks!

closed 2019-11-09 by FormerLurker