FormerLurker/Octolapse #334: Use a LED as "flash" for the snapshots

sqweezr, 2019-04-11:
I am fairly new to scripting and all, so i hope you can help me out.
The plan is to fire a strip of LED's just before the snapshot and turn them off after the snapshot, so i found the pre an post snapshot scripts. 

The thing is that i thought i could use Gcode in there, so i studied for a day or 2 how to use a auxilary pin on my ramps 1.4 to control the LED's using a M42 gcode command (M42 P4 S255 to turn them on and M42 P4 S0 to turn them back off).

But it turns out i cannot use gcode as a script.......and now i am stuck :)
There is not much to be found on these scripts so i really hope you can help me out!

Kind regards,

Koen

FormerLurker, 2019-04-12:
Thanks for posting Koen!  I added formatting to your scripts so that it is a bit easier to read, I hope you don't mind.

sqweezr, 2019-04-26:
> Thanks for posting Koen! I added formatting to your scripts so that it is a bit easier to read, I hope you don't mind.

Off course not! Looks way better now :).

I tried the latest version of octopi and still the same problem........Do you have any more options?

MineThingIssues, 2019-07-22:
Looks like a really interesting idea! Would you mind telling me what error you are getting? Does your script work when run on its own in the terminal?

I have always used GPIO.output(21, HIGH(or LOW))
But if true/false works then there shouldn't be an issue there. Make sure to also do GPIO.cleanup() at the end of your turn off script to make sure the pins are reset and prevent future errors.

HeneryH, 2020-06-23:
I got came up with these commands through various sources but then I execute the script, the LED only stays on until the script completes then it goes off.  

I have to put a timer at the end for x seconds to stay on for x secs.  This is actually good for me since that is the behavior I want, but can someone explain why that is?

I'm guessing the pin has a default state of "off" and when the script completes the pin goes to default regardless of how the script left it.

`import time

from gpiozero import OutputDevice

GPIO_PIN = 27  # Which GPIO pin you're using to control the LED.
SLEEP_INTERVAL = 5  # (seconds) How often we check the core temperature.

led = OutputDevice(GPIO_PIN)
led.on()
time.sleep(SLEEP_INTERVAL)
`

FormerLurker, 2020-06-23:
Maybe try the [LED module](https://gpiozero.readthedocs.io/en/stable/migrating_from_rpigpio.html#output-devices)?  Not sure if you will need to disable warnings or not...

veldriph, 2021-01-03:
look at relay No.4
startup raspberry pi

起動直後
[image]

push [test] Before Snapshot Script
[image]

push [test] After Snapshot Script
[image]

setup
[image]

files
[image]

vi light-off.py

#!/usr/bin/python
# -*- coding: utf-8 -*-
import RPi.GPIO as GPIO
import time
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(25,GPIO.OUT)
GPIO.output(25,False)
time.sleep(0.1)

vi light-on.py

#!/usr/bin/python
# -*- coding: utf-8 -*-
import RPi.GPIO as GPIO
import time
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
time.sleep(0.1)
GPIO.setup(25,GPIO.OUT)
GPIO.output(25,True)
GPIO.cleanup()

light-on.sh

#!/bin/sh
python /home/pi/oprint/lib/python2.7/site-packages/octoprint_octolapse/test/light-on.py

light-off.sh

#!/bin/sh
python /home/pi/oprint/lib/python2.7/site-packages/octoprint_octolapse/test/light-off.py

veldriph, 2021-01-03:
time.sleep(0.1)

this time is not tuning.

13rtn, 2022-05-02:
Hi everyone, I need some help. I followed the procedure for creating the scripts step by step, the problem is that I get an error from octolapse when I go to test the script. in the log file I find this: 

_2022-05-02 10: 28: 01,314 - octolapse.script - ERROR - Error output (stderr) for 'Logitech c920 - Before Snapshot Camera Script':
[sudo] password for pi:_

I can't figure out where the problem is, the password inside the .sh file is correct

Thanks for your help