FormerLurker/Octolapse #335: Camera Script Not Working - Cant Make folder

TheNore, 2019-04-12:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: .0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

Will not take photo in octoprint/octolapse plugin. diagnosis run from SSH commands
1.Created DSLR script exactly like Wiki
2. Redid walk through twice even with echo manual Password in snapshot script 
3. Added new take-snapshot file to /etc/sudoers.d/ folder with correct permission text - same result

#### What should have happened?

New file is in location /capt0000.jpg on the camera
Saving file as /home/pi/scripts/test.jpg
Deleting file /capt0000.jpg on the camera

#### What happened instead?
sudo /home/pi/scripts/take-snapshot.sh
Creating directory:
mkdir: cannot create directory ‘’: No such file or directory
New file is in location /capt0000.jpg on the camera
File capt0000.jpg exists. Overwrite? [y|n] y
Saving file as capt0000.jpg
Deleting file /capt0000.jpg on the camera
The snapshot was not found in the expected directory: ''.

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian Stretch
Os Version: 9.3

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3s+MMU2s
Printer Firmware Version: 3.7.0

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: happens on multiple gcode and in ssh

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/TheNore/50d7376578fe3d00762d1d8a00683df4#file-gistfile1-txt

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/TheNore/144e13cbb7c40290785ae1dede537d6b

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-04-12:
Thanks for posting an issue!  I have a few questions.

1,  Are you running OctoPi?  If not, the directory names will be wrong in that sample command.
2.  You noted above that you ran this command:  ```sudo /home/pi/scripts/take-snapshot.sh```  However, this is the correct test command: ```/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"```  Are you sure you included the parameters after take-snapshot.sh?  If not the script definitely won't work.
3.  Are you sure you included this as the first line of your script: ```#!/bin/sh```
4.  Would you take a screencap of your terminal after running the above test command?  That might be helpful.
5.  What camera do you have?

Thanks!

TheNore, 2019-04-12:
Thank you for the fast response, sorry I have used the forward slash / in
execution in octoprint and command line. I just didn't reference them in
the ticket.
I custom built my Octoprint build, i did not use the prebuilt image.
I am using a canon Rebel 550D (t2i) and it is fully supported in the
gphoto2 documentation

Should i be using the full script syntrax above in octolapse, or just the
/home/pi/scripts/take-snapshot.sh

[image: image.png]

TheNore, 2019-04-12:
when triggering the scrip tin octolapse i get the following error 

There were 1 errors while applying custom camera settings/scripts:
camera_initialization_error: The custom camera initialization script failed with the following error message: mkdir: cannot create directory ‘’: No such file or directory
ERROR: Could not get image.
The snapshot was not found in the expected directory: ''.

FormerLurker, 2019-04-12:
You should be using the entire script.  That stuff after take-snapshot.sh is critically important, and is telling the script where to put the snapshot.  However, keep in mind that you're only testing the script there, and that's not the way you should enter it into your Octolapse camera settings.  I'll make this clear below.

Your other issue is that you are putting your script in the wrong place.  Remove the script path from the 'Before Print Start Script' setting and put ```/home/pi/scripts/take-snapshot.sh` in the 'Snapshot Acquire Script' setting like so:

[image]

Make sure you have 'external camera - script' selected in the camera type drop down too.

It might also be a good idea to re-read the wiki page since a few details were overlooked.  Be sure to let me know if anything is unclear, or if I can do anything to make it better.

Thanks!

TheNore, 2019-04-12:
I am Sorry for the confusion, I am trying to diagnose the issue and provide all information. I did not realize that I couldn't test the script in octolapse before committing to a print. it takes time to run a test print. below are the settings I have at the time o

f documenting the issue. As are the logs i submitted earlier.[image]

FormerLurker, 2019-04-12:
Hey, no worries at all!  I need to work on a better way to test these scripts, as you probably noticed.

If you do a test print, just use something very small like a calibration tower or some kind.  You should only have to print a layer or two.  If you run in test mode you won't have to wait as long or waste plastic.

One last thing.  Have you seen the video guide for this process?  If not, [it can be found here](https://www.youtube.com/watch?v=J5SskYwP5JQ).  That might make it a bit more clear as to what to expect.

Keep me posted.  I'm happy to help!

TheNore, 2019-04-12:
Ok... Honestly I feel like i should have tired this from the beginning. I upped the 4000ms wait time to 5000ms and boom. The script and photos executed perfectly, I just tried on 3x test prints. 
From the errors I was getting it seemed like a permissions issue and I chased that ;/ 
Then I went back to recheck everything like you said, and upped the time to 1 more second. and it completed successfully. Sorry for all of the confusion along the way, I should have figured this out solo.

FormerLurker, 2019-04-12:
No problem!  5 seconds to take a photo is pretty steep.  Fair warning: you'll probably have some print quality issues because of that.  However, you can greatly reduce that by deferring download of the snapshots until the print is complete.  I have created a beta guide for that process which can be found [here](https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download).  More warnings:  This guide is new so there might be some mistakes or omissions.  Also, the script includes a section that will DELETE ALL IMAGES ON YOUR SD CARD, and is not compatible with all cameras.  On the bright side, if it works for you the process will take much less time and your prints will turn out better.  On the negative side you won't see any snapshots in the Octolapse tab at all, since they will not be downloaded to the pi until later.

TheNore, 2019-04-12:
That is a good point.
Has there been any other situations where the script fails due to the
trigger being too short?
I will definitely look into the new walk through. on the first small test
print i did with 10 sec pauses, print quality was definitely an issue.

FormerLurker, 2019-04-12:
Have you had any similar quality issues using a webcam with Octolapse?  If so, it's best to work those out first, since fixing it for longer delays is more difficult.  Also, when using a DSLR make sure your snapshot delay is set to 0 (the timeout can be higher).

If the timeout is too short, Octolapse will kill the process and continue, so you'll miss some snapshots but your print shouldn't fail.  Also, turn your resolution down since a bigger file takes longer to download, and 22 megapixels (or whatever your max resolution is) is a bit overkill for a timelapse :)  Similarly the jpeg quality probably doesn't need to be super high since you're just going to render a compressed video anyway.  Anything you can do to reduce the size of the image will cut down on the snapshot time.  Also, your exposure will affect snapshot time too, so more light + lower exposure = less time.

I'm working on some new low impact stabilizations that have improved quality a LOT for me (bordering on 0 impact, even for vase mode in some cases), but they aren't ready yet :(

TheNore, 2019-04-12:
Perfect. This is what I was going to get into next - I have the camera in full manual mode, but I will turn down the Capture resolution, and tweak the WB and ISO settings. I had thought this was an issue before, but I wanted to iron out any other problems.

I have had flawless experience with print quality on my octolapse (400+) time lapse videos done with my Logitech C920.  then DSLR option is just incredible because i have many lenses and dolly systems that would be perfect for time-lapse 3D prints, especially with the MMU. So thank you very much for you work on expanding the capabilities of this plugin 👍

FormerLurker, 2019-04-12:
My pleasure!  I'm very glad you have had success with the plugin.  Are you putting your timelapses online somewhere?  If so please shoot me a link.  I love seeing what's out there.  

If you have any issues with the deferred download script let me know, though it might be better to create a new issue.  I have another 'no download' guide, but you have to manually render your timelapse with that one.

I'll leave this issue open for now in case you run into any unexpected issues getting your current config working, or if you just want to post your results or suggest improvements.

TheNore, 2019-04-15:
successful first time lapse print! thank you for your help. I can only get it to work on "layer change - compatibility" for some reason, but the quality is not bad at all! any downside to this? 

https://youtu.be/LU3vB8E_SrY

FormerLurker, 2019-04-15:
Awesome, thanks for posting!  Believe it or not, that print is the very first timelapse I ever did (before Octolapse was even thought of).  [Here is a link](https://www.youtube.com/watch?v=i7ATXC3fiwQ) to that if you're interested.

If your quality is good there is no downside to using compatibility mode.  Feature detection, which is what is added when you choose 'High Quality' mode, is not very easy to configure properly, and can be finicky (especially if you hit the 'minimum layer time' cooling setting, then it just won't work).  I'm probably going to remove it from the next release and replace it with a new much easier to configure option that should give equivalent results in most cases, and much better results in others.  I just completed this option this weekend and finished my first test print.  Lots of stuff to do still, but it's coming along nicely :)

TheNore, 2019-04-16:
Ha! that is so cool. great minds ;p Thanks for the information on the
settings, that makes sense. I am also using Slic3r PE 1.42.0 Beta1 so that
might have something to do with it also.
If you need anything tested I wouldn't mind being a beta tester if you
needed additional info. thanks again for your help :)

MineThingIssues, 2019-07-20:
I understand that this issue has now been resolved, but I would just like to mention that when I first started using coctolapse with a DSLR I had a very similar issue. I was using a much older camera but very high file size and even 5 seconds was not enough. I ended up spending 36 hours printing something that should only have taken 6 hours (for multiple reasons, not just octolapse).

Anyway, I fixed it with the deferred download script (which came with its own issues), and now my time-lapses are flawless! (apart from having to change the batteries every few hours).

FormerLurker, 2019-10-07:
So, the deferred download script and no-download script instructions have finally been updated to cover the snapshot target issues lots of people were having.  I'm going to go ahead and close this issue, but if anyone wants me to reopen it, please let me know.  Thank you all for taking time to post here!

closed 2019-10-07 by FormerLurker