FormerLurker/Octolapse #336: Code M81 (with IP address) causes Octolapse to fail

jkkato, 2019-04-12:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

Note I have Octolapse installed on my other printer and I don't use it much but it does work without issues. I did have to change a few settings on my new printer to get it to work as I wanted it (with my cura settings), like absolute positioning and priming height, but now I am at the following point.
1) I slice and start the print as normal with Octolapse on.
2) Just at the end, during my ending gcode, Octolapse errors. This causes Octoprint to think the print "failed" at the final line of gcode, but the print is finished. Octoprint web page is unresponsive.

#### What should have happened?
After the print finishes, the last two lines of my ending code set the hot end temperature to M109 70 and waits until the temperature is reached.  Then a M81 192.168.1.84 (turn off the power supply) command is sent to the Wemo Smart switch plugin to turn off the printer power via the smart switch at a certain IP address. End of file.
That is the last line of the .gcode file. I have been using the same setup for 3 months no problem.

#### What happened instead?
1) On the very last line in the code, when the command to turn off the printer is sent, Octolapse errors (An error occurred while parsing the command string.  Details: Cannot parse the gcode address, multiple periods seen.), even though the print did finish properly.
2) The M81 code is properly executed and the printer power (Wemo Switch) turns off as designed.
3) but Octolapse causes Octoprint to fail and basically freeze at the last command. The Web interface is stopped and the print job is shown as not finished. 
4) Octolapse does complete the timelapse video, but indicates it as a FAILED, and does not add the proper Post Roll (with the print head parked out of the way).
5) Other plugin's do not finish, like Pushbullet, because octoprint is frozen.
6) I have to reload the Octoprint page, as it does not respond to any mouse input or scrolling.
7) After reloading the page, Octoprint functions as normal.
I tried to upload the full Plugin_Octolapse.log files, but it is huge. I cut the last few snapshots of the log file. I will also upload the last cgode file (I think) I ran, but I had this same problem with three different files.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: Chrome Version 73.0.3683.86 (Official Build) (64-bit)
Browser OS: Windows 10 (also happens on Samsung S9 Edge and Chrome)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: 
[CE3_OctoLapse test.gcode.txt](https://github.com/FormerLurker/Octolapse/files/3074242/CE3_OctoLapse.test.gcode.txt)

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [octoprint.log](https://github.com/FormerLurker/Octolapse/files/3074202/octoprint.log)

#### Link to octoprint.log

Link to octoprint.log:  
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/3074245/octoprint.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-04-12:
First, thanks for posting all of the required info!  It will be critical in solving this issue.

M81 was not successfully parsed, though it looks like that gcode is being used in a rather unconventional way, I'm guessing by design from the wemoswitch plugin.

Octolapse was designed to cancel the print if this sort of error occurs, but ONLY in test mode (obviously I need to document this better).  If this happened during a live print, Octolapse would just ignore the error (after logging it as a warning), and would have allowed the gcode to be sent to the printer.

However, I see this error too in your Octoprint log file:
```
2019-04-11 17:18:12,069 - octoprint - ERROR - Exception on /api/plugin/wemoswitch [POST]
Traceback (most recent call last):
  ... removed for brevity - the info here isn't really useful for us anyway
ConnectionError: HTTPConnectionPool(host='192.168.1.84', port=80): Max retries exceeded with url: /setup.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x625eb390>: Failed to establish a new connection: [Errno 111] Connection refused',))
```

This indicates to me that something is up with your power strip.

Regarding the Octolapse parser, I've completely rewritten it in C++, and will make sure I test this command to make sure it doesn't cause problems in the next release.

Thanks for reporting your issue.  Let me know if you feel there's anything else I could help with.

jkkato, 2019-04-12:
Thanks for the quick reply. 

The M81 command gets intercepted by the Wemo switch plug in, and I IP Address is because you can have multiple switches (I have two). I think the M81 gets sent to the printer too, but gets ignored (not sure). This Wemo switch is only connected to the Ender 3 power supply. The RPi and all my other accessories are not plugged in to the switch. It is supposed to disconnect octoprint from the printer communication beforehand. Maybe there is a timing issue that comes up when Octolapse sees the extra "." of the IP Address. 

Octolapse does pop up an error message about too many ".", during a regular live print. At least I am 99% sure it was on live print. I will double check. I was "printing" in test mode. Then switched to live. My last pring was live with data logging (I forget the exact name), after two "failed" Octolapse prints. I don't have the exact wording of the error right now, but I think I cut and pasted it in my description.

I didn't look at the octoprint log, just the Octolapse log.

FormerLurker, 2019-04-12:
You will get this error popup if there are problems parsing any gcodes during a live print:

There was an error parsing your gcode.  Stopping Timelapse

That will be followed by the error details, which are not important.  In this case the timelapse is stopped (no further snapshots taken).  I'm thinking I should reconsider that.  I'm planning to reduce the number of popups and move errors/warnings like this somewhere else, but I haven't decided exactly how yet.

However, I'm still not sure why an exception was thrown in the wemo plugin, or why Octoprint locked up (it shouldn't have).  I'm especially confused regarding the timelapse video that's generated.  If Octoprint really is locked up, I'm not sure how could Octolapse generate a timelapse.

jkkato, 2019-04-12:
To be more clear, I don't think octoprint itself locked up, just the web interface/sever instance. All I seem to be able to do is reload the webpage. When it reloads the print job is stopped and the percentage bar is at ~98%(depending on the print job). 

On the other hand, the Wemo switch does turn off. I thought that the Timelapse did render right away too, but now I am not 100% sure. When I get home I will compare time Stamps from the render, when it gets transferred to my Dropbox (another plugin) and the time of the snapshot (since I have a clock beside the printer in the frame). To see if it renders and uploads right away when while the interface is not working...

jkkato, 2019-04-13:
I restarted Octoprint and tried again on the "No Logging" Mode. It seemed to work. An error message popped up, but the printed part finished as it was suppose to and it generated the Timelapse (without the "Failed" prefix), but did not take the final snapshot.

I tried to print the same file again on "Diagnostic Live Print" mode. I got the same error message, and the printed part did finish, but Octoprint stopped. It looked like the print job just cancelled. It generated the timelapse right away with the "Failed" prefix.

Then I tried another print file with "No Logging" mode. Again the error message, the printed part finished, but the Octoprint interface was not working properly. It said 73% complete.
[image]
 I could switch tabs, but the buttons didn't work, like pause, cancel or print. The temperature graph stopped. It said that it was connected to the printer, but there was nothing on the terminal feed. I took a couple of videos, but they are too big to upload here. It did render the timelapse right away, and added the "Failed" prefix. It is puzzling that the interface locked up a lot earlier, but still finished the print job and rendered the video. Also, it was in "No Logging" mode, but it still got marked as "Failed".

I am trying again with the "No Logging" mode right now. I might try again with Octolapse off after, before heading to bed.

jkkato, 2019-04-14:
So it looks like there may be two issues, and the freezing interface may be not related to Octolapse.

The last print with "No Logging", also seemed to just "failed" at the last line, quit the print job and the video render was prefixed with failed. Similar to the second attempt in the previous post. The interface was fine. 

I then ran two longer prints with Octolapse not running (but still enabled in the plugin manager). The first print printed without any issue at all. The second print the interface froze up at 68%, but again octoprint was still working in the background and finished without issues. So it is likely that the freezing interface is unrelated to Octolapse (although Octolapse was still enabled but not recording), and it was just by chance that the interface froze at the end of the prints.

I may try a few prints without the M81 command at the end and Octolapse on.

FormerLurker, 2019-04-14:
@jkkato, thanks for all of your testing.  I recommend disabling your other plugins, and re-enabling them one at a time until you can figure out which one is causing the lockup.  It could be OctoPrint too.  If so, you'll see the same thing running with all plugins disabled (or better yet, in safe mode).

Regarding Octolapse, if it is disabled, the odds of having a problem are very low.  There is very minimal interaction if Octolapse is disabled.  If errors do occur you should see errors in either octoprint.log or plugin_octolapse.log, but it is unlikely.

Regarding the 'failed' message caused by the IP address in M81, for now just ignore that.  You adjust the naming template in the rendering settings to get rid of the annoying prefix for now.  Change the default of:

```{FAILEDFLAG}{FAILEDSEPARATOR}{GCODEFILENAME}_{PRINTENDTIME}```

to

```{GCODEFILENAME}_{PRINTENDTIME}```

within the rendering profile under 'advanced settings' here:

[image]

Again, thanks for your testing.  I'll try my best to fix that parsing issue, or at least prevent an entire timelapse cancel, before the next release.

FormerLurker, 2019-06-03:
Hey, did this help at all?  I'm about at the point where I can get the parsing of M81 with an IP to work, and can shoot you instructions to upgrade to the development version for a test run if you'd like.  I'm hoping to get it finished in a week or two.  I've been saying that for a while, but this time I have a really good feeling!

jkkato, 2019-06-05:
I had turned Octoprint off for a while. I tried it with two prints (and the M81 turned off), just now. For some reason the camera is taking photos in random positions, in between photos in the parked position. Happened on both Prints. I don't think I changed any settings from when it was taking proper photos before. I will have to look at this issue.

Do you want me to run Octoprint with the M81 command and see what happens?

jkkato, 2019-06-05:
https://www.dropbox.com/s/yvgrozivjw27i0f/CE3_Bear_wTree_20190604020759.mp4?dl=0
Heres a Video of the random photos. I had set it to go to the fixed coordinate 10 x 140 mm, which is about half way on the side.

FormerLurker, 2019-06-06:
Hmm, i saw this just today from another user.  Hopefully it isn't a trend.

FormerLurker, 2019-06-06:
If you would be willing to run a print with the 'full diagnostic - live print' setting and shoot me the log, I will try to see what is going on.

Regarding M81, the fix is not in the live version yet.  If you want to try it, ill let you know when I've got it stable enough for testing.

jkkato, 2019-06-06:
Here is a log file. I hope it is what you are looking for. (it maybe still uploading)
https://www.dropbox.com/s/lngcwfu7d28cxdd/plugin_octolapse%20%28Debug%20Live%29.log?dl=0

Here is the resulting video.
https://www.dropbox.com/s/uzcnz7ot60hh66w/CE3_Pole_Base_Foot_20190606070511.mp4?dl=0

FormerLurker, 2019-06-06:
Ug..  From the log everything looks fine.  Each snapshot takes about the same amount of time (around 2.2-2.6 seconds), and none look odd in any way.  I looked every snapshot taken, and your printer is reporting that it's at the following position for every snapshot:  2019-06-06 05:53:34,559 Received from printer: line:X:10.00Y:140.00.  I'm not used to seeing M400 respond with an 'OK' immeidately after being sent.  My printer will respond with 'echo:busy: processing' until all movements are completed, then it will respond with 'OK' after the movement is finished.  Your printer seems to respond to the M400 with an OK, then I send an M114 (request current position) and it waits for a bit (around 2 seconds for most of your snapshots), then responds with the position.  It's possible that the M114 is coming back before the movement is finished, but I would expect to see some obvious differences in timing, but I'm not.  However, I cannot rule it out without running some tests on the actual hardware.  I can say I've had lots of issues with the Ender 3 specifically here recently (I've asked the MFG to send me a test unit, but they haven't gotten back to me yet), so perhaps it has something to do with the firmware.

Perhaps you could try this test for me since I don't have access to an Ender 3:  Open the Octoprint terminal and enter the following (with the bed clear and the extruder up off the bed at least several MM):
```
G28 X Y; Home X and Y
G1 F500; set the axis speed so that it's extremely slow
```
Then enter this command, which will move your print head to the snapshot position you configured:
```
G1 X10.000 Y140.000
```
Now, as quickly as you can without making mistakes (you will have a couple seconds) enter these two commands one at a time:
```
M400
M114
```
Then let me know how it responds.  I expect you'll get an OK from M400 immediately (which is odd), but the M114 will respond with the position ( something like X:10.00Y:140.00 + other stuff) AFTER the movement completes.  Maybe try all of the code 4 or 5 times (you can increase the feedrate command G1 F500 to something higher if it's too painful to wait) and see if the results are consistent?

Another possibility is that the snapshots are somehow being cached, or are late for some reason.  Is there anything special about your mjpegstreamer stream, or is it the standard setup that came with OctoPi?  If it's being cached, I have some ideas about how to fix it, but unless you have some special setup on your Pi, I've never seen that happen before.  Is your framerate very low?  

Finally, you might go into the main settings (Octoprint Settings->Octolapse->Main Settings) and turn off 'Display real snapshot time'.  This adds several M400 commands, and some printers don't like this.  In fact, I've removed this setting from the next version because it can cause misbehavior in some printers that is firmware related.  It's possible this will solve the problem too.

Hopefully one of those things will work!

jkkato, 2019-06-08:
> Another possibility is that the snapshots are somehow being cached, or are late for some reason. Is there anything special about your mjpegstreamer stream, or is it the standard setup that came with [...]

I am using just the regular Raspberry Pi camera and the standard setup. [The Camera is Like this one](https://www.amazon.ca/dp/B07F73QPV8/ref=sspa_dk_detail_0?psc=1&pd_rd_i=B07F73QPV8)

I will look into the other stuff when I have a chance.

FormerLurker, 2019-06-08:
You don't happen to be using octoprint anywhere beta are you?  I just heard about a similar issue when using that.

jkkato, 2019-06-08:
Actually I did start using that recently. I think I installed it a week ago. Do you think it is related?

FormerLurker, 2019-06-09:
Yeah, the author has been in contact with me about it (v nice person).  An awesome feature was added to allow more efficient and higher quality streaming, but it adds (not verified) 200ms extra delay to snapshots.  You can work around it by adding 200MS extra camera delay in the octolapse camera profile to compensate for now (maybe more, you will need to experiment).  If your printer isnt dialed in, or if you are using some less viscous filament (petg), you may have some quality issues, but im hoping to find a better long term solution.  Let me know if that works for you.  Will leave this open so i remember to get back to you about any solutions we find.

flippedcracker, 2019-08-18:
So, I'm having a similar issue with my GCODE using an ip address is causing Octolapse to fail. I have a TPLINK plug that I have a light plugged in to and use an M80 xxx.xxx.xxx.xx command to turn on when a print starts, so that the camera has enough light. But Octolapse fails for having multiple decimals. Does the development version have any work done on that would fix this issue as well?

FormerLurker, 2019-08-18:
@flippedcracker, can you post a small gcode file somewhere?  I will try it out and will see if it works.

flippedcracker, 2019-08-19:
So, I'm having the TP-Link plugin turn on and off an outlet. So Octoprint is adding the Gcode to do that. I have :
`M80 192.XX.XX.XX`

as the starting Gcode script, and 

`M81 192.XX.XX.XX`

as the ending script. And this is the specific error I get: 

```
Octolapse Timelapse Stopped
There was an error parsing your gcode. Stopping Timelapse. Error Details: Cannot parse the gcode address, multiple periods seen.
```

FormerLurker, 2019-08-19:
@flippedcracker, it looks like this should work.  However, I feel like this command should have been implemented by the TP-Link plugin a bit differently.  It should have had some parameter code in front of the IP address, and sending this command to your printer might result in an error.

In any case, Octolapse will no longer stop the timelapse when encountering this command starting with V0.4.  Let me know if you'd be willing to test out the new version, and I can send you instructions for doing so.  Be sure to include an email address (which you will be free to delete after I respond to prevent spam due to email harvesting).

Thanks!

flippedcracker, 2019-08-19:
@FormerLurker It looks like they added another command to the TP-Link plugin to use @TPLINKON instead of M80. I should have used that. But since you've made this change, I'll test it out. You can send the instructions to flippedcracker+github@gmail.com . Thanks.

FormerLurker, 2019-10-07:
So, this has been resolved and tested in the next version of Octolapse!  Let me know if anyone would like to test it too.  I'm going to close this issue for now, since the primary problem has been resolved.

closed 2019-10-07 by FormerLurker

Tollbringer, 2020-01-21:
I am using the Tasmota plugin and I can not test octolapse without disabling the M81 command that turns off my printer (using 192.168.x.x) address command.

I would very much like to have auto-off and octolapse!

FormerLurker, 2020-01-21:
@Tollbringer, you are welcome to try out the alpha of v0.4.  I'm hoping to promote it to RC status soon.  If you're interested, you can use [these instructions](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse), and install using the devel link in the instructions (not the rc/devel link, it says 'Use this link ONLY if you've been asked to install the development version' fyi.).  Pay extra close attention to the section about automatic slicer settings configuration (it's the best feature maybe).  Please download your settings.json file (OctoPrint settings->Octolapse->Download Settings) BEFORE upgrading in case you  have any migration problems, and be sure to let me know how it goes!

Tollbringer, 2020-01-21:
Thanks for quick reply. I am afraid to try DEV as I am very new to OctoLapse and I feel I can't get the basics working right  (not sure what happened but the errors for the ip address have stopped. (I commented the command out and later put it back in).

FormerLurker, 2020-01-21:
> I am afraid to try DEV
I understand completely.  It's not as well vetted as the master.  However, there are lots of bug fixes in 0.4 that were never applied to the master branch.  I think most problems in the dev version are related to initial installation (settings/file migration, which is very tricky), rendering and scripting (DSLR/rendering bash and batch files), as well as some subtle UI bugs.  The portion of the program that actually drives the snapshots is quite stable.  It's the stuff before starting a print and after the print ends that seems to need some work.

Also, setup is much easier in V0.4.  If your printer has a profile ([see this link for a file list](https://github.com/FormerLurker/Octolapse-Profiles/tree/master/0.4.0rc1.dev3/printer)), you can just select your make and model and be done with that part.  If your printer is NOT in the list, let me know the make/model and shoot me a link to the MFG page, and I'll try to create one for you.  Automatic slicer settings detection is supported (as I mentioned in my previous email), which automates the tedious process of copying your settings manually.

If you want to try the dev version, I'd recommend installing it and trying it out on a few test objects (like a benchy or calibration cube), and disabling Octolapse for any prints that you actually care about, at least until you're comfortable with it.

Also, not sure how your error could have gone away, but i'm glad it did!  The previous parser didn't handle commands that have pure text parameters well (like M81).  I'll make sure to re-test this command in the new parser to ensure it will work for others in any case.

Tollbringer, 2020-01-21:
This is very much appreciated. I have stumbled  across a severe error that has me puzzled. My printer is pausing significantly since I have been playing with webcam's and timelapse. I am not sure if the error is OctoLapse or other- I tried a test print with octolapse (turned off); the problem persisted. I have removed the G4 P1 commands from the gcode scripts and I have now disabled ocotolapse to try an track down the issue. I did follow Teaching Tech's tutorial video if that means anything. If you know what the issue could be, shout it out please.

FormerLurker, 2020-01-21:
@Tollbringer, I'm not sure what you mean by 'Pausing Significantly'.  Can you describe this more completely?  Additionally, this seems like a separate issue.  Try this:

If your print works as expected with ALL third party plugins disabled (including Octolapse), try enabling ONLY Octolapse and trying again.  If the pause problem occurs, then create a new issue here, and we'll work it out.  It could be that G4 P1 is being interpreted by your printer's firmware differently than we expect.  It should wait for 1 millisecond, but perhaps that's not the case.  

Also, in V0.4 I've added a new suite of commands (@OCTOLAPSE TAKE-SNAPSHOT, @OCTOLAPSE STOP-SNAPSHOTS and @OCTOLAPSE START-SNAPSHOTS) that can be used to initiate snapshots via gcode.  These commands will  never be sent to your printer through OctoPrint.  However, the new 'Smart Layer Trigger', which is available starting in V0.4, is generally superior to the gcode triggers, since they try to minimize travel distance and artifacts by choosing better places to take a snapshot than simply on layer change.

Tollbringer, 2020-01-21:
I appreciate the help. It seems somehow PrusaSlicer 2.0 has become messed up. After not being able to determine the issue in Octoprint = I focused on trying a different slicer- and Cura did not pause slicing the same file. I will have to resolve this issue before continuing- And I will beta test the latest version. Thank you.

FormerLurker, 2020-01-21:
No problem.  FYI, slowdowns while printing PrusaSlicer are likely related to your speed settings.  Make sure you are using the correct speeds for retraction/deretraction, z lift, and XY travel.  You might want to use the default profiles if you've made any modifications.

vaderag, 2021-03-09:
> h V0.4. Let me know if you'd be willing to test out the new version, and I can send you instructions for doing so. Be sure to include an email address (which you will be free to delete after I respond to prevent spam due to email harvesting).

I appreciate this is SUPER old, but I'm having this same issue with the Tuya plugin which works the same way...
Every print I do the timelapse receives the fail tag, even though from my perspective it completed.
I have the command with ip address to turn off my printer so suspect this is what is happening, but seems like whatever fix you put in a while back is not working in this instance... :(

FormerLurker, 2021-03-10:
Can you create a new issue @vaderag, and include gcode samples that cause Octolapse to fail?  Also, mention the version you are using please :)

vaderag, 2021-03-10:
> Can you create a new issue @vaderag, and include gcode samples that cause Octolapse to fail? Also, mention the version you are using please :)

will do :)