FormerLurker/Octolapse #337: [Request] adding "Failed Snapshot Count: " to bottom of octolapse

MineThingIssues, 2019-04-12:
#### If this is a feature request describe it here
It would be quite helpful for debugging purposes to have a count of all the failed snapshots, and if possible a sort of graph that could be used to find the frequency of failures (The graph could be snapshots going horizontally with a green line above for success  and a red line above for failure. Similar to: [This](https://lh3.googleusercontent.com/bK4KvJngJFGAAY9gyNqSiViBlRZOHqSYwvOe_JnsE-CMiDVS6kkR6XENk74EBZ-52eHW8Vv81C3c9fkLKv_4VTcvGaanqyBsA01WPpduVVhTCT3WMnhKEayeY7kuw34bZedu9i56xKtb-nZ3hroFJzYYo1OdhW5RUHAdxvuWbstXLQsQMqfTCMsyJwxj_YCanbSHDWruI0bV9uoHvLDAyA-JmG_UjIKMUThfhiUGNITw9Irdy-wyA7WmZmlwGLPA9q_WXIUP8xfWKaUn9xPUmWB8yMpKwKyz_dVWpgtISgO7sSyoDrqc2vK6yqsKaVQMLO9mGyxGcxQ6-R_J26BL4SUgVjx4p3t7EINjkkaIAkuiyIZF1TPbAAxM3RJnRwv6-RT3F-ohagwZ6-SynfaHifqKpZR84m0eWbeRcBeTbxS3OoBHNOnYkNzhcBes3sMQlSHF_1iJIu-HAgzb_7iK-bFyNf6P2FZQi-1jH6vXlziBRbRML9CGBy-BipB-I7xjqXa2JWlJigidyYHzqDWom2fBplXonz1CH3uIeBBr36V1cb3MuZp1thyzNyV3-FpVw1dG01oHVI9x8obmLiLnfGgo1fDFre5fcqVc16bLncCxLnkC8Ta3SyvmpwQKx9vqtMg62wnypMH7ghl7ev0GMW6dVfifDX4=w170-h32-nol) but with green instead of white and red instead of blue). It is a little thing but I think that it could be a useful tool for debugging (such as trying out different timeout settings to see how that affects the reliability of snapshots)

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

FormerLurker, 2019-04-13:
Interesting idea.  I was thinking of adding some debugging screens that would include things like position errors, gcode errors, warnings and other useful info, but I don't think I can fit this on the current tab without some design changes.  Eventually I also want to add setup and debugging tools like automatic printer setup too, which might also work well on such a screen.  However, a failed snapshot count could pretty easily fit on the existing tab, though I think the graph would work better in a different view.  Ideas?

I think the next step is just to sketch this out and try to flush out what works.

FormerLurker, 2019-10-07:
Update (after months...):  Still working on this.  I may yet include this in the next version of octolapse.  Will keep you posted.

FormerLurker, 2020-02-26:
This feature is now available in the latest devel/rc and devel branches.  Let me know if you want to try it out.  The number of snapshot errors (failed snapshots) now displays IF there is at least one error.  If you try it and have problems, please post on [this thread.](https://github.com/FormerLurker/Octolapse/issues/455) 

Closed at long last!

closed 2020-02-26 by FormerLurker