FormerLurker/Octolapse #339: Octolapse - Camera Error [Request]

kazzthehobbyist, 2019-04-26:
#### If this is a feature request describe it here
snapshot_script_error: The snapshot script failed with the following error message: 
*** Error ***

#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: OctoPrint 1.3.10 running on OctoPi 0.16.0

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/3120361/plugin_octolapse.log)

#### What were you doing when the problem occurred
1. Step: Had already setup my DSLR before and that worked fine. disconnected the camera for some time.
2. step: Hook up the camera and to try it out
3. step: Running a test in Octoprint/Octolapse the camera is focusing, but no snapping
4. step: Running through the youtube video to see if anything is missing. The test are doing well
5. step: Trying to run a test in Octoprint/Octolapse, but Its the same as in step 3.

#### What should have happened?
The camera should have snapped and waiting 25000MS.

#### What happened instead?
The camera starts to focus, but dont snap and the print continue after a couple of secounds

#### Operating System running OctoPrint and Octolapse
OS Name: Windows 10 home

#### Printer model & used firmware incl. version
Printer Model: Creality Cr-10s
Printer Firmware Version: Marlin 1.1.9

#### Browser and version of browser, operating system running browser
Browser: Google Chrome
Browser OS: Version 73.0.3683.103

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: <script src="https://gist.github.com/kazzthehobbyist/97176b09e579df51868c88159ad20b6b.js"></script>

#### Link to settings.json
Link to settings.json with all passwords removed: I could not find there it be found

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  [plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/3120481/plugin_octolapse.log)

#### Link to octoprint.log
Link to octoprint.log:  [octoprint.log](https://github.com/FormerLurker/Octolapse/files/3120483/octoprint.log)

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: https://youtu.be/TMbc983op0M
[image]

kazzthehobbyist, 2019-04-28:
I found a solotion on this problem myself.
It turns out that My camera did not focus in time to take a picture.

closed 2019-04-28 by kazzthehobbyist

FormerLurker, 2019-04-28:
Hi Kazzthehobbyist!  I was just looking for the solution to this because I've seen that error before.  I'm very glad you got it solved, and sorry about my lateness in responding (work's been crazy).

Thanks for submitting your issue, and happy printing!

kazzthehobbyist, 2019-04-28:
Its good. It runs pretty smooth, Thou then printing The camera (canon eos 1200D) it needs at least 10 secs to process my picture and its causing plastic to ooze to much. If I shorten the pause it gives me an error which is why wanted to use the capture without downloade option.

FormerLurker, 2019-04-28:
@kazzthehobbyist, maybe you're talking about the snapshot timeout and not the delay?  I agree that the no-download or deferred download option would be the best.  I strongly recommend turning autofocus off if you can, as well as reducing the resolution/jpeg quality to minimize the snapshot time.  You should be shooting for no more than 2.5 seconds to acquire the image, which should be doable.  Increasing the shutter speed, decreasing the exposure, and increasing your lighting should all help.

kazzthehobbyist, 2019-04-28:
@FormerLurker Yes I meant the timeout (Its getting kinda late). I have already turned off the auto focus, but its like everything bellow 5000ms is not possible and adjusting retraction settings dont seem to help to much.

FormerLurker, 2019-04-28:
Not much is going to help considering it's taking 25 seconds.  We'll just continue this in the other thread because it's looking like deferred/no download will be the way to go.