FormerLurker/Octolapse #340: Defered image download and gphoto2 in shell mode

yet-another-average-joe, 2019-04-29:
This is neither a feature request nor as issue ! It is experimentation...

There's a little problem (IPHO) with the defered image download procedure : the SD card has to be erased.

 In shell mode, we can change the destination directory for the images. We can also create directories (but for some reason it did'nt work for me...). So, I created a directory on the card using a computer. On Nikon D5100, it was created under DCIM, and has to be like so : "nnnD5100". Obviously, it depends on the DSLR brand and model. Let's call it 999D5100

Then, using the shell mode, the images can be stored in this new directory. The command lines : 

`echo cd  store_00010001/DCIM/999D5100 | sudo gphoto2 --shell`

Now, this directory will be the current directory for capture calls : the images will be stored in this directory. There will be no need to erase the entire SD card before a timelapse, but just the contents of 999D5100.

BUT : the camera will keep this directory as the current directory when used normally (with the shutter). So, the current directory will  have to be restored. I don't know how to programmatically save/restore this path.

There's also a great feature with shell mode. We can move the mirror up and download previews !!!! Yes, LiveView can be turned on when the D5100 camera is in USB mode !!! I don't know if other cameras work this way, but it is most likely portable.

```
set-config /main/actions/viewfinder=1 | sudo gphoto2 --shell
capture-preview
capture-image
```

BUT the mirror goes down as soon as we quit shell mode. There's workaround : the DSLR must be in shell mode from the begining. This means starting a named pipe, and sending commands to this pipe for all commands, until the end of timelapse.

This means that the idea exposed in issue #331 could be applied to other cameras than mirrorless ones.

Some thoughts ? Is it worth exploring this ?

MineThingIssues, 2019-05-22:
I think that this idea has a lot of potential! I have never heard of shell mode so I have no idea what it does, but being able to download previews would be great for deferred image download. Having the sensor always exposed would allow for image capture to be a lot faster if it doesn't have to lift the mirror up. This could decrease the print time by a fair amount. I do not know how to do this on a canon camera such as the Canon 1300D (T6) that I own, but it would be good to experiment with it a bit and see what works.

If you could reply with the commands to put the camera into shell mode and a few more gphoto2 commands to take pictures and save them then I could potentially try and help get this working on canon cameras. I am not a developer and know quite little about shell scripts or programming in general, but I am sure that I can do something that may help.

I was thinking that you could potentially take one image at the default camera directory after it is done which could potentially reset the current directory back to the main one. It is not ideal, but it would only result in 1 image that the user could delete easily manually, or there could be some way to delete it using gphoto2.

FormerLurker, 2019-05-22:
I've been thinking about writing a 'camera plugin' that would support things like GoPros and DSLRs.  A pretty complete understanding of how exactly gphoto2 works, and what is required for each camera would be critical.

The basic idea would be that the plugin would create an API similar to a webcam that can return a still image or thumbnail (or maybe even a live stream, but that might be pushing it).  Some additional functionality could allow for deferred image download, or other goodies that could hopefully integrate with both Octolapse and the built-in timelapse plugin.

Gphoto2 has been very difficult to support since every camera seems to function differently.  Recently I've been trying to help users get some of the newer mirror-less Sony cameras working, and it's been... difficult to say the least.  Capture and download seems to work well, but I've not been able to get deferred downloads to work.  It doesn't help that I don't have access to one of these cameras :)

@yet-another-average-joe, sorry I didn't respond sooner.  Been pretty much 100% preoccupied with getting the next release in working order.  Please continue to post what you learn about about shell mode.  If I write a plugin it would probably interface with libgphoto2 directly, but I imagine the knowledge you gain would be applicable.

@MineThingIssues, please also post anything you learn here too so we can keep it all in one place.

MineThingIssues, 2019-05-23:
When I have some time I will try and experiment a bit with shell mode in gphoto2. It looks to be quite easy to use, but I will have to try and see.

`cd DIRECTORY` allows you to specify a directory for the camera, and `lcd DIRECTORY` allows you to choose a directory on the local machine. the documentation can be found [here](http://www.gphoto.org/doc/manual/ref-gphoto2-cli.html)

@FormerLurker a lot of the mirrorless cameras that are out now are quite modern and so may be newer than the gphoto2 version that is downloaded using wget. If someone could try and set it up with a new version of gphoto2, then it could potentially solve a few issues. Mirrorless cameras are supported on the website (such as the Canon EOS R), but I do not know about older versions of gphoto2. running the command `--list-cameras` will tell the user if their camera is supported.
To be clear, just because it doesn't say that it is supported, doesn't mean that it won't work. I presume that the ones on the website are just the ones that have been mildly tested for basic functions

yet-another-average-joe, 2019-05-26:
In fact, shell mode is not that interesting, at least with the Nikon D5100.

set-config /main/actions/viewfinder=1     > the mirror goes up (= LiveView)
capture-preview                                        > a preview image is shot and downloaded
capture-image                                           > a picture is shot and saved to the SD card, but after the shutter does its job, the mirror moves down, then up. Why ?  (tested with autofocus off and manual exposure, with no lens on the camera).

I don't know why it behaves this way... There is no real benefits in terms of latency or timeout. It is very disapointing.

Could someone test with another camera ? 

About the directory : how to restore the default target directory after a timelapse ? There is no shell funtion that returns the current target folder on the SD card (or I missed something...). It is not very convenient for the user to first explore the SD card before a timelapse, write down the current folder, and later issue "cd store_00010001/DCIM/xxxD5100" in shell mode in order to restore the normal target folder ! A tech enthousiast will be happy with this (maybe...), but probably not normal users.

I feel like it is a dead end, at least for the Nikon D5100.

FormerLurker, 2019-10-07:
Yeah, I've had no luck with shell mode :( Going to close this issue, but will reopen it upon request.  I'm doing a bit of issue housekeeping today in preparation for the next release.

closed 2019-10-07 by FormerLurker