FormerLurker/Octolapse #342: Installation Issue - Pillow

jalanjarosz, 2019-05-02:
System: ASUS Tinkerboard SMB (similar to Raspberry Pi, but more powerful)
OS: armbian (debian based)
Octoprint: 1.3.10

Octolapse is not installing. Here is the install log just for Octolapse:

```
2019-05-02 16:15:48,463   /home/pi/OctoPrint/venv/bin/python -m pip install https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip --no-cache-dir
2019-05-02 16:15:49,997 ! DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
2019-05-02 16:15:50,714 > Collecting https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
2019-05-02 16:15:50,715 > Downloading https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip (4.5MB)
2019-05-02 16:15:52,717 > Collecting pillow (from Octolapse==0.3.4)
2019-05-02 16:15:54,720 > Downloading https://files.pythonhosted.org/packages/81/1a/6b2971adc1bca55b9a53ed1efa372acff7e8b9913982a396f3fa046efaf8/Pillow-6.0.0.tar.gz (29.5MB)
2019-05-02 16:16:01,951 > Requirement already satisfied: sarge in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from Octolapse==0.3.4) (0.1.5.post0)
2019-05-02 16:16:01,952 > Requirement already satisfied: six in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from Octolapse==0.3.4) (1.12.0)
2019-05-02 16:16:01,952 > Requirement already satisfied: OctoPrint>1.3.8 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from Octolapse==0.3.4) (1.3.10)
2019-05-02 16:16:01,952 > Requirement already satisfied: psutil in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from Octolapse==0.3.4) (5.4.8)
2019-05-02 16:16:01,952 > Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.4)
2019-05-02 16:16:01,953 > Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.9)
2019-05-02 16:16:01,953 > Requirement already satisfied: semantic-version<2.5,>=2.4.2 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.4.2)
2019-05-02 16:16:01,953 > Requirement already satisfied: future<0.16,>=0.15.2 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.15.2)
2019-05-02 16:16:01,953 > Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.3)
2019-05-02 16:16:01,953 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.6.5)
2019-05-02 16:16:01,954 > Requirement already satisfied: futures<3.2,>=3.1.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.1.1)
2019-05-02 16:16:01,954 > Requirement already satisfied: rsa<3.3,>=3.2.3 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.2.3)
2019-05-02 16:16:01,954 > Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10)
2019-05-02 16:16:01,954 > Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.7.19)
2019-05-02 16:16:01,954 > Requirement already satisfied: tornado==4.5.3 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (4.5.3)
2019-05-02 16:16:01,954 > Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.1)
2019-05-02 16:16:01,955 > Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (6.2)
2019-05-02 16:16:01,955 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2019.4.14)
2019-05-02 16:16:01,955 > Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.3)
2019-05-02 16:16:01,955 > Requirement already satisfied: wrapt<1.11,>=1.10.11 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.10.11)
2019-05-02 16:16:01,955 > Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.10)
2019-05-02 16:16:01,956 > Requirement already satisfied: monotonic<1.4,>=1.3 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.3)
2019-05-02 16:16:01,956 > Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.2)
2019-05-02 16:16:01,956 > Requirement already satisfied: netifaces<0.11,>=0.10.6 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.9)
2019-05-02 16:16:01,956 > Requirement already satisfied: pkginfo<1.5,>=1.4.2 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.4.2)
2019-05-02 16:16:01,956 > Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.8.3)
2019-05-02 16:16:01,957 > Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.6.1)
2019-05-02 16:16:01,957 > Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (5.2.1)
2019-05-02 16:16:01,957 > Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.2.11)
2019-05-02 16:16:01,957 > Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.21.0)
2019-05-02 16:16:01,957 > Requirement already satisfied: websocket-client<0.41,>=0.40.0 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.40.0)
2019-05-02 16:16:01,957 > Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.8.3)
2019-05-02 16:16:01,958 > Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.8.1)
2019-05-02 16:16:01,958 > Requirement already satisfied: emoji<0.5,>=0.4.5 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.4.5)
2019-05-02 16:16:01,958 > Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.9)
2019-05-02 16:16:01,958 > Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.3.5)
2019-05-02 16:16:01,958 > Requirement already satisfied: markdown<2.7,>=2.6.9 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.6.11)
2019-05-02 16:16:01,958 > Requirement already satisfied: Unidecode<0.05,>=0.04.14 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from awesome-slugify<1.7,>=1.6.5->OctoPrint>1.3.8->Octolapse==0.3.4) (0.4.21)
2019-05-02 16:16:01,959 > Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from rsa<3.3,>=3.2.3->OctoPrint>1.3.8->Octolapse==0.3.4) (0.4.5)
2019-05-02 16:16:01,959 > Requirement already satisfied: webassets>=0.10 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from Flask-Assets<0.11,>=0.10->OctoPrint>1.3.8->Octolapse==0.3.4) (0.12.1)
2019-05-02 16:16:01,959 > Requirement already satisfied: singledispatch in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from tornado==4.5.3->OctoPrint>1.3.8->Octolapse==0.3.4) (3.4.0.3)
[... cut ...]
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

jalanjarosz, 2019-05-02:
Noticed issue with zlib in the log and I confirmed that I have zlib installed:

```
pi@tinkerboard:~/.octoprint/logs$ sudo apt show zlib1g
Package: zlib1g
Version: 1:1.2.8.dfsg-5
Priority: required
Section: libs
Source: zlib
Maintainer: Mark Brown <broonie@debian.org>
Installed-Size: 126 kB
Provides: libz1
Depends: libc6 (>= 2.4)
Conflicts: zlib1 (<= 1:1.0.4-7)
Breaks: libxml2 (<< 2.7.6.dfsg-2), texlive-binaries (<< 2009-12)
Homepage: http://zlib.net/
Tag: role::shared-lib
Download-Size: 82.4 kB
APT-Manual-Installed: yes
APT-Sources: http://httpredir.debian.org/debian stretch/main armhf Packages
Description: compression library - runtime
```

jalanjarosz, 2019-05-02:
I reviewed issue #170: [](https://github.com/FormerLurker/Octolapse/issues/170)

After executing: `sudo apt-get install libjpeg-dev libfreetype6 libfreetype6-dev zlib1g-dev`

I was able to manually install pillow with: 
`pi@tinkerboard:~/OctoPrint/venv/bin$ ./pip --no-cache-dir install pillow`

jalanjarosz, 2019-05-02:
Issue was end-user resolved with the help of previous issue. 
Documented process to assist developer and other users.

closed 2019-05-02 by jalanjarosz

FormerLurker, 2019-05-02:
Wow, you are fast!  Feel free to post again if you have any additional trouble.  Pillow has always been a bit iffy to install.  I'm considering looking for a replacement that has fewer dependencies.  It is very fast and free, so I can't complain just too much :)