FormerLurker/Octolapse #343: MP4 Timelapse is not playable because of unsupported file type

Dragon2781, 2019-05-05:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

My setup is a Prusa i3MK3S with a Raspberry Pi 3B+
I followed your video to add a Canon EOS 7D
1. I deleted all Logfiles and restarted Octoprint
2. I started a Print, the gcode was uploaded locally to Octoprint.
3. Everything worked fine, the print was successfull, the Canon took pictures and I was able to see the pictures in the Octolapse Tab.
4. The rendering of the pictures took about 15min.
5. I downloaded the MP4 file to my Mobile Phone (Android) and it was not playable because the file type is not supported (even it can play MP4 files of course).
6. I downloaded the MP4 file to my PC (Win 10) and it was not playable with WinMediaPlayer because the file type is not supported.
7. I downloaded VLC Player, it was able to play it.

-> I ziped all files you need and uploaded it to file-upload.net
- MP4
- gcode
- octoprint.log
- plugin_octolapse.log

https://www.file-upload.net/download-13593776/FileTypeNotSupported.zip.html

#### What should have happened?
I should be able to play it whth the default players of Windows and Android

#### What happened instead?
I got error messages saying that the file type is not supported

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: Original Prusa i3 MK3S
Printer Firmware Version: 3.7.0-2201

#### Browser and version of browser, operating system running browser

Browser: Chrome 74.0.3729.131
Browser OS: ?

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://www.file-upload.net/download-13593776/FileTypeNotSupported.zip.html

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://www.file-upload.net/download-13593776/FileTypeNotSupported.zip.html

#### Link to octoprint.log

Link to octoprint.log: https://www.file-upload.net/download-13593776/FileTypeNotSupported.zip.html

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://www.file-upload.net/download-13593776/FileTypeNotSupported.zip.html

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-05-07:
Hey, sorry I didn't answer right away.  Work has been very busy, and I'm having a hard time keeping up with tickets.

The MP4 works with VLC.  I did change some things in the rendering settings a while back that might have fixed this, but I've been struggling to complete the next release, so that fix isn't in there (it also fixes the max framerate of 25fps bug that I see in this video).  

Can you upload your settings.json file for me?  You can export them from the settings page (wrench icon->octolapse).  I need to see your rendering settings.  I do notice that the resolution is insanely high (more than my monitor), so that could be part of the issue.  The other two items I'd like to look at are the 
bitrate and the codec.

Dragon2781, 2019-05-07:
Hi, thank you very much for your answer. I think everybody is busy this days.... :)
Here is the settings.json
https://www.file-upload.net/download-13597075/Settings.zip.html

FormerLurker, 2019-05-07:
Hey, is there any way you could paste into gist.github.com instead of file-upload.net?  That place is virus central apparently.  I already had to dodge a few sneaky exe files and moving download boxes.

Dragon2781, 2019-05-07:
Oh, sorry about that. The plugin_octolapse.log was to big for the gist thing.

Here is the settings.json
https://gist.github.com/Dragon2781/21f3bb96ddb9f161cc5e9e27cae55587

FormerLurker, 2019-05-07:
Thanks!  So, everything looks good unfortunately.  Maybe set your rendering to 'Fixed FPS' and use 25FPS?  That MIGHT work.  I think the issue happens when the output FPS doesn't match the input FPS (output set to 25FPS due to a bug).

Dragon2781, 2019-05-08:
For other people running into this issue:
We finally managed to fix the problem by lowering the resoultion in the settings of the camera from 18M (L) to 4.5M (S).

FormerLurker, 2019-05-16:
Since this appears to be solved, I'm going to close the issue.  However, I will do some research to see if this problem can be eliminated without the need to reduce quality so much.

closed 2019-05-16 by FormerLurker

dimo414, 2021-04-25:
I'm observing (I think) the same issue. I'm using a Raspberry Pi camera on an existing Raspbian install (i.e. not OctoPi; I followed [this guide](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian-or-raspberry-pi-os/2337)). Octoprint's built-in timelapse renders videos that are playable everywhere I've tried (Chrome and Firefox on Windows, OSX, and Android), but the Octolapse videos are only playable in dedicated software like VLC, they don't play in-browser. Firefox reports an [error message](https://user-images.githubusercontent.com/1885701/115985916-d858ac00-a562-11eb-9de9-76c38a2de2e9.png) "No video with supported format and MIME type found."

I have not mucked with the rendering profiles in Octolapse at all, and I've tried a couple different profiles but no luck. I haven't tried changing the camera's resolution; given that the built-in timelapse works I'm assuming this is an issue with Octolapse, not my hardware.

Here's my [settings.json](https://gist.github.com/dimo414/2941622a4b3622136e0e6cc45ca81f22), as well as an example timelapse from [Octoprint](https://drive.google.com/file/d/15TiE5oDhxQvTekkO9C5uF-7KWHOUTMS7/view) vs. [Octolapse](https://user-images.githubusercontent.com/1885701/115985310-2c15c600-a560-11eb-902b-c25750287310.mp4). Let me know if there's any further diagnostics I can share.

FormerLurker, 2021-04-25:
@dimo414, have you tried a fixed fps profile?  It os usually either the resolution or the framerate that gives some players problems.

dimo414, 2021-05-19:
Yes I'm using the default profiles and have tried both `MP4 - 30 FPS` and `MP4 - 60 FPS`, which both say they use a constant framerate.

jwhb, 2022-03-24:
Firefox 99.0b7 (64-bit) shows me the error `NS_ERROR_DOM_MEDIA_DEMUXER_ERR` in dev console. It's similar to this bugzilla report:
https://bugzilla.mozilla.org/show_bug.cgi?id=1427278

So apparently it's not supposed to work.
I'm now trying with H264 codec instead of MP4.