FormerLurker/Octolapse #345: onedrive?

BlackNet, 2019-05-16:
Any way possible or future upgrade type that would save the video's to onedrive storage, or even google drive, or icloud or etc.??

#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-05-16:
I've been considering custom rendering/snapshot folders, and will think about how this can be accomplished.  I don't think it could be done before the next release, but I've gotten similar requests in the past, so it seems reasonable.

In the meanwhile, it's possible to do this by writing a custom 'After Render Script' and adding the path to your camera profile:

[image]

I'm not sure how exactly you could copy to onedrive, but I bet there is a way.  That would at least get you by until a built-in feature is available.

BlackNet, 2019-05-16:
that should work.  

I did find this project. https://github.com/RaspiRepo/Raspberry-PI-OneDrive and that's a mount point and a standard cp or what not should be sufficient.

FormerLurker, 2020-02-26:
Hey, FYI, I pushed an update to the Dropbox Timelapse plugin so it now supports Octolapse uploads.  It's not onedrive, but it's not bad.  You can test it out (not yet merged with master) by installing from this URL in the plugin manger:  https://github.com/jslay88/OctoPrint-Dropbox-Timelapse/archive/octolapse-hooks-v2.zip

FormerLurker, 2020-02-26:
Let me know about the above solution if you try it out.  If you have any issues, feel free to open a new issue.  For now I'm going to close this out, and sorry about the super late response!!!!

closed 2020-02-26 by FormerLurker