FormerLurker/Octolapse #346: Octolapse with Octoprint-Cancel Object

rmilyard, 2019-05-22:
#### Version of Octolapse

Octolapse Version: Octolapse 0.3.4

I had Octolapse setup and running fine.  PrusaSlicer 2 now supports Octoprint-Cancel Objects so I installed current version and config for it.  Now Octolapse doesn't seem to trigger for pictures now.  How can we troubleshoot this?

FormerLurker, 2019-05-22:
It looks like the issue template got messed up somehow.  I see the 'version of Octolapse' but not any other data.  I'm not sure right now why you are not getting snapshots.

If you could generate a test print gcode with two objects (something very small like a couple calibration cubes), and set the Octolapse debug profile to 'Test Mode - Full Diagnostics', run the print, then upload the resulting plugin_octolapse.log file from OctoPrint settings (wrench/spanner icon)->logs, that would be very helpful.

Just so you know, getting Octolapse to play nicely with other plugins that block/alter gcodes is pretty challenging.  For example, it's possible that the cancel objects plugin could alter gcodes sent by Octolapse, which would not be good.  It's also possible that Octolapse could intercept a gcode and process it before the cancel objects plugin has a chance to block the gcode (though I don't know exactly how that plugin works yet, I could be mistaken)  

A working solution might require changes to both the Cancel Object plugin as well as Octolapse depending on what isn't working.

rmilyard, 2019-05-23:
I printed file xyzCalibration_cube_0.2mm_PLA_MK3S.gcode
[plugin_octolapse.zip](https://github.com/FormerLurker/Octolapse/files/3210150/plugin_octolapse.zip)

FormerLurker, 2019-05-23:
Great, thanks!  Ill be able to take a look at this tomorrow and hopefully figure out what's going on.

rmilyard, 2019-05-23:
Good.  If need more let me know.

rmilyard, 2019-05-25:
Find anything?

FormerLurker, 2019-05-25:
Sorry, ended up working late.  Will take a look tomorrow.

FormerLurker, 2019-05-26:
There are a lot of rewritten gcodes in the log you sent, but no evidence that Octolapse was having any particular problem, so I'm a bit confused.  Can you send me your settings.json file (see octoprint settings->octolapse->Download Settings.  Also, a copy of the gcode file that you used to generate the log file would be useful.

rmilyard, 2019-05-26:
Clearing all the logs and redoing this.  I will send all in about an hour.

rmilyard, 2019-05-26:
[octoprint.zip](https://github.com/FormerLurker/Octolapse/files/3220524/octoprint.zip)
Here is all the files.

FormerLurker, 2019-05-26:
Thanks!  It looks like I'm going to have to slice and run a print with cancel objects enabled to figure this out.  I've got a few tests to finish running, then I'll try to do this.  I've changed so much in the dev version that I'd like to get it working there instead of fixing it in an update to the current master version.  Once I think I have it solved, would you be willing to test the fix on an experimental version?  It might be up to a week before I have it stable enough for alpha testing though :(

rmilyard, 2019-05-26:
Sure. I can test it out.

FormerLurker, 2019-06-03:
I wanted to give you a status update.  As usual, things are taking a bit longer than I'd hoped with the next version.  I'm out of town most of this week, but should be back at it after that.  Once I get the dev version rolled out I'll be in testing mode, and this issue will be on my radar again.

Twilek-de, 2020-09-28:
I am experiencing a problem with cancel objects as well. Octolapse works but cancel objects shows an empty screen in the web interface. BUT it seems to work from Octopod which seems to have it´s own connection to cancel-objects.