FormerLurker/Octolapse #347: Webcam stream works with timelapse but not Octolapse

JoeK1973, 2019-05-22:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

v0.3.4

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

v1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

NO

#### What were you doing when the problem occurred

1. I was setting up Octolapse to connect to my webcam running on my network
2. I put in my IP address (http://192.xx.x.xxx:8080/video/) and I changed the snapshot URL to what it needs to be for snapshots {camera_address}shot.jpg and hit the test button
3. I get a 404 or similar error message telling me it couldn't find the stream despite this address being the one that Octoprint's timelapse software uses (the stream works on Octoprint built-in timelapse)

#### What should have happened?

It should have connected to the webcam stream correctly

#### What happened instead?

Symptoms as described above.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.14.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 Mk 2 clone
Printer Firmware Version: Marlin v1.1.6

#### Browser and version of browser, operating system running browser

Browser: Chrome 74.0.3729.157
Browser OS: Windows 10 64-Bit

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://drive.google.com/open?id=16qGzBMcMtUZ-qyzxMgsDMeu3ICr_Y9bD

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-05-22:
What streaming software are you using?  I'm not familiar with the URL setup you are describing.

Try this:  

In the camera profile, leave the Base Address alone, then for the Snapshot Address Template enter this:

```
http://192.xx.x.xxx:8080/video/shot.jpg
```

See if that test works.  If it fails, paste that same address into your browser and see if that works.

If you can send me a snapshot of the following settings: OctoPrint Settings (wrench/spanner)->Webcam&Timelapse->Snapshot URL, just so I can see it in place.  You might want to click 'test' while you're there too just to double check.

Thanks!

JoeK1973, 2019-05-22:
Hi, just printing right now (so unable to enable octolapse to try your suggestion) but I took a couple of screenshots of the timelapse settings:

[https://drive.google.com/open?id=1U4jT1SkAb0HOt0fqYLIVNeU2tjpLhWcN](url)
[https://drive.google.com/open?id=1BCzP6YHVbHN5mzVmj2DDoWi1bBvxu8Yi](url)

Will try your suggestion as soon as I can, many thanks for getting back to me so quickly!

Joe

JoeK1973, 2019-05-22:
Oh and I'm running [IP Webcam](https://play.google.com/store/apps/details?id=com.pas.webcam&hl=en_GB) on an old Android phone, it runs a server streaming the camera from the phone onto the LAN?

FormerLurker, 2019-05-22:
No problem!  You caught me at the right time.

The images was helpful enough that I think it can be solved without additional info.  Set your base address to this:
```
http://192.xx.x.xxx:8080/
```
then set your Snapshot Address Template to this:
```
{camera_address}shot.jpg
```
The problem is that your stream URL has a /video/ path, but the snapshot URL does not, hence the 404 error.

You can edit the profile even while you are printing if you want.  Enabling Octolapse is not necessary (though it shouldn't hurt anything either) since you can do this from the Octolapse tab by clicking the gear icon to the right of the camera profile.  It shouldn't hurt anything.  Just change those two settings in your camera profile and click test.  If it the test is successful, save the changes and you should be good to go!

JoeK1973, 2019-05-22:
Awesome! Worked a charm! Thanks very much!

Looking forward to getting into Octolapse tomorrow evening after work, cheers!

FormerLurker, 2019-05-22:
Glad I could help, and good luck!  Let me know how it goes.

Closed!

closed 2019-05-22 by FormerLurker