FormerLurker/Octolapse #348: [REQUEST] External Octolapse trigger using Pi pins instead of USB

nyanroberts, 2019-05-26:
#### If this is a feature request describe it here

Had great success using external DSLR control over USB. The one limitation I'm finding is being able to synchronize Octolapse with my motion timelapse rigs (Emotimo Spectrum ST4, and Dynamic Perception MX2). I would able to manually trigger both timelapse controllers using an I/O port (basic TRS terminal) in order to perfectly synchronize Octolapse and my camera movement. Hopefully there is an easy way to modify the camera script to use the pins on the Raspi (5V and GND) and then make a custom cable. The script for my Canon 7D required feedback from my camera to verify that a picture was taken. Not sure how to get around that when there won't be any transferring of files, and there won't be any way to give feedback to Octolapse. I can handle making the cable myself as long I know which pins to connect to (specified in script), and then I imagine changing the script would do the rest. Below is the response from Emotimo about how they suggest to send the signal. -

"For wiring, connect the sleeve of the connector to your GND.  Then connect both the tip and ring to GPIO set to output with a pullup engaged.  This needs to be 5 volt tolerant as there are 5 volt pullup on these.  Once wired, set the ST4 to be in ext trigger mode, then set up any timelapse move and start it, then pause it.  Then drive both tip and ring from your RPi low . . . this should . . .explode . . . or advance the ST4 to the next shot position."

#### Version of Octolapse
 
Octolapse Version: 0.3.4rc1.dev3

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-06-03:
Sorry it's taken me a while to respond.  I've been busy getting the next release finished.

I know that some folks have been able to use the 'Snapshot Acquire Script' to call python code to toggle an IO pin/pins.  I'll dig around and see if I can find an example.

Also, some printers have the ability to toggle an internal pin in the printer's board via gcode (the MK2/MK3 has this I believe).  You could either use the gcode trigger and add this command manually to your layer change script, or you could add a 'gcode camera' profile, and enter the appropriate code to toggle the pin there.  You'll have to play with the delay on BOTH your webcam/dslr AND the 'gcode camera' profile to get this to work.

FYI, I do want to add this capability natively at some point.  Once the next release is finished I plan to focus on making a wider array of cameras work with Octolapse.

FormerLurker, 2019-06-03:
Take a look at [this](https://github.com/FormerLurker/Octolapse/issues/334) issue, where I submitted some sample code to a user for triggering GPIO pins via script.  Note that this user is still having some issues, so if you discover any problems or (better yet) solutions, let me know so that I can help that user out.

nyanroberts, 2019-07-09:
On the official Emotimo site, they say to test your trigger cable by connecting tip, ring, and sleeve together with foil. I decided it was safer to accomplish this action by using a relay instead of just the GPIO pins directly off the board. By finding a compatible relay (3.3v input or optocoupler) and running a simple “low/high” python script via the trigger.sh, it works! Still have some test prints to do, but I should be able to fine tune the snapshot delay using octolapse, or add a timeout delay into the python script before the initial relay trigger. Will post photos/video once the test prints are complete. Thanks!

FormerLurker, 2019-07-09:
Oh neat!  I'd love to see some videos.  If you ever create a guide, let me know and I'll put it in the wiki!  Let me know if you need help figuring out the delay.

Good work!

FormerLurker, 2020-02-26:
I neglected to close this issue after it was resolved.  Sorry about that.  Closed!

closed 2020-02-26 by FormerLurker