FormerLurker/Octolapse #349: [Request] Preview Stabilization Positions

ggrote, 2019-05-27:
It would be cool to have a button to preview the Stabilization positions to get the camera into right position.

Octolapse Version: 0.3.4
OctoPrint Version: OctoPrint 1.3.11 auf OctoPi 0.16.0

FormerLurker, 2019-06-03:
Hay, I wanted to let you know that this feature should be available in the next release.  I've been considering this for a while, and since I've updated the webcam image preferences screen so much, it makes sense to have this.

I plan on adding an 'on before stabilization preview' gcode section to the 'main settings' that will, by default, home the X and Y axis before moving to the stabilization point.  For animated stabilizations, I'm hoping to get it to advance to the next position each time you press the preview button.  For 'disabled' axis and the new (unreleased) snap-to-print trigger, the button will either return an error, or will be disabled with a message.  I haven't decided between those two options yet.

FormerLurker, 2019-06-06:
This is complete.  The gcode script used to home the axis is stored in the printer profile.  There is a button to stabilize the axis in both the camera profile, and on the image adjustment screen.  The travel speed comes from octoprint at the moment, but i will improve this before release.  I wasn't able to get animated stabilizations to advance as I had d hoped, but i will think about it some more.

FormerLurker, 2019-10-07:
I'm going to close this issue since I've added the requested feature to the next version!  Enjoy :)

closed 2019-10-07 by FormerLurker