FormerLurker/Octolapse #350: Exception when using external script

csmith105, 2019-05-28:
Octolapse Version: 0.3.4
OctoPrint Version: 1.3.11
OctoPi Version: 0.16.0

Diagnostic Logging was Enabled: Yes

```2019-05-28 01:21:39,754 Octolapse - loaded and active.
2019-05-28 01:21:43,009 Octolapse - is loading template configurations.
2019-05-28 01:21:59,611 Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__
init__.py", line 1318, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/ti
melapse.py", line 568, in on_gcode_queuing
    if self._is_snapshot_command(self.Settings.current_printer(), command_string
):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/ti
melapse.py", line 1012, in _is_snapshot_command
    return command_string.lower() == printer.snapshot_command.lower()
AttributeError: 'NoneType' object has no attribute 'snapshot_command'```

#### What were you doing when the problem occurred

Trying to capture an image from my Sony A5100 using the external script feature. The script is the default suggested for this feature.

#### What should have happened?

Ideally the image would be catapulted. 

#### What happened instead?

#### Printer model & used firmware incl. version

Printer Model: Ultimaker 2

#### Browser and version of browser, operating system running browser
Browser: Chrome (current)
Browser OS: Win 10 x64

#### Link to the gcode file you were printing when the problem 
Link to Gcode File: It was a benchy exported with S3D

#### Link to settings.json
```{
  "auto_reload_latest_snapshot": true, 
  "current_printer_profile_guid": "a6783118-51fe-41bd-8b00-1df9deda97f0", 
  "default_printer_profile": {
    "small_perimeter_speed": null, 
    "bridging_speed_multiplier": 100, 
    "perimeter_speed": null, 
    "outline_speed_multiplier": 100, 
    "first_layer_travel_speed": null, 
    "top_solid_infill_speed": null, 
    "top_solid_infill_speed_text": null, 
    "movement_speed": 6000, 
    "solid_infill_speed": null, 
    "ooze_shield_speed": null, 
    "guid": "5d39248f-5e11-4c42-b7f4-810c7acc287e", 
    "priming_height": 0.75, 
    "small_perimeter_speed_multiplier": 100, 
    "has_been_saved_by_user": false, 
    "origin_z": null, 
    "abort_out_of_bounds": true, 
    "min_y": 0.0, 
    "min_z": 0.0, 
    "detract_speed": 3000, 
    "suppress_snapshot_command_always": true, 
    "default_firmware_retractions": false, 
    "speed_tolerance": 0.6, 
    "z_hop": 0.5, 
[... 2491 lines cut ...]
      "name": "Off", 
      "value": "off"
    }, 
    {
      "name": "Blink", 
      "value": "blink"
    }, 
    {
      "name": "Auto", 
      "value": "auto"
    }
  ], 
  "camera_powerline_frequency_options": [
    {
      "name": "50 HZ (Europe, China, India, etc)", 
      "value": "50"
    }, 
    {
      "name": "60 HZ (North/South America, Japan, etc)", 
      "value": "60"
    }
  ], 
  "show_position_changes": false, 
  "current_debug_profile_guid": "68d000d6-0ae3-40f4-9b16-bdcb882604a6"
}```

csmith105, 2019-05-28:
Solved by reinstalling Octolapse and going to the release version of gphoto2, 2.5.20

closed 2019-05-28 by csmith105

FormerLurker, 2019-05-28:
Hey, glad you figured this out!  However, I'm not sure what could have caused an error here:

```
AttributeError: 'NoneType' object has no attribute 'snapshot_command'
```

Probably an error thrown somewhere else caused this.  Would you mind sending me your full log that contains that error message?  I'd like to check it out because it should have failed gracefully even if gphoto2 crashed.