FormerLurker/Octolapse #351: Octoprint FlipCam 90° not for snapshots ?

tardar, 2019-05-30:
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE
OctoPrint Version: 1.13.11

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Configured FlipCam to 90°
2. Print
3. Make a snapshot and it wasn't flipped

#### What should have happened?
The Snapshot should been flipped like the live Cam video.

#### What happened instead?
The liveCam Picture wasnÄt flipped and looks bad on mky smartphone ^^

OctoPi, Linux
OS Name: OctoPi
Os Version: 0.16.0

FormerLurker, 2019-05-30:
Look in your octolapse camera profile at the transpose setting.  You can rotate/flip/invert there

FormerLurker, 2019-06-03:
Did that do the trick for you?

tardar, 2019-06-03:
No it didn't.
I flipped the cam in OctoPrint so i can see the right webcam picture.
Now i try to send via FHEM a picture to my smartphone and it uses the Link from the webcam (snapshot).

I think i have to manipulate the link (with a parameter) to flip the picture.
Any ideas ? :)

Thanks really alot

FormerLurker, 2019-06-06:
Wait, are you talking about the snapshot url, and you are using that with fehm?  So really this has nothing to do with octoprint or octolapse.  The software running the camera stream is called mjpegstreamer.  It is installed by default on OctoPi.  I think it is possible to tell mjpegstreamer to pre rotate your stream for the pi cam, but I don't think it is possible for usb cameras (I could easily be wrong).  In any case there is no way to rotate the video via a url.  You could also create a script on your pi to rotate the image, or if you are using octolapse, you can download the latest snapshot via a url.  It wont be real time, of course.  Let me know if you'd like to try that.

You could also install a different streaming server that supports rotation.  Shouldn't be hard.  Better yet, install it on a second pi and offload all that workload!

FormerLurker, 2019-07-14:
Were you ever able to get this working?  Since I'm pretty sure this isn't an Octolapse issue at all, I'm going to go ahead and close the ticket.  If you have any more questions, just let me know and I'll reopen it.

Thanks for posting!

closed 2019-07-14 by FormerLurker