FormerLurker/Octolapse #353: Octolapse takes pictures before print head and hotbed are correctly positioned

ajdavids, 2019-06-03:
#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.11 running on OctoPi 0.15.0PE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Add "snap" to the gcode in the PrusaSlicer printer profile, in the "after layer change" section
2. Slice an stl and upload the gcode to octoprint
3. Enable octolapse. Set the snapshot mode to "gcode". Choose any print head position; I've tried "center right" and "animated - printing"
4. Print the object

#### What should have happened?
I should have gotten a timelapse with the bed in the same position every time and the printhead where it was supposed to be every time

#### What happened instead?
OctoLapse takes snapshots before it finishes moving the print head or bed, so the printhead is still directly above the print and the print is not in a stable position.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.0PE

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3S
Printer Firmware Version: 3.7.1

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: I've tried with several gcode files. Here's one of them:
https://gist.githubusercontent.com/ajdavids/caf65adbde56ba7465ae9da702c5bf9f/raw/a9568fadf8789393c18bc41adeba001caeb54f53/Whistle_0.15mm_PLA_mk3S_44m.gcode

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/ajdavids/f00cb7b41ebecb37edf4c24c60a2fe47

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/ajdavids/5ccd50cc06eff9f19aa3fa50cda15235

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/ajdavids/74155812617e79f97980bdd3d1c16970

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://youtu.be/qOXLWLMcR1E

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-06-05:
So all of the usual culprits for this problem don't seem to be involved here, but there's one other thing I'd like you to try before I get too technical on you.  Can you disable all of your other plugins, especially anything that reports on positions (might not be obvious which plugins are doing this).

Octolapse relies on the timing of OK responses from an M114 command to know when the extruder is stabilized (or a very large camera delay).  If some other plugin sends the same code at the wrong time, it can trick Octolapse into thinking that the printer has responded to the M114 command (on some printers, G91 can cause this same issue because it replies with a current position response).

If that doesn't help, we might need to bring out the big guns.  You can set Octolapse's debug mode to 'Full Diagnostic - Live Print', print about 25 - 30 layers using the same settings and gcode you attached to your original issue, then stop the print and send me the new log file.

If you need a quick fix (though it might impact your print quality), you could increase your camera delay by a couple seconds to give the extruder time to stabilize.

ajdavids, 2019-06-07:
Hey FormerLurker, one of my plugins turned out to be causing it. I've let the plugin's creator know. It's a plugin that's currently in beta, and the beta version of the plugin does not send OK responses to M114 commands, but somehow seems otherwise incompatible with Octolapse. Are there any other ways, besides that one response, that a plugin might cause the behavior being discussed?

FormerLurker, 2019-06-07:
Ok, cool.  Have the author message me if they have any questions ok?

ajdavids, 2019-06-07:
Will do!

kennethjiang, 2019-06-07:
The author of OctoPrint Anywhere  here. The beta version of OctoPrint Anywhere replaced `mjpeg-streamer` with a different mechanism so that it can do H.264 streaming. But the new mechanism may introduce up to 200ms in delay to snapshotting. I guess this delay will be too much for Octolapse because it requires very precise timing? If so, what is the tolerance for the timing of a snapshot?

FormerLurker, 2019-06-07:
Hmm.  So the snapshots taken are from 200ms ago?  In that case 200ms needs to be added to the snapshot delay to compensate, which will affect print quality somewhat.  Maybe ill try installing it next weekend to see if there is a workaround.

Lots of other stuff in Octolapse wont work yet without mjpegstreamer, like camera controls, but those could be added since i anticipate supporting other streaming servers.  Which one are you using?

kennethjiang, 2019-06-07:
Sounds like things are more complicated than I expected - as always. :)

I wrote a simple snapshot server here at: https://github.com/kennethjiang/OctoPrint-Anywhere/blob/beta/octoprint_anywhere/h264_stream.py#L63

The reason for having to replace mjpeg-streamer is that, in order to produce H.264 video stream, I need to use Pi Camera module as in `/dev/video0`, which allows only exclusive usage.

FormerLurker, 2019-06-07:
@kennethjiang, 

that's pretty neat!  I'm going to be in the woods for a couple of days, but I might take a look at this when I get back so that I can understand what's happening.

>Sounds like things are more complicated than I expected - as always. :)

Tell me about it, lol!

kennethjiang, 2019-06-09:
I prototyped a solution that can respond to a snapshot in about 50-100ms (the api response time varies for some reason I don't know yet).

What's the for `mjpeg-streamer` to respond to a snapshot request? Does it also vary in your experience? @FormerLurker

FormerLurker, 2019-06-09:
I usually see 10-30ms delay over Ethernet (more over wireless) to acquire an image, plus any framerate delay (1/frame rate*1000).  I'm guessing that the snapshot capture is probably returning the previous frame for some reason, or the framerate is just really low.  At 5fps the frame rate delay up to 200ms.  Maybe that is a component of this issue?

kennethjiang, 2019-06-09:
The frame rate for snapshotting goes anywhere between 20FPS and 35 FPS (note this is different from the framerate of H.264 streaming, which uses a different "port" in Pi Camera module). This 20-35FPS accounts for the bulk of 50-100ms delay I'm getting in my prototype (not the current production version in which 200ms comes from some other factors that are too much to elaborate here).

kennethjiang, 2019-06-10:
I'm trying to test my fix that has much lower snapshotting delays. However, I found out that my Octolapse has stopped working. No snapshots are ever taken. At the end of the print it just said that Octolapse failed because there are no sufficient snapshots.

See attached screenshot for what happened. I highlighted a couple things that might be related to the problem. Also it could have been the new OctoPrint 1.3.11 release (I don't remember I have had an Octolapse timelapse since then).

This is the case with out without OctoPrint Anywhere enabled.

[image]

FormerLurker, 2019-06-10:
Yes, it is dry running.  If you want that, just use 'test mode' in the octolapse debug profile.  I think they both ise basically the.same.code anyway :)

kennethjiang, 2019-06-10:
Cool thx!

After some more digging (thank you @ajdavids so much for helping me with it), it turns out that the [picamera python module](https://picamera.readthedocs.io/en/release-1.13/) I'm using somehow causes unpredictable delay ranging between 40ms to 200ms.

I'm wondering if a consistent delay is more important than a delay that's most times pretty short but some times awfully long. If so, I can insert some artificial delay in my snapshotting api so that it more or less delays for 200ms for each call. Do you think this is a viable solution @FormerLurker  ?

FormerLurker, 2019-06-10:
@kennethjiang,

If the snapshot that's returned is for the current frame (not the previous) it should be OK.  I think the least possible delay would be better.  Octolapse will happily wait up to the snapshot timeout (5 seconds by default) to acquire an image from a get request.  The problem that OP had was that the frame was stale, which resulted in a timelapse that's not stabilized.

What about using the capture_continuous function, and always storing the result with the same filename (snapshot.jpeg or something similar), then serving that up when a snapshot request is made?  You'd have to deal with the case (threading) where a snapshot request is being made by a client, but you are in the middle of capturing the next frame.  Not sure if the same delay would be present in this case or not, but in at least some cases it would reduce the delay, no?

kennethjiang, 2019-06-10:
Thank you for the details! I'm using `capture_continuous` and it is this function that somehow gives me very unpredictable delays, probably due to some weird internal mechanism. Will try the artificial delay later today when I get home.

FormerLurker, 2019-06-10:
> I'm using capture_continuous
Looks like we think alike :)  Too bad it's unpredictable ATM.  Let me know how it goes.  This is a really interesting project you are working on!

FormerLurker, 2019-07-23:
@kennethjiang, any progress on your streaming server?  I thought I'd share something I learned that may help you pin down the random delay.  I've been working on reducing the lag acquiring images through mjpg-streamer, and noticed that there seemed to be a variable delay somewhere between 50MS and 150MS.  What I discovered was that the 100MS of variability was likely due to the 10FPS framerate I had set.  This makes total sense if mjpg-streamer waits until the next frame is available before delivering an image.  1/FPS*1000 should give you the approximate variation in delay if this is the case.  The other 50MS seems to come from the get request + the download time.  I was able to reduce the delay to between 10MS and 50MS by using a 30FPS stream and also using a streaming download in a separate thread and signaling success as soon as I receive a response from the get request.  Then Octolapse continues to download the image in the background while Octoprint keeps on running.  However, mjpg-streamer will still wait until the next frame is available, hence the approx 40MS variation in image acquisition time.

kennethjiang, 2019-07-24:
Thank you so much @FormerLurker for this detailed info!

I fixed this issue by taking 3 consecutive images and returning only the last one to make sure the old cached images won't get returned. My observation is the same as yours (although I am using python `picamera` module) that the latency is mostly below 50ms with 25FPS, hence the latency for each snapshot now takes less than 200ms (3x50ms + some download time). Not ideal but nobody has complained about the added delay yet. :)

@ajdavids has confirmed with me that the bug is fixed. Feel free to close this issue. Thanks again for your help.

FormerLurker, 2019-07-24:
@kennethjiang, no problem at all, and thanks for dealing with this side issue.  I appreciate it, and all of the other great work that you do!

Closed!

closed 2019-07-24 by FormerLurker