FormerLurker/Octolapse #354: Fails to start because the delay from Android app IP Web Cam

mattbruman, 2019-06-07:
I am using IP Webcam on android phone and because it takes a few seconds to take the snapshot octolapse says it fails. If I hit test it actually is successful every 3-4 times I hit test. Are logs needed? I think if we can just add a longer delay before it says it failed it would solve the issue, I have tried setting the timeout to the max but it still says failed.

Or maybe there is a better app to use on android that plays better with octolapse?

FormerLurker, 2019-06-07:
Check the time out setting in the camera profile.  That should solve your problem.

mattbruman, 2019-06-07:
I said I set it to the max but it still errors, its weird, because I can hit take snapshot in the web interface for the app and it pops up but for some reason octolapse says it fails.

FormerLurker, 2019-06-07:
Hmm.  Can you tell me what the error message says exactly?  Also, export your settings file (octoprint setting->Octolapse->export settings), and attach it?  Don't post it if your camera has a password, it will be in the settings file.  I need to deal with that soon.

FormerLurker, 2019-07-14:
Did you ever get this figured out?  I'm going to close the ticket for now, and will re-open it if you need more help, so just let me know!

Thanks for posting!

closed 2019-07-14 by FormerLurker