FormerLurker/Octolapse #355: [Request] Trigger camera by external signal for SD print

Block137, 2019-06-12:
#### If this is a feature request describe it here

A new mode that trigger camera when a signal is received from GPIO, to allow layer-based time lapse when printing from SD card.
Given that the G-code on SD contains commands to control one main board pin.
https://community.octoprint.org/t/sd-print-trigger-camera-at-signal-receive/10075/6

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-07-23:
@Block137, I'm still considering how best to accomplish this.  I'm thinking that the printer would trigger a pin, which could be connected to a GPIO pin on the Pi.  When Octolapse senses this pin is active (or deactivated, probably allowing for custom triggering) it would acquire an image.  If you want a stable timelapse, you could add the appropriate gcode on layer change to move the hotend, then signal the pin, followed by a pause long enough to acquire the snapshot.

My thoughts so far would be to add this as another trigger profile type called something like 'External Trigger'.  One of the triggers could be GPIO.  All Octolapse would do is watch for the GPIO signal as defined by the trigger, then grab an image on all enabled cameras.  I could probably also optionally trigger a pin or something once the capture is complete, but I can't think of any way to make that useful on the printer side.

While I'm at it I could also at HTML triggering via some API command, or plugin triggering via a custom hook since both of these would be pretty easy to do at the same time.

I'm considering adding this to V0.5.  Let me know what you think.

Block137, 2019-07-23:
Triggering a pin once capture is completed might be used to signal the printer to continue printing or move camera gimbal for panning timelapse.

IMO there should be options to choose:
1. What signal to look for (falling, rising or both)
2. Delay after signal detect
3. (Maybe out of scope) Do something other than snap such as turn on lights if controlled by Octopi.

FormerLurker, 2019-07-23:
Item 1 makes sense.  I sort of envisioned Pin number, rising/falling/toggle.

The standard camera delay will apply, so item 2 is already implemented.

Item 3 may already be implemented, but it's not the most straightforward.  One could use the camera profiles to trigger scripts or send gcode to the printer.  It would make a bit more sense to add some scripting/pin control to the trigger profile (currently called the snapshot profile, but changed to trigger profile in the next version) instead of the camera profile, however.

szafran81, 2019-10-19:
@FormerLurker I think that my req may have some thing in common with this one. It's pretty simple (to describe) - add a specific octolapse trigger that can be inserted info g-code that will trigger taking the snapshots.

Something like:
```@OCTOLAPSE```

This way one can completely control when and where the snaps are taken, and not to rely on any automatic detection of g-code moves.

FormerLurker, 2019-10-19:
@szafran81, you will be pleased to know that this already exists!  See the 'gcode trigger'.

szafran81, 2019-10-19:
Great, thank you.
Have to take a look at implementing this into my g-codes. Are there any examples somewhere on how it's used maybe ?

FormerLurker, 2019-10-19:
Well, the default gcode is ```snap```, so just add that to your layer change script in your slicer and you can see how it works.  I looked and I can't seem to find a good guide, but basically Octolapse will take a snapshot whenever it encounters the ```snap``` command.

You can even edit your printer profile and change the default command.  You could use @Octolapse if you want.

szafran81, 2019-10-21:
Allready changed it to @OCTOLAPSE :)
To bad that there are no use cases. I'll manage without it.
The one thing that would be nice to know before I'll be doing some test with it (it'll take me some time as my print queue will be done in a week or two) is if the set g-code does only snapshot or the whole thing that is setup (eg. goes to the specified point in settings, does stabilization, and then takes a snapshot).

FormerLurker, 2019-10-21:
It uses the current stabilization settings.  If stabilization is disabled, it just takes a snapshot.

szafran81, 2019-10-21:
Great, thank you.