FormerLurker/Octolapse #356: Octolapse - Camera Error snapshot-thumbnail-create-error

ThePapu, 2019-06-14:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: Version 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1.  Set up Octoprint on my Raspberry Pi 3 Model B + following tutorials using my Logitech C920 webcam, this was detected in octoprint for control and I could take a standard timelapse in Octoprint, great, but now I wanted to get octolapse working for that beautiful timelapse functionality

2. Set up Octolapse following instructions, making sure to add G90 for starting gcode in Cura and setting xyz axis to absolute in Octolapse

3. Started a print with gcode which was uploaded to OctoPrint sliced in Cura 4.1.0. and got the Camera Error Message detailed below.

4. Spent a few hours googling, reading this git hub, and I couldn't find any reference to the specific error message I was getting, so I created this ticket to see if it's a bug or something I've done wrong, I'm sorry but I don't understand how to solve or troubleshoot
"Exception: 'module' object has no attribute 'open'"

#### What should have happened?
A beautiful timelapse of the print from octolapse

#### What happened instead?
Received an Error message  from Octolapse 

"Octolapse - Camera Error
snapshot-thumbnail-create-error: Create Thumbnail - An unexpected exception occurred while creating a snapshot thumbnail. Check the log file (plugin_octolapse.log) for details. - Inner Exception: 'module' object has no attribute 'open'"

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: OctoPi 0.16.0

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3
Printer Firmware Version: 1.1.6

#### Browser and version of browser, operating system running browser

Browser: Mozilla Firefox
Browser OS: 67.0.2

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://pastebin.com/TgjLugWn

#### Link to settings.json

Link to settings.json with all passwords removed: 
https://www.dropbox.com/s/8yv6xjd1pf0eey1/Settings.json?dl=0

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/SW05JdhC

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/LkAG4rP2

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 

https://www.dropbox.com/s/ww6c0txzcnzopba/octolapse%20error.PNG?dl=0

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-06-14:
This is indeed unusual.  It looks like the pillow module wasn't loaded correctly, or there is some kind of version conflict.  I recommend [updating your pi](https://www.raspberrypi.org/documentation/raspbian/updating.md) and trying to manually update pip and pillow via the following:

```
pi@octopi:~ $ source oprint/bin/activate
(oprint)pi@octopi:~ $ pip install --upgrade pip
(oprint)pi@octopi:~ $ pip install pillow --no-cache-dir
```

ThePapu, 2019-06-14:
Thanks for the super speedy response, I'm very impressed!
I tried updating everything and sadly it didn't fix the issue, I got the same error message. 

So I redownloaded, reinstalled and reconfigured Octopi on a new SD card and it seems to be working just fine now! Hopefully my journey will help someone else!
All the best!

closed 2019-06-14 by ThePapu

FormerLurker, 2019-06-14:
No problem!  I only wish we figured out what exactly was happening, but I'm guessing Pillow failed to initialize, probably due to conflicts with the requirements.  Pillow has given me lots of install headaches, but it's pretty much the best game in town for working with the snapshot images.