FormerLurker/Octolapse #357: [Request] Delta printer help needed (From youtube comments)

yourics, 2019-06-20:
#### If this is a feature request describe it here

We had a chat in a youtube comment section. I can't seem to figure out how to get octolapse working on my anycubic kossel linear plus (delta) printer.
I was wondering if you could help me with this.
I'm having the issue right now where it can't determine it's current position and tells me to home the printer, but even if I do it still won't work.

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: Anycubic Kossel Linear Plus
Printer Firmware Version: Marlin 1.1.9

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://pastebin.com/Je0wVy7Q

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-06-20:
So it looks like your bed has a diameter of 230MM, and judging from your gcode, the center is at X0,Y0, so you will want to change the following settings in your Octolapse printer profile:

Print Volume: 
  Override Octoprint Printer Profile: Enabled
  X-Width Min: -115.0
  X-Width Max:  115.0
  Y-Depth Min:  -115.0
  Y-Depth Max:  115.0
  Z-Height Min:      0.0
  Z-Height Max:    300

Since you are using Marlin, I think that your printer should ignore any commands that move outside of your printer area.  To check this and to verify your origin, you could enter the following commands ONE AT A TIME(after removing the comments) in the Octoprint terminal window:
```
G90; set to absolute xyz
G28; home the printer
M114;  Get the current position.  It should respond with x:0.0 y:0.0 z:0.0 (Z might be higher than this)
G1 X0 Y0 Z5 ; Move the axis to the center and up a bit to prevent scraping the bed
G1 X-120 ; This might hit your end stop, but not by much.  I don't think it will, but if it does let me know
```
If the ```G1 X-120``` doesn't grind anything, it's time to try something a bit more extreme - try to move to the back left FAR PAST the edge of your bed:
```
G1 X-120 Y-120 ; This might hit your end stops too!  If not, the print head should be in the back left of your build plate and you are good!
```
If either the ```G1 X-120``` or ```G1 X-120 Y-120``` hits your end stop (or whatever the equivalent is on a delta) let me know because we'll need to do a few special things to the stabilization profiles to make them work.  I hope to fix this in an upcoming (possibly the next) release.  Octolapse will auto-detect your origin, but double check the results of the M114 command above and make sure it returns X0.0 Y0.0 (or something equivalent).  Also, please let me know what the Z coordinate is there if you would.

I have to warn you that I've never even stood in front of a delta, so I could have some fundamental misunderstanding about how they work.  Please proceed with caution.

Let me know if that helps!

yourics, 2019-06-20:
Hi, I get this when I type M114, note the z:
Send: M114
Recv: X:0.00 Y:0.00 Z:291.41 E:0.00 Count X:42205 Y:42205 Z:42205

FormerLurker, 2019-06-20:
Before I forget, I noticed that your gcode is sliced for vase mode.  I don't recommend you use Octolapse with vase mode prints since you'll get really bad quality issues.  I've got a new stabilization in the next (unreleased) version that works well for vases, so best that you wait for that.  However, if you do want to use it anyway, and you are using a layer trigger, make sure you change your 'snapshot' profile (soon to be renamed trigger profile) so that the 'Height' setting is above 0 (preferably to your layer height), else it will try to take a snapshot after every gcode command, and you don't want that.  I've also fixed this in the next release.

Now, back to business.  Where is the extruder after the G28?  Is it all the way at the top, or next to the bed?

Can you try this and let me know what M114 reports:
```
G90
G28
G1 X0 Y0 Z5
M114
```
I'm hoping it reports X:0.00 Y:0.00 Z:5.00.  The nozzle should be at the center 5mm away from the bed, let me know if it isn't.

yourics, 2019-06-20:
It's where you thought it would be. And this is the output :)
Send: M114
Recv: X:0.00 Y:0.00 Z:5.00 E:0.00 Count X:19292 Y:19292 Z:19292

The vase mode gcode was by accident. It was the first small gcode that I found. Do you need another gcode?

FormerLurker, 2019-06-20:
> Do you need another gcode?

No, it was just fine for my purposes.  I just wanted to make sure you knew of the implications of vase mode + octolapse.  

I'm a little concerned about the initial homing position, but I think it should be fine since the slicer will inject a Z coordinate before any actual printing happens.  It may report that it's got a 'position error' in the beginning of the print because of this, but if things are working well it will continue along just fine once it's sees a Z coordinate in any Gcode while your xyz axis is in absolute mode.

The other thing you might want to do is change your simplify settings to slice in relative E coordinates.  There are some bugs with absolute E in the current version (I believe these are fixed in the next release already).

Once you do that you should be ready to try a test mode print.  Slice a small calibration cube, set the Octolapse debug profile to 'Test Mode', use the 'Layer Trigger' and the 'Center' stabilization profile.  Then unload your filament and click print.  It should start immediately without warming your bed, nozzle, or extruding your filament (but better to remove it to be on the safe side).  If everything is good you should get a rendered timelapse (in the timelapse tab) once the test print is complete.  I recommend that you monitor the entire print, and be ready to power down if necessary.  I don't expect any problems, but better safe than sorry.  

If that goes OK, you'll be ready to try out a real print.  Just load your filament, switch the debug profile from 'test mode' to 'live print' and try again.  This time it will warm up the bed and extruder, and you should get a physical print and a timelapse video after (available in the 'timelapse' tab).

Let me know how it goes!  Fingers crossed...

yourics, 2019-06-20:
Ok so I've made some progress. It's not giving me the coordinates error anymore. However now it's not taking pictures..
These are my current settings. Let me know if you need to see one of these options in more detail 
[image]

FormerLurker, 2019-06-20:
It looks like you have two webcams enabled, and that the default webcam is taking pictured (judging from the snapshot that appears on the tab).  Try disabling the 'logitech webcam' by clicking the check box to the left, then edit the 'default webcam' profile and click on the 'test' button next to the 'Snapshot Address Template' setting and see if it reports success.  If it does, try your test print again.

yourics, 2019-06-20:
It told me the default webcam works OK and I have disabled the logitech one (its the same as default though).
Yet still stuck at only 1 snapshot

FormerLurker, 2019-06-20:
Can you post the gcode file you are using?

yourics, 2019-06-20:
Ofcourse, thank you for your help so far by the way. Can't wait to get this working :)
On my phone right now, so I had to post the file itself. Hope it's not a problem.

https://drive.google.com/file/d/1yTPFHqVFOTvxW44YD7yi0xkmtD5GyfFy/view?usp=drivesdk

FormerLurker, 2019-06-20:
I don't have access to download it.  Can you make it public for a few minutes?

yourics, 2019-06-20:
Oops my bad! Should be public now 😜

FormerLurker, 2019-06-20:
Ok, so I think that odd position reporting after a home is causing Octolapse to think that it is on a high layer (you can verify this by turning on the position info panel and see what layer/z height you are at during the test print).

You might be able to get it to work by changing this part of your start gcode:
```
G28   ;move to endstops
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
```
To this:
```
G28   ;move to endstops
G92 E0                  ;zero the extruded length
G1 Z0.5 ; Move the z axis to a known height of 0.5mm
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
```
Now your printer will prime at 0.5mm, then (at least based on your current gcode file) it will move to the proper height and start your print.  If my guess is correct, that should fix it.

yourics, 2019-06-20:
Alright will give that a try tomorrow. Time to go to bed.
Once again, thank you so much for your time so far!

yourics, 2019-06-21:
OMG it works!!!! :open_mouth: 
I am so so happy! :smile: 
One more question, is there any way to keep the images itself before they get rendered into a timelapse video? And does octolapse have a quality setting? If so, I'm using the logitech C920 and the quality is not as good as it should be.

You deserve a huge shoutout in my next video!!
You made my day :smile:

FormerLurker, 2019-06-21:
You can increase the bitrate in the rendering settings.  The default is pretty low.  Double or triple it for a nicer video.  You can also increase your resolution if your webcam supports it.  1080P at 10fps works pretty well.  Also, you can keep the images to render yourself, but you have to download them from your pi manually (working on improving this too).  Go to your rendering profile and turn off snapshot cleaning after rendering, but be sure to delete them at some point else your pi will get filled pretty fast :)

I'm glad to have helped you get this working!  Once you are sure your printer settings are good, maybe you can upload your settings.json file so I can add the profile to the new Octolapse profile library?  You can export your settings in Octoprint Settings (wrench/spanner icon)->Octolapse->Download Settings.

Send me a link to your next video please!  I'd like to see your results.

yourics, 2019-06-22:
I will improve the quality of the video output, then I'll look into moving the head into different positions, make a video and upload the settings.json file then once I'm 100% sure it's working.
I'll be sure to send you a video! :)

FormerLurker, 2019-07-14:
Did you ever get it working?  For now I'm going to assume so, and close the ticket.  I can always reopen it!

closed 2019-07-14 by FormerLurker