FormerLurker/Octolapse #358: Minor issue with

leoheck, 2019-06-22:
Speed and Features text issue, using Octolapse version `0.3.4`
[image]

FormerLurker, 2019-06-22:
This might have already been fixed in the dev branch.  When I have a chance to check I'll post back.

leoheck, 2019-06-24:
Ah, that is cool, thanks for the answer.

FormerLurker, 2019-10-16:
So, feature detection works differently now.  I"m no longer using speed based detection, and it's all handled via preprocessing.  Feature detection has been completely removed when using the real time triggers (all triggers in 0.3.4).  

Here is a screen shot

[image]

Feature detection is now based on gcode file comments, and only works with the new Smart Layer triggers (v0.4).  If the proper comments are available in the gocde file, feature detection automatically works.  If they are not there the user is warned of possible quality ramifications (not a popup).  Here is a screenshot of that:

[image]

Note that clicking on the question mark above opens help that details how to deal with these quality issues:

[image]

I know that involves more than the issue you reported, but I wanted to include the full story.

FormerLurker, 2019-12-16:
I've confirmed that the bug you reported is fixed, so I'm calling this CLOSED!  Thanks for the report!

closed 2019-12-16 by FormerLurker