FormerLurker/Octolapse #359: Multiples Error Messages

leoheck, 2019-06-22:
Using Octolapse version `0.3.4` and Octoprint `1.3.11`

This error message keeps popping periodically every 0.5 seconds.
I actually don't know if this issue belongs here but I am reporting it anyway.
[image]

I was not using any G90 on my gcode. It was an error. But the popup should be fixed. It keeps appearing/disappearing periodically.

FormerLurker, 2019-06-22:
Yes, the errors are rate limited, but will continue to show in v0.3.4.  Position errors rarely show in the new version.  I'll post back once I have a chance to verify the exact behavior you are seeing (missing G90, repeating error messages) has been fixed.

leoheck, 2019-06-24:
So, the fix should show a single fixed popup with issue octolapse is facing. Am I right?

FormerLurker, 2019-06-24:
I'm hoping.  It's on my list of things to verify before I deploy the dev build.  It's not a problem for the new pre-calculated triggers (new feature), since all of the code is preprocessed, and only shows one error message at the end of the precalculation process, but before the print starts.  I need to test the real time triggers (all of them in the current master release:  Gcode, Timer and Layer) to be sure those are still good to go.  I'll post back here once that's verified.

FormerLurker, 2019-10-16:
@leoheck, I think the position error has been removed entirely now.  In v0.4 the position state panel will open for real time triggers (those available in v0.3.4), showing all possible errors.  For the new 'Smart Layer Trigger' I've started adding custom messages that explain possible problems, and provide links to specific help for each issue.  This is all displayed exactly once (after gcode preprocessing but before the print starts) for the smart trigger, and there is an option to ignore the errors and continue printing without Octolapse (available in the main settings page).

I'm still working on the help files, and detecting problem situations, but basically the error messages have been augmented so that there is a 'Help' and 'Close' button.  The 'Help' button opens a popup with specific help for the situation.  There are still quite a few cases where there is no help offered, and some others where the help page is too broad, but it's being incrementally improved. 

Here is an example where some required slicer settings are missing from the gcode file showing both the error and the resulting help file (if you click help):

[image]

FormerLurker, 2019-12-16:
I have verified that this has been fixed.  I'm closing this issue.  Thanks for this and the other report, just sorry it all took so long to deal with :)

closed 2019-12-16 by FormerLurker