FormerLurker/Octolapse #362: Octolapse with DSLR not working with read-only root partition

The-M1k3y, 2019-07-05:
Ok, I might have a really weird problem here, so there are some additional information.
I run Octopi (and Octolapse) on a Raspberry Pi 3 and modified the system so that the root partition is mounted as read-only, with an additional home partition that is mounted rw.

As long as the root partition is mounted ro, no photos that the DSLR shoots appear in the snapshot directory, although the sub-directories for this timelapse are created. But when I take the photos manually
`/home/pi/scripts/take-snapshot.sh 0 1.0 /home/pi/.octoprint/data/octolapse /home/pi/.octoprint/data/octolapse/snapshots/44c91b60-18cd-428f-a6ee-1f3919b12b3d/4ff5eb89-de68-4c13-bd42-f9c320912eab huawei-stand_octolapse_0.2mm_PLA_MK3SMMU2S000000.jpg /home/pi/.octoprint/data/octolapse/snapshots/44c91b60-18cd-428f-a6ee-1f3919b12b3d/4ff5eb89-de68-4c13-bd42-f9c320912eab/huawei-stand_octolapse_0.2mm_PLA_MK3SMMU2S000000.jpg`
it works as intended.

The moment I remount the root partition as writable `mount -o rw,remount /` octolapse also works and the photos appear in the directories.

Partition layout:

> pi@octopi:~ $ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0  7.4G  0 disk
├─mmcblk0p1 179:1    0 43.1M  0 part /boot
├─mmcblk0p2 179:2    0    4G  0 part /
└─mmcblk0p3 179:3    0  3.4G  0 part /home

Mounts (shortened some system mounts like /proc or /sys for readability):

> pi@octopi:~ $ mount
/dev/mmcblk0p2 on / type ext4 (ro,noatime)
tmpfs on /var/log type tmpfs (rw,relatime,size=102400k)
tmpfs on /tmp type tmpfs (rw,relatime,size=102400k)
/dev/mmcblk0p3 on /home type ext4 (rw,noatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=89552k,mode=700,uid=1000,gid=1000)

#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.3.11 

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
1. Printing with "Script" Camera enabled, using the default script from the github wiki for a DSLR

#### What should have happened?
Photos should be taken and saved in the directory /home/pi/.octoprint/data/octolapse/snapshots/$id/$id/

#### What happened instead?
-The camera took the photo, but no file appears in the relevant directorys
-Error Message:
> /home/pi/.octoprint/data/octolapse/snapshots/e7421dbd-4a91-4e22-927a-b81509a6d645/4ff5eb89-de68-4c13-bd42-f9c320912eab/huawei-stand_octolapse_0.2mm_PLA_MK3SMMU2S000000.jpg'.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version
Also happens on the virtual test printer, so model and version are irrelevant to the problem

#### Link to settings.json
Link to settings.json with all passwords removed: https://pastebin.com/JYvRu22R

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://gist.github.com/The-M1k3y/5def1afa657084d52205dbe6d0481d0c

#### Link to octoprint.log
Link to octoprint.log:  https://gist.github.com/The-M1k3y/1defd4d647d441f076c2ec6b26c8b2d5

FormerLurker, 2019-07-05:
Ok, thanks for reporting this.  My Linux skills are poor atm, but i'll see what i can find out after this weekend (family stuff going on).  Also, thank you for filling out such a detailed report!  That is much appreciated 👍

The-M1k3y, 2019-07-05:
Take your time, personal life goes first. I can work around this by remounting as rw. I don't really like it, but it works.
If I can assist in any possible way, let me know.

FormerLurker, 2019-07-05:
I probably will take you up on that, thanks!

The-M1k3y, 2019-07-17:
I did some more testing today and had a look at the environment variables and I found the culprit. When Octolapse calls the script, the PWD variable (current working directory) is '/'. During all previous tests, I called the script from the home directory. When calling the script myself from the root directory I can replicate the error. When adding the line `cd /home/pi` to the start of the script it works.

I have no idea why, but it seems that gphoto2 must be called from a writable directory.

FormerLurker, 2019-07-23:
Glad you got this solved.  I'm going to close the issue.  If you have any other issues I can reopen it.  Let me know and happy Octolapsing!

closed 2019-07-23 by FormerLurker